mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #2: Slowdown (or little freeze) during the animation

FrenchLab47, 2018-11-23:
Hello,

I come back to you to ask if you have noticed a slowdown problem during the running of the animated gifs. I first asked the author of the AnimatedGIFs library, I specified all the details that I observed, but it does not use too much esp32.
If you have a moment, I suggest you read directly what I wrote here :
https://github.com/pixelmatix/AnimatedGIFs/issues/24

It's not a serious or urgent problem, it's just to know if you have an idea :-)

Best regards.
Patrick BESSE.

mrfaptastic, 2018-11-27:
Hi Patrick,

Unfortunately I have no idea why this might happen. The animated gif example with this library is just a proof of concept to test the display. I suspect there must be some memory read event that occurs with a large .gif that might cause this. ESP / FreeRTOS stuff I guess?

FrenchLab47, 2018-11-27:
That's also what I think. A memory stack that saturates, or something like that. I'll do other tests, and if I find something I'll come back to tell you.
Thanks.

mrfaptastic, 2018-12-30:
Hi FrenchLab47, did you ever figure out the cause of the pause?

FrenchLab47, 2019-01-01:
Hi mrfaptastic,

No, sorry, but I have not found anything new to mitigate, or cancel this defect.
I wanted to test with other models of ESP32, but I did not have time.
Currently I am looking for solutions to animate text, more attractive than simple scrolling.
As I am not a programmer or developer, I am not fast and I spend a lot of time.
In fact I'm more comfortable with a soldering iron 

I will, in a few times, inform you of my progress on this project.
Cordially.

mrfaptastic, 2019-01-03:
Hi FrenchLab - I'm not really a programmer either so I understand (Code, cross-fingers, compile, fail, fix, repeat type of coder)

> Currently I am looking for solutions to animate text, more attractive than simple scrolling

I might have something you want here. I have modified a version of [MD_Parola](https://github.com/MajicDesigns/MD_Parola). This library was originally built of single color 8x8 pixel blocks that could be 'chained' together to make a matrix text display (typically join 8 or so of these together - so 64 wide x 8 pixel high)

https://www.youtube.com/watch?v=X8fuDIhxNH0

What I have done is modify this library to split one of these panels into four lines of 64x8 pixels.

Text can then be animated, but the restriction is that each line is only 8 pixels high - as per the original library. This might be of use, if so I'll try get it working reliably and create a git repository.

FrenchLab47, 2019-01-03:
Hi mrfaptastic,

I also used MD_Parola with MAX7219 modules, and indeed, the text animations are very pretty.
For my part, I try to develop a class with some effects.
At the base I use Adafruit_GFX, and U8G2 with unicode fonts, (French accents, euro symbols, etc).
For now I can display basic scrolling effects, in several directions and it works well with all font sizes.
I can also animate each character of a string, independently. It's almost like a "type writer" effect, or the "slide" effect of MD_Parola.
I can also combine several lines, with different fonts, colors, speed, and effects.

But that's all for the moment, for the future, I would like to isolate the pixels of each character to make other effects.
I'll go a little further, clean my code (very rough) and I'll publish it here.

csloz, 2019-04-18:
@FrenchLab47 

The Issue is the SPIFFS filesystem gets sluggish with seeks on larger files.
I made some testing on FFAT vs SPIFFS with the same files I was having slowdown issues with.
When using FFAT it runs smoothly without the slowdown.

Marc Merlin has done some work on this in GitHub that can easily be ported to this library (mostly its a matter of adding another ifdef in the animated gif_config.h file and removing (commenting out) some of the ifdef's for smartmatrix in Gifdecoder_lmpl.h

https://github.com/marcmerlin/AnimatedGIFs

I maybe see if he wants to patch his code to be compatible with this lib, its a 5 min job +-

FrenchLab47, 2019-04-18:
Hello csloz.
This idea seems entirely logical. I would try to test that as soon as possible.

Thanks

FrenchLab47, 2019-05-01:
Hello all,

I'm coming back here to confirm that animated Gifs work well, without slowing down, when they are stored in a FAT partition.
It's easy to test with an SD card, but my application uses only Flash memory with some images and my files html, css, js. And I also wanted to see the loading improvement on my html pages.

The new version of ESP32 ARDUINO offers FAT formatting options for flash memory (select ESP32 Dev module). But I did not find a simple way to upload DATA files into the FAT partition.
The code examples for ISP-IDF are not up to date, I have tried several without success.

I only managed to create a .img file (with some 8.3 files, and no folder)
But no code or no python command can write in my flash.
The write operation proceeds without error (0 ... 100%) but no file is written.
Same result with ESP DOWNLOAD TOOL: The writing is complete at the correct address, but the FAT partition remains empty.

Finally the only solution that worked for me is the following (in three steps)

1 - To begin, I load a sketch of FTP server (ESP32FtpServer) on my ESP32 (ARDUINO IDE with FAT default formatting).
2 - I use FILEZILLA to access the FTP server of the ESP32, and I transfer my files in the FAT partition.
3 - Then I load the sketch of my application (the FTP server is overwritten and replaced, but not the data files)

After that, my files are readable in the FAT partition :)

It remains an important problem: I need to reload the 2 sketches every time I modify a Data file (html, css or js).
I tried to integrate the FTP server in my application but it causes errors with the web server (AsyncWebServer).

Looking forward to find a simpler upload solution ...
I will ask the question in other forums.

Thanks for your help (and google translate)

csloz, 2019-05-04:
Follow instructions for working esp idf toolchain here - https://docs.espressif.com/projects/esp-idf/en/latest/get-started/
Git clone the tool below, and build, can then create FFAT images.
https://github.com/TobleMiner/mk_esp32fat

Alternately clone Marks repo and do same - http://marc.merlins.org/perso/arduino/post_2019-03-30_Using-FatFS-FFat-on-ESP32-Flash-With-Arduino.html

Let me know if you need more help.

I've uploaded my (work in progress) here using his newer gif libraries - https://github.com/csloz/pixelated/

FrenchLab47, 2019-05-06:
@csloz 

Thank you for your links and your ideas.
I have already tried all this for several days, and for now I have not found any solution that runs on Windows 10.
I tested several versions of ESP-IDF, in d: then in c:
Everything is well configured, but MAKE always causes path errors with these components that seem made for Linux.
I also tried with PlatformIO, but it brings me nothing more than Arduino IDE that I master better.
I think I will keep my FTP server, waiting for these components to work on Windows.

Thanks

csloz, 2019-05-06:
Getting a bit offtopic here I think, as this issue is solved (or at least not mrfaptastic's code)

Maybe you can try this as a solution for you:

In Windows 10 you can install a ubuntu subsystem.
https://docs.microsoft.com/en-us/windows/wsl/install-win10

Once installed, there is a compiled version of the mkfatfs tool for ubuntu in here - https://github.com/marcmerlin/esp32_fatfsimage
- this file
( https://github.com/marcmerlin/esp32_fatfsimage/blob/master/fatfsimage )

That can be run inside the ubuntu subsystem in windows 10 to create the filesystem if you can't compile otherwise.

Hope that helps!

mrfaptastic, 2019-06-19:
Closing as not related to this library.

closed 2019-06-19 by mrfaptastic

mrfaptastic, 2020-08-10:
FYI, updated the AnimatedGIF example based on yet another ['AnimatedGIF'](https://github.com/bitbank2/AnimatedGIF) library which is super fast and doesn't suffer from these stuttering issues.

https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/tree/master/examples/AnimatedGIFPanel