mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #4: Make ESP32 pinout compatible between your lib and SmartMatrix?

marcmerlin, 2018-12-25:
Howdy again, I tried your library to see if it would look better on video than SmartMatrix which is ok with the naked eye, but looks very flickery on video, see
https://community.pixelmatix.com/t/working-around-sync-offsets-and-flicker-when-filming-taking-pictures/346
https://www.youtube.com/watch?v=1f2aDTCmLcE

Louis recommended that I try your lib, which I got working in 32x64 (not 64x64 yet). One issue though is the two libs use different pinout. See 
https://github.com/pixelmatix/SmartMatrix/blob/teensylc/src/MatrixHardware_ESP32_V0.h#L64
Would you be willing to change your pinout to be compatible with it so that people can switch between both libs without rewiring?

ESP32-RGB64x32MatrixPanel-I2S-DMA.h:
#define R1_PIN  2
#define G1_PIN  15
#define B1_PIN  4
#define R2_PIN  16
#define G2_PIN  27
#define B2_PIN  12

#define A_PIN   5
#define B_PIN   18
#define C_PIN   19
#define D_PIN   21
#define E_PIN   22

#define LAT_PIN 26
#define OE_PIN  25

#define CLK_PIN 22

Thank you and Merry Xmas,
Marc

marcmerlin, 2018-12-25:
BTW, it looks like you may have been able to use my ESP32 AnimatedGifs port. If so, glad it worked https://github.com/marcmerlin/AnimatedGIFs

mrfaptastic, 2018-12-30:
Hi Marc, I'll look to enhance it so the pin-out can be defined on loading of the library. The pin is currently as it is such that hooking up the LOLIN D32 to one of these HUB75 panels is a lot more straight-forward, in that the pins on one side of the HUB75 connector connect to pins on one side of the LOLIN D32 (top to bottom) etc.

As for the AnimatedGif port - I based this off the example code in the original SmartMatrix library. I'll look at your enhancements and potentially just port your example instead. Good work.

marcmerlin, 2018-12-31:
Ah yes, the stupid pinout of ESP32 chips that is different on every other chip, what a mess :(
I get what you're saying on having pins that happen to be contiguous on one side of the chip, that's convenient. 
Do I however understand that you're not using a level shifter (74HC245 or equivalent)? If so, nice to know it works without one.

For AnimatedGifs, you can look at 
http://marc.merlins.org/perso/arduino/post_2018-07-13_AnimatedGIFs-for-SmartMatrix-or-NeoMatrix-_Neopixel-WS2812B_-from-SDcard-or-SPIFFS_-on-Teensy-3_x_-ESP8266_-or-ESP32.html for a quick demo, but mostly it adds the file browser (next/previous), jump to gif #x, and gamma correction.

See https://github.com/marcmerlin/AnimatedGIFs/blob/master/AnimatedGIFs.ino#L313

Feel free to close this issue, I totally understand that making 13 or so pins arguments of the lib, is kind of a pain. It's not so bad to change them in the .h file. Cheers.

marcmerlin, 2019-01-03:
Cool, thank you.
I'm still stuck on https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/issues/3 which sadly is a dealbreaker for me, but on the plus side, it gave me the opportunity to add a GFX + FastLED layer to SmartMatrix which supports up to 128x128. It doesn't do DMA though, but it supports 24bit color instead of 12bit color. https://github.com/marcmerlin/SmartMatrix_GFX  and demo code https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos 
I'll send you another PR with details. Cheers

marcmerlin, 2019-01-03:
This issue is resolved, so I'll close it.
Thanks.

closed 2019-01-03 by marcmerlin