mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #6: Animations can look bad on video to refresh sync mismatch between camera and panel

marcmerlin, 2018-12-26:
See https://www.youtube.com/watch?v=MZSxv4Knk0c

This is probably totally unfixable, and in some way the artifact looks better on your library than SmartMatix since yours does full frame updates before looping back for PWM levels.
That being said, it seems that there is no good way to get those panels to look good in pictures or video whether you do PWM per line or PWM per full frame.

Feel free to close this if you agree that there is no good fix and it's an issue with the technology. Fix of course would be to use a neomatrix panel instead, but those don't come in pitch 5, 4, or 3 last I checked.

marcmerlin, 2018-12-31:
Hi. Thanks for your answers.
I tried 
        if (actualReactualRefreshRatefreshRate > 500) // HACK Hard Coded: Minimum frame rate of 160
and it totally fixes the problem for me

        ramrequired += 100000; // HACK Hard Coded: Keep at least 100k free!
I wasn't sure how to adjust it, I tried to double it and it made the picture dimmer.

As why to film/take pictures, many of those panels are used in displays seen by others who in turn take pictures, and they look truncated with black bars if the refresh is too slow.

closed 2019-04-28 by mrfaptastic

huukhoa1412, 2020-08-10:
How to Sync mismatch between camera and panel when play gif ?

marcmerlin, 2020-08-10:
It's not a matter of syncing, the panels need to refresh faster than the shutter speed of the camera.  
IF the panels cannot go faster, then you have to slow down the shutter speed of the camera.