mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #7: Please help review RGBPanel library comparison

marcmerlin, 2018-12-31:
If you have a bit of time, could I ask you to review this writeup I just did?

The choice of APIs and backend drivers can be a bit confusing, so I've tried to write a summary.
https://github.com/marcmerlin/SmartMatrix_GFX/blob/master/README.md

Could you look it over and send me edits if applicable you can click edit in github and it'll send me a pull request.
Once it looks ok, I'll share it with Adafruit, Louis and you so that each lib can summarize the ecosystem out there and help users make the right choice for their needs.

mrfaptastic, 2020-08-10:
Hi. I never responded to this directly, but your summary seems in-line. I don't think there's a great deal between SmartMatrix and this library other than the fact that this was originally built as a drop-in replacement for people familiar with Adafruit GFX.

Thanks to Covid though I’ve had some time in the past few weeks to work on the library that I never had in the past two years, so have fixed a bunch of memory issues and Brian Lough’s fantastic contribution helped resolve chaining support.

The difference between this library really is that it doesn't provide any fancy stuff like layers etc. It's really a raw pixel co-ordinate to DMA interface, to keep memory usage to the bare minimum. In any case, your GFX library provides this higher level pixel-persistence stuff, which I'll look at next.

marcmerlin, 2020-08-10:
Thanks for the update, and good job with the progress.
So, what follows is an observation and not a critcism :)

I've been working with OSS for close to 30 years, and I've seen so many examples of  several project that solve 90% of the problem, but not the same 90% :) which in turn makes it difficult for users to pick what's the best code for them to use.

I'm not blaming you for writing this, it's a really cool learning project (I did a simpler version for RGB Matrices http://marc.merlins.org/perso/arduino/post_2015-01-06_Driver-for-direct-driving-single-to-3-color-LED-Matrices-with-software-PWM.html  and just got lucky that no one else had, before me).

I also agree with you that I don't need the extra layers and everything in the SmartMatrix API. My SmartMatrix::GFX glue library ignores all of it and uses the library as a framebuffer :)
Are you able to drive 128x128 on ESP32? SmartMatrix stops around 128x100 or so if you have PSRAM
If you can do, 128x128, then your lib is a clear win over SmartMatrix.
If not, it's a different lib with a different 90% support :)

As for the last 10% which sadly is becoming more than 10%:
There is the issue that panels are getting more of a mess. Between FM6126A (now FM6127) chips that are incompatible, and Louis who barely got it working for one panel he had but it never really worked for me, and the many more ways those panels can be messed up (ABCDE vs ABC for 128x64, or the AB panels, and all the different ways of doing row mapping to save a tiny bit of copper traces, it's a terrible mess).
I'm sure you're aware of  https://github.com/hzeller/rpi-rgb-led-matrix which is currently the best lib for supporting as many of those panels as possible (and yet, people still end up getting new weird ones at random), but it's only for rPi.

Ths is kind of where I wish that the two libs could join effort/code to help solve that 10%, but that's just my wish, and I totally understand that you get to work on what's interesting and worthwhile to you :)

marcmerlin, 2020-08-10:
In the meantime, because I've grown beyond 128x64, I've switched to rPi and wrote glue code/library to run arduino code on rPi:
[image]

http://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html

While you can't use the rpi code as is, if you encounter weird panels, you may be able to re-use the support that's already been written in that lib.
Have fun with your project, and thanks for the update.

mrfaptastic, 2020-08-11:
Totally agree with you on the missing 10% issue, but agree that's open source since well since open source began... "Will XXXX be the year for Linux on the Desktop" is a classic example.

RPi/SBC for any serious projects (like yours) should be the go-to choice anyway I think (or even better an FPGA) so a vastly more capable library makes sense. I'll look to see if there's portable code of use to this Arduino library, but for now nobody has raised any issues or a pull request for the FMXXXX chips. Perhaps it's more of an issue for non-32x64 panels.

To be honest though, if the Chinese start pumping out piles of crap panels to save a dollarh, then I'll probably just design a PCB myself that will always work with the library, or provide a list of 'working panels'. Rather support the panels that don’t have shortcuts.

> 128x128 should be easily doable.

marcmerlin, 2020-08-11:
Eheh, I'm not surprised to read we've had the same experiences :)
Few comments:
- I'd love an easy to buy small FPGA that does this and takes some FB input I can feed from a uC, but haven't found one with code yet (not really wanting to write that)
- FM chips, you are correct, I have mostly only seen them on 128x64, but I do have an array made out of 64x32 panels with FM6126A that were returned by users who couldn't use them. It's the exception more than the rule, though
- and yes, it's super annoying that the chinese keep finding new incompatible ways of making those panels, my hat to you if you can get your own designed to your spec.

What is the highest res you can do right now even if you don't have the panels? Can you blind output 128x128 at 100Hz or so? If so, that makes your library more capable than SmartMatrix on ESP32.

mrfaptastic, 2020-08-11:
The library, like SmartMatrix drops the colour gradients / PWM depth silently if there's going to be memory constraints, or refresh issues. Also, how things are chained, what GPIO bus speed is set and what other junk is running in the sketch influences this. Quick compile and run on a ESP32 floating around:

128x128 works
256x128 works, but I think it's dropped to about 5-6bit colour.

```
Raised lsbMsbTransitionBit to 0/4 to fit in remaining RAM
lsbMsbTransitionBit of 0 gives 19 Hz refresh: 
lsbMsbTransitionBit of 1 gives 38 Hz refresh: 
lsbMsbTransitionBit of 2 gives 67 Hz refresh: 
lsbMsbTransitionBit of 3 gives 101 Hz refresh: 
Raised lsbMsbTransitionBit to 3 to meet minimum refresh rate
```

But in the end, it depends on how it looks to the naked eye.

marcmerlin, 2020-08-11:
256x128 for an ESP32 is really great. The way SmartMatrix manages memory and bitplanes, it cannot reach 128x128 (runs out of RAM).
So, congratulations, your lib and how you manage memory, is 3-4x more efficient than SmartMatrix.

You definitely should advertise this on your main page, something like "Support for up to 128x256 at reduced color depth on ESP32 when SmartMatrix tops out a bit after 128x64"

Without having looked at the SmartMatrix code in details, it still uses more bitplanes than you do, or uses them in a way that is less memory efficient than you. Well done!

marcmerlin, 2020-08-15:
I posted your update in https://community.pixelmatix.com/t/up-to-256x128-on-esp32-with-esp32-rgb64x32matrixpanel-i2s-dma/733 . This will hopefully help people looking for 128x128 or above on ESP32.

Galaxy-Man, 2020-08-24:
mrfaptastic, thanks for the updated library. All works no issues, not got a SD card so running in memory. Waiting for ribbon cables to test 2x2.

If you are thinking of going bigger let me know and Ill get you some more panels sent over. Im currently using FM6124/FM6126 and seems to be fine.

mrfaptastic, 2020-08-24:
> If you are thinking of going bigger let me know and Ill get you some more panels sent over. Im currently using FM6124/FM6126 and seems to be fine.

As always, you're more than kind.  A FM6124/FM6126 panel(s) would be of use to test with. @atc1441 seemed to imply that the FMXXXX example doesn't work and he had to do additional stuff. https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/issues/23#issuecomment-677929742 

Interested to know what happens when chaining non-FM with FM models of panels together... this will no doubt be the next problem as people chain whatever they have laying around in the basement, together.

Galaxy-Man, 2020-08-24:
FM6126 + one of the icnd2038s panels I sent to you do work together. I loaded the reset routine and the icnd2038s panel ignored it but the FM accepted and both work together. I'll try and get two icnd2038s panels and built 2x2, it's like roulette when prdering even from Amzon. Watch this space.

embedded-creations, 2020-09-04:
> Copyright (c) 2015 Pixelmatix

You might want to update your license file :-)

Galaxy-Man, 2020-09-04:
Hi Louis

I am being stupid here, which licence file?

mrfaptastic, 2020-09-04:
@Galaxy-Man I think that's a message to me. @embedded-creations - I'm all for giving credit where credit is due ha!

closed 2021-02-22 by mrfaptastic