mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #8: all columns shifted one pixel left, column 0 is shown on right side of panel

Markusenz, 2019-01-29:
Thanks for this work, I plan to use that lib as it works best of all I saw.
However one thing I cannot easily find how to correct:
I use a standard p3 panel 64x32, 1/16 scan. All works so far, but all pixels with x=0 show on the rightmost column instead of the leftmost, and all pixels with x=63 show on column 63 instead 64. So all pixels are shifted left one column, and the leftmost appears at column 64.

I wonder why since my panel is standard and I also used it already with other libs.
Thanks in advance!

Markusenz, 2019-01-29:
Another problem is it starts to flicker as soon as WiFi is active (as soon as the ESP32 joins an access point), which makes it useless for me...

mrfaptastic, 2019-01-29:
Can you post an example sketch of your code ? Ghosting is probably caused by electrical interference due to the wiring - not much that can be done there. What module are you using?

Markusenz, 2019-01-30:
Thanks a lot for your reply. In the meantime I updated to current ESP32 Arduino core 1.0.1, and the ghosting when using WiFi has gone.
However I still see the problem with columns shifted one pixel to the left. I attach the link to the picture showing that, and my code:
[https://imgur.com/a/wBa1YCS](url)
``` 
#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>

const byte ROWS=32;
const byte COLUMNS=64;

#define R1_PIN  17
#define G1_PIN  16
#define B1_PIN  4
#define R2_PIN  0
#define G2_PIN  2
#define B2_PIN  15

#define A_PIN   21
#define B_PIN   19 
#define C_PIN   18
#define D_PIN   5
#define E_PIN   -1
          
#define LAT_PIN 3
#define OE_PIN  23
#define CLK_PIN 22

RGB64x32MatrixPanel_I2S_DMA matrix; 

void weatherDemo() {
  byte br = 127;
  auto white = matrix.color444(br, br, br);
  auto red = matrix.color444(br, 0, 0);
  auto cyan = matrix.color444(0, br, br);
  auto blue = matrix.color444(0, 0, br);
  matrix.fillScreen(matrix.color444(0, 0, 0));
  matrix.setTextWrap(false);
  matrix.setTextSize(2);
  matrix.setCursor(2, 1);
  matrix.setTextColor(white);
  matrix.print("12");
  matrix.print(":");
  matrix.print("34");
  matrix.setCursor(2, 16);
  matrix.setTextColor(red);
  matrix.print("abcdef");
  matrix.drawPixel(0, 0, matrix.color444(br, br, br));
}

void setup() {
  Serial.begin(115200);
  matrix.begin(R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN );  // setup the LED matrix
  matrix.fillScreen(matrix.color444(0, 0, 0));
  weatherDemo();

}

void loop() {
  delay(100);
}
```

Markusenz, 2019-01-30:
When I set a green pixel at (0,23), the pixel is shown also on the right side, and the ghosting pixel one row above with reduced intensity, see second picture on that page:
[https://imgur.com/a/wBa1YCS](https://imgur.com/a/wBa1YCS)

  matrix.drawPixel(0, 0, matrix.color444(255, 255, 255));
  matrix.drawPixel(0, 23, matrix.color444(0, 255, 0));

mrfaptastic, 2019-03-22:
In line 251 of ESP32-RGB64x32MatrixPanel-I2S-DMA.cpp, there is:

`        // need to turn off OE one clock before latch, otherwise can get ghosting
        if((x_coord)==PIXELS_PER_LATCH-1) v|=BIT_OE;`

Can you comment out this bit and see what happens?

If that changes nothing, on line 236 there's:
'        // drive latch while shifting out last bit of RGB data
        if((x_coord) == PIXELS_PER_LATCH-1) v|=BIT_LAT;'

Can you change it to:

'        // drive latch while shifting out last bit of RGB data
        if((x_coord) == 0) v|=BIT_LAT;'

And see what happens?

I can't replicate this issue with my panel. Not all of these panels are made exactly the same it seems.

Thanks.

closed 2019-04-28 by mrfaptastic

colonelwatch, 2021-06-07:
Hello, I see this issue went stale, but I have a panel with a similar problem. All of the columns are also shifted one pixel left, where column 0 is on the right side of the panel. I don't know if this is relevant, but there is also a little bit of vertical ghosting in column 0 (nowhere else). My panel is 64x64 and a MBI5124 variant, and I put together my code from the examples.

[image]
[image]

I see that you mentioned some possible fixes before you closed the issue, but I can't find those lines of code in the most recent release.

```c++
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <FastLED.h>

// Configure for your panel(s) as appropriate!
#define PANEL_WIDTH 64
#define PANEL_HEIGHT 64    // Panel height of 64 will required PIN_E to be defined.
#define PANELS_NUMBER 1   // Number of chained panels, if just a single panel, obviously set to 1
#define PIN_E 32

MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t colorWheel(uint8_t pos) {
  if(pos < 85) {
    return dma_display->color565(pos * 3, 255 - pos * 3, 0);
  } else if(pos < 170) {
    pos -= 85;
    return dma_display->color565(255 - pos * 3, 0, pos * 3);
  } else {
    pos -= 170;
    return dma_display->color565(0, pos * 3, 255 - pos * 3);
  }
}

void drawText(int colorWheelOffset)
{
  // draw text with a rotating colour
  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // Don't wrap at end of line - will do ourselves

  dma_display->setCursor(5, 0);    // start at top left, with 8 pixel of spacing
  uint8_t w = 0;
  const char *str = "ESP32 DMA";
  for (w=0; w<strlen(str); w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print(str[w]);
  }

  dma_display->println();
  dma_display->print(" ");
  for (w=9; w<18; w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print("*");
  }
  
  dma_display->println();

  dma_display->setTextColor(dma_display->color444(15,15,15));
  dma_display->println("LED MATRIX!");

  // print each letter with a fixed rainbow color
  dma_display->setTextColor(dma_display->color444(0,8,15));
  dma_display->print('3');
  dma_display->setTextColor(dma_display->color444(15,4,0));
  dma_display->print('2');
  dma_display->setTextColor(dma_display->color444(15,15,0));
  dma_display->print('x');
  dma_display->setTextColor(dma_display->color444(8,15,0));
  dma_display->print('6');
  dma_display->setTextColor(dma_display->color444(8,0,15));
  dma_display->print('4');

  // Jump a half character
  dma_display->setCursor(34, 24);
  dma_display->setTextColor(dma_display->color444(0,15,15));
  dma_display->print("*");
  dma_display->setTextColor(dma_display->color444(15,0,0));
  dma_display->print('R');
  dma_display->setTextColor(dma_display->color444(0,15,0));
  dma_display->print('G');
  dma_display->setTextColor(dma_display->color444(0,0,15));
  dma_display->print("B");
  dma_display->setTextColor(dma_display->color444(15,0,8));
  dma_display->println("*");

}

void setup() {
  HUB75_I2S_CFG mxconfig;
  mxconfig.mx_height = PANEL_HEIGHT;      // we have 64 pix heigh panels
  mxconfig.chain_length = PANELS_NUMBER;  // we have 2 panels chained
  mxconfig.gpio.e = PIN_E;                // we MUST assign pin e to some free pin on a board to drive 64 pix height panels with 1/32 scan
  mxconfig.driver = HUB75_I2S_CFG::MBI5124;
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  dma_display->begin(); // use default pins
  dma_display->setPanelBrightness(64);
  dma_display->setLatBlanking(2);
}

uint8_t wheelval = 0;
void loop() {
  drawText(wheelval);
  wheelval +=1;

  delay(20); 
}
```

ovdessel, 2023-10-02:
Experienced the same issue as the above comments. First column (x_coord = 0 ) was shifted to the right side of the board rather than the left. 

The solution involved editing ESP32-HUB75-MatrixPanel-I2S-DMA.cpp line 366
`/* When using the drawPixel, we are obviously only changing the value of one x,y position,
   * however, the two-scan panels paint TWO lines at the same time
   * and this reflects the parallel in-DMA-memory data structure of uint16_t's that are getting
   * pumped out at high speed.
   *
   * So we need to ensure we persist the bits (8 of them) of the uint16_t for the row we aren't changing.
   *
   * The DMA buffer order has also been reversed (refer to the last code in this function)
   * so we have to check for this and check the correct position of the MATRIX_DATA_STORAGE_TYPE
   * data.
   */
  x_coord = ESP32_TX_FIFO_POSITION_ADJUST(x_coord)+1;

  uint16_t _colourbitclear = BITMASK_RGB1_CLEAR, _colourbitoffset = 0;
`

mrfaptastic, 2023-10-09:
> Experienced the same issue as the above comments. First column (x_coord = 0 ) was shifted to the right side of the board rather than the left.

Did you try adjusting the 'clkphase' value?