mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #9: 64x64 Panel

csloz, 2019-03-22:
FYI - 

I have tested with a P2.5 64x64 panel with Line E, and H file MATRIX_HEIGHT modified for 64x64

Testing working, and have a plasma test running, but the colors seem slightly different on the left / right side, which looks like image is cut in 2.

I see the code is doing Panel Upper / Panel Lower, will fiddle a little see if I can resolve.

[image]

mrfaptastic, 2019-03-22:
Hello. Thanks for testing this out on a 64x64 panel, good to know it works in theory. Can you test with just a single color on the whole display? Red, green or blue? Does this happen with any other libraries? 

Not sure why this would occur. It might be related to #8

csloz, 2019-03-22:
I have ordered some other panels to test, will see next week if its panel or other issue.

On another note, would be happy to send you a pcb or two that I've made, makes things much easier for interfacing to the ESP32  Let me know via lawrence at computersolutions dot cn, and I'll send you some.

csloz, 2019-03-24:
Was something funky on that panel, works well on another one. - no color iffyness on bottom.

[image]

[image]

closed 2019-03-24 by csloz