mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #17: Brownouts / Power usage increase on new version of library

csloz, 2019-07-30:
Good news - no more green boogies.

Bad news - the power consumption seems to have  increased.
Before I could run plasma test on USB for days without issue, now it lasts about 1/2 second before rebooting without external power.

New library draw - 
10% coverage (Conway game of life)
5v/ .46A - Brightness 50
5v/ .32A - Brightness 20
5v/ .27A - Brightness 10 
5v/ .25A - Brightness 5

100% coverage (plasma)
5v/ .36A - Brightness 5
5v/ .48A - Brightness 10
5v/  .59A - Brightness 15
5v/ .70A - Brightness 20
Over that = brownout/reset.

Win some, lose some!

Will add in a cap to see if that improves USB power brownout, otherwise will need to add in some code to try modulate power if its going to go over hard limits.

mrfaptastic, 2019-07-30:
I'll look at this again in the coming days. That is too much of a
regression.

I like my RGB panels, but prefer if they're more energy efficient.

mrfaptastic, 2019-07-30:
Resolved with better library defaults.

closed 2019-07-30 by mrfaptastic