mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #18: More than 2 displays

Spindlerkoo, 2019-07-30:
Hi,
how many 64x32 RGB displays can your library control?

Thanks, Adam

mrfaptastic, 2019-07-30:
It's a hardware limitation opposed to a software one. ESP32 only has a small amount of memory, so you'll be hard pressed to show any more than 64x64 pixels with 24bit RGB.

You could experiment with the color depth etc. However, two chained displays would be the limit. I have updated the readme.

Spindlerkoo, 2019-08-01:
And which memory? Flash or SRAM?

mrfaptastic, 2019-08-01:
SRAM

Spindlerkoo, 2019-08-05:
And will this help? https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/external-ram.html

mrfaptastic, 2019-08-05:
I can't say as I have never used such a thing. Feel free to try it out and report back.

mrfaptastic, 2019-08-05:
Hi,

I just had a read of the link you posted, and it says under the 'Restrictions' section:

> External RAM cannot be used as a place to store DMA transaction descriptors or as a buffer for a DMA transfer to read from or write into. Any buffers that will be used in combination with DMA must be allocated using heap_caps_malloc(size, MALLOC_CAP_DMA) and can be freed using a standard free() call.

So you will not be able to use this.

closed 2019-08-05 by mrfaptastic