mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #19: drawIcon

csloz, 2019-08-14:
Another small function for the library. 
Could I get commit access?

void drawIcon (int *ico, int16_t x, int16_t y, int16_t h, int16_t w);

inline void RGB64x32MatrixPanel_I2S_DMA::drawIcon (int *ico, int16_t x, int16_t y, int16_t cols, int16_t rows) {
  int i, j;
  for (i = 0; i < rows; i++) {
    for (j = 0; j < cols; j++) {
      drawPixelRGB565 (x + j, y + i, ico[i * cols + j]);
    }
  }  
}

mrfaptastic, 2019-08-14:
Sounds good. Check your email.

mrfaptastic, 2019-08-18:
I think you forgot to commit the function body as well. I only see the prototype.

csloz, 2019-08-18:
Its there. -  Line 327

mrfaptastic, 2019-08-18:
Apologies. You did, not sure how I missed that.

As for bounds checking it's not really required as in lines 181-185 of ESP32-RGB64x32MatrixPanel-I2S-DMA.cpp will catch anything dubious.

closed 2019-08-18 by mrfaptastic