mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #20: where the problems is ?

cxandy, 2019-11-21:
[image]

mrfaptastic, 2019-11-21:
Hardware issue. Check your wiring and grounding.

OE pin especially.

Not a software matter.

cxandy, 2019-11-22:
this firmware is work fine. 
https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel

Galaxy-Man, 2020-07-19:
This firmware works great on my ESP32 DoIT, I have managed to add in TimeAlarms and WiFiConnectLite. I cannot find where to say thank you for the support and library. Thanks
Only issue I have is WiFiConnectLite does not save settings to my ESP32.
These are the only changes made to setting so I could use I2c
#define R1 25 
#define G1 26 
#define BL1 27
#define R2 5    // 21 SDA
#define G2 19   // 22 SDL
#define BL2 23 
#define CH_A 12 
#define CH_B 16 
#define CH_C 17 
#define CH_D 18 
#define CH_E -1 // assign to pin if using two panels
#define CLK 15  
#define LAT 32  
#define OE 33

mrfaptastic, 2020-07-19:
@Galaxy-Man thank you for your kind words. I also discovered the WiFiConnectLite issue last night as well, and have implemented a [fix to this](https://github.com/mrfaptastic/WiFiConnectLite/commit/c1d41649f6fa38dd1ab1ef95dc9c5fc95933de83). Please download a new .zip of the library. 

As for these new 'FM6126' - I do not have one of these so cannot test. What is it you needed to connect the RESET pin to? Other than that, just to confirm my understanding, you only needed to change the following pin mappings:
```
#define R2 5 // 21 SDA
#define G2 19 // 22 SDL
```

Galaxy-Man, 2020-07-19:
I changed all the pins, it stated out I wanted to use a TTGO oled ESP32 with battery, Wemos ESP32 and DoIT ESP32 so tried to get the pins to work on all boards with having to change layout between them. I had more issues with the TTGO board so rather than waste time I opted for the DoIT board and the pin layout is what I end up with which worked for me and frees the SDA and SDL pins, I should of stuck with your layout. 

Your libray for  the matrix is impressive, is there anything else you are thingking of adding or improving? 

The reset is Bob's code which you need ot run before initializing your library, it initialise the chipset correctly.

It should be on my github page, if not I'll upload it.

I Will try two FM6126 boards this week, I have them with me but forgot the cable from work, doh!

I have not had so much fun with these matrix baords since my Atari days lol

Galaxy-Man, 2020-07-19:
Tested new code works well, thanks.

mrfaptastic, 2020-07-20:
> Your libray for the matrix is impressive, is there anything else you are thinking of adding or improving?
Thanks again. I can thank the original guy who came up with a working DMA implementation for this. I just wanted to make something that was straightforward and that worked. I can't think of anything further that needs to be done. Perhaps share a link to Bob's code so I can understand it more.

> I have not had so much fun with these matrix boards since my Atari days lol

To be honest, I haven't touched my panel since I wrote the original library out of my own curiosity. However, you've got me interested again. So last night found an old [Gyro](https://www.aliexpress.com/item/32401358878.html) laying around. You can connect it to the two  I2C pins (22 and 21) that this library keeps free, and have a lot of fun. 

https://github.com/mrfaptastic/ESP32-Gyro-Clock

Galaxy-Man, 2020-07-20:
This is not good news, these panels are taking up loads of my time! I like it :-)

Bob's code to follow...

Galaxy-Man, 2020-07-20:
#include <Arduino.h> 
#include <TomThumb.h> // this should be in the adafruit-gfx library
#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>

RGB64x32MatrixPanel_I2S_DMA matrix;

////////////////////////////////////////////////////////////////////
// Reset Panel
// This needs to be near the top of the code
//
// Change these to whatever suits
// recommended settings and patches are by
//
// pinout for ESP38 38pin module
// http://arduinoinfo.mywikis.net/wiki/Esp32#KS0413_keyestudio_ESP32_Core_Board
//

//  R1 | G1
//  B1 | GND
//  R2 | G2
//  B2 | E 
//  A  | B
//  C  | D
//  CLK| LAT
//  OE | GND

#define R1 25  
#define G1 26  
#define BL1 27
#define R2 5    // 21 SDA
#define G2 19   // 22 SDL
#define BL2 23  
#define CH_A 12  
#define CH_B 16 
#define CH_C 17  
#define CH_D 18  
#define CH_E -1 // assign to pin if using two panels
#define CLK 15  
#define LAT 32  
#define OE 33    

void resetPanel()
{
  int MaxLed = 64;

  int C12[16] = {0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1};
  int C13[16] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0};

  pinMode(R1, OUTPUT);
  pinMode(G1, OUTPUT);
  pinMode(BL1, OUTPUT);
  pinMode(R2, OUTPUT);
  pinMode(G2, OUTPUT);
  pinMode(BL2, OUTPUT);
  pinMode(CH_A, OUTPUT);
  pinMode(CH_B, OUTPUT);
  pinMode(CH_C, OUTPUT);
  pinMode(CH_D, OUTPUT);
  pinMode(CH_E, OUTPUT);
  pinMode(CLK, OUTPUT);
  pinMode(LAT, OUTPUT);
  pinMode(OE, OUTPUT);

  // Send Data to control register 11
  digitalWrite(OE, HIGH); // Display reset
  digitalWrite(LAT, LOW);
  digitalWrite(CLK, LOW);
  for (int l = 0; l < MaxLed; l++)
  {
    int y = l % 16;
    digitalWrite(R1, LOW);
    digitalWrite(G1, LOW);
    digitalWrite(BL1, LOW);
    digitalWrite(R2, LOW);
    digitalWrite(G2, LOW);
    digitalWrite(BL2, LOW);
    if (C12[y] == 1)
    {
      digitalWrite(R1, HIGH);
      digitalWrite(G1, HIGH);
      digitalWrite(BL1, HIGH);
      digitalWrite(R2, HIGH);
      digitalWrite(G2, HIGH);
      digitalWrite(BL2, HIGH);
    }
    if (l > MaxLed - 12)
    {
      digitalWrite(LAT, HIGH);
    }
    else
    {
      digitalWrite(LAT, LOW);
    }
    digitalWrite(CLK, HIGH);
    digitalWrite(CLK, LOW);
  }
  digitalWrite(LAT, LOW);
  digitalWrite(CLK, LOW);
  // Send Data to control register 12
  for (int l = 0; l < MaxLed; l++)
  {
    int y = l % 16;
    digitalWrite(R1, LOW);
    digitalWrite(G1, LOW);
    digitalWrite(BL1, LOW);
    digitalWrite(R2, LOW);
    digitalWrite(G2, LOW);
    digitalWrite(BL2, LOW);
    if (C13[y] == 1)
    {
      digitalWrite(R1, HIGH);
      digitalWrite(G1, HIGH);
      digitalWrite(BL1, HIGH);
      digitalWrite(R2, HIGH);
      digitalWrite(G2, HIGH);
      digitalWrite(BL2, HIGH);
    }
    if (l > MaxLed - 13)
    {
      digitalWrite(LAT, HIGH);
    }
    else
    {
      digitalWrite(LAT, LOW);
    }
    digitalWrite(CLK, HIGH);
    digitalWrite(CLK, LOW);
  }
  digitalWrite(LAT, LOW);
  digitalWrite(CLK, LOW);
}
// End of default setup for RGB Matrix 64x32 panel  
///////////////////////////////////////////////////////////////

void setup(){

  resetPanel(); // do this before matrix.begin

  // If you experience ghosting, you will need to reduce the brightness level, not all RGB Matrix
  // Panels are the same - some seem to display ghosting artefacts at lower brightness levels.
  // In the setup() function do something like:

  matrix.setPanelBrightness(10); // SETS THE BRIGHTNESS HERE. 60 OR LOWER IDEAL.

  matrix.begin(R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK); // setup the LED matrix
 // matrix.setFont(&TomThumb.h);
 matrix.setTextColor(matrix.Color(96, 0, 96)); // r,g,b
 matrix.setCursor(1, 15);
 matrix.println("Hello World");

}

void loop(){

}

Galaxy-Man, 2020-07-20:
Let me know how you get on with this...or need a hand! I'm sure you can work it out :-)

mrfaptastic, 2020-07-21:
I'm not going to code anything I can't test directly, so if I get a FM6126 based panel one day I might implement a routine. However, your code seems straight forward enough to re-purpose.

Galaxy-Man, 2020-07-21:
Understood, I was going to make it as a standalone file for neatness but never got round to it.

Got the Morphing Clock working with your library, works well and is neat.

If you need inspiration  for a project then an alarm addin would be good. 

Q. Does your DailTimer library work ok on ESP32, reason for question is when I build it reports TimeLib errors!

mrfaptastic, 2020-07-21:
Not sure what library you're using, but this is the TimeLib library.

https://github.com/PaulStoffregen/Time

Galaxy-Man, 2020-07-21:
the build was referencing a very old TimeLib, sorted now thanks.

mrfaptastic, 2020-07-28:
@cxandy - Can you do me a favour and try the development branch and see if that's improves anything?

closed 2020-08-03 by mrfaptastic