mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #21: updates and other libs

marcmerlin, 2020-01-03:
Hi mrfaptastic, I tried to send you Email but it bounced. If you Email me at marc at merlins.org, happy to take this off a github issue where it might not quite belong ;)
I also Cc Rory who did the 128x64 ESP32 furret with SmartMatrix. I'm not sure
if your lib would offer him more free memory in that resolution
https://github.com/rorosaurus/FurretTotem https://www.youtube.com/watch?v=Q7uGj0VHFyQ
------------
As you know, last year, I ended up concentrating on SmartMatrix as it
supported more panels types than your library, including chaining in
various ways all the way to 128x128 (which actually can work on teensy
3.6, although only barely. ESP32 is limited to 128x64 due to DMA RAM).

I saw you made some updates to your lib, and I was curious if it's still
superior in any way to SmartMatrix, or if it's just a personal project
that you're happy to keep maintaining, because why not? :)

Since last year, I did a crapload of work with RGBPanels.

The most relevant thing is a big collection of multi API backends (I
support Adafruit::GFX, FastLED:NeoMatrix, and LEDMatrix).
I saw you spent time re-porting Aurora demos to GFX independently from me :)
(sorry, I wish we had been able to share and I'd have avoided you the
work of doing it too).

Please have all my demos here:
https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos
to steal all my demos :)

That said, for the non GFX ones, if you're interested in using your
library over SmartMatrix, please look at
https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/blob/master/neomatrix_config.h
and
https://github.com/marcmerlin/Framebuffer_GFX
and
https://github.com/marcmerlin/SmartMatrix_GFX

You'll notice that
https://github.com/marcmerlin/SmartMatrix_GFX/blob/master/SmartMatrix_GFX.cpp
does mostly nothing except handle showptr
The magic happens in
https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/blob/beddfb3278852f64bbf2c74886385a2aed15bbeb/neomatrix_
config.h#L162

if you write something equivalent for your lib, then you get all this
demo code for free, including the really cool TableMarkEstes
screenshots:
http://marc.merlins.org/perso/arduino/2019-04.html#Comparing-FastLED_NeoMatrix-and-SmartMatrix_GFX-with-PixelMatrix-Aurora-and-Table-ME-
Demos
old video demo on low res Neopixel array:
https://www.youtube.com/watch?v=SSllLL5SGCg

Now, the reason I bother you :) is to sell you neomatrix_config and
https://github.com/marcmerlin/Framebuffer_GFX :)

If you use them (by simply using SmartMatrix and SmartMatrix::GFX as is,
or making a small glue driver for your library), once you write code
against it using any of the 3 APIs, I mentioned, you get to run it on a
lot of backends:
- SMARTMATRIX
- SSD1331 (96x64 TFT)
- ST7735_128b128
- ST7735_128b160
- ILI9341
- NeoMatrix (M32B8X3 M16BY16T4 M64BY64 are 3 examples of NEOMATRIX defines)
- FASTLED_SDL is auto defined if you use https://github.com/MarcFork/FastLEDonPc
- ARDUINOONPC is auto defined by https://github.com/marcmerlin/ArduinoOnPc-FastLED-GFX-LEDMatrix

The last 2 are pretty exciting (I just wrote them a few days ago), as they
allow you to run your code on a linux PC for testing, no arduino required.
The 2nd one is even more cool, it allows you to run your arduino code on a
Raspberry Pi and display on much bigger displays up to at least 256x256.
https://community.pixelmatix.com/t/preannounce-smartmatrix-gfx-and-fastled-neomatrix-on-top-of-rpi/579
https://github.com/marcmerlin/ArduinoOnPc-FastLED-GFX-LEDMatrix
https://github.com/marcmerlin/FastLED_RPIRGBPanel_GFX

Best,
Marc

csloz, 2020-01-03:
Slightly OT, but I've tried emailing you before Marc, but no reply.  
Interested in getting it working also.  I have some video of what I'm doing at https://pixelated.cn
Can send you a pcb or two if you are interested (although not till march, as I'm out of china at the moment).

mrfaptastic, 2020-01-03:
Hi @marcmerlin. Thanks for your continued interest in the library. I read through your blog and I'm most impressed in your dedication to LED raving.

So what you want is an implementaiton of Framebuffer_GFX that ultimately calls ESP32-RGB64x32MatrixPanel-I2S-DMA right? 

Cheers.

marcmerlin, 2020-01-04:
@csloz : please try again: marc at merlins.org. I'm usually pretty good with Email but sorry if I dropped yours. I'm happy to discuss code/APIs/etc, but I don't have time to help people write or debug their own code because I have too much of mine that already needs care :) but for SmartMatrix, I do try to help out on the smartmatrix forum (as do others)

@mrfaptastic , so to be clear, I don't really "want" anything :)
The Email when I now pasted here since I couldn't Email it to you, was mostly to compare notes with you one year later.

I was wondering if this lib still had advantages over SmartMatrix, or not?
If we're talking GFX compatibility, I added that (and 2 more APIs, FastLED and LEDMatrix), so that's a non issue.

As for example code, I was giving you a pointer to FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos which should have code that you may enjoy on your matrices.

The rest, is mostly to give you other pointers to what's available, including my multi hardware backend layer (indeed Framebuffer_GFX) which allows you to run your same code on 7 different display methods, without having to change a line of it (not unlike GFX, but while adding 24bit support unlike GFX which is 16bit, and support for code that uses FastLED and LEDMatrix).

If this is not useful to you, or you are happy to port your code to my API layer if and when you want to switch to a different hardware backend (including rPi and now even your own linux PC to code and debug locally), then there is no need for you to do anything.

This was mostly just a FYI which you can use for whatever value it has to you :)
(and you can also close this comment if you wish)

closed 2021-02-22 by mrfaptastic