mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #23: FM6126 based HUB75 Panels

mrfaptastic, 2020-07-21:
Same issue will affect this library, reference: https://github.com/hzeller/rpi-rgb-led-matrix/issues/746

Potential fix required before using: .begin() -> https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/issues/20

```
#include <Arduino.h>
#include <TomThumb.h> // this should be in the adafruit-gfx library
#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>

RGB64x32MatrixPanel_I2S_DMA matrix;

////////////////////////////////////////////////////////////////////
// Reset Panel
// This needs to be near the top of the code
//
// Change these to whatever suits
// recommended settings and patches are by
//
// pinout for ESP38 38pin module
// http://arduinoinfo.mywikis.net/wiki/Esp32#KS0413_keyestudio_ESP32_Core_Board
//

// R1 | G1
// B1 | GND
// R2 | G2
// B2 | E
// A | B
// C | D
// CLK| LAT
// OE | GND

#define R1 25
#define G1 26
#define BL1 27
#define R2 5 // 21 SDA
#define G2 19 // 22 SDL
#define BL2 23
#define CH_A 12
#define CH_B 16
#define CH_C 17
#define CH_D 18
#define CH_E -1 // assign to pin if using two panels
#define CLK 15
#define LAT 32
#define OE 33

void resetPanel()
{
int MaxLed = 64;

int C12[16] = {0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1};
int C13[16] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0};

pinMode(R1, OUTPUT);
pinMode(G1, OUTPUT);
pinMode(BL1, OUTPUT);
pinMode(R2, OUTPUT);
pinMode(G2, OUTPUT);
pinMode(BL2, OUTPUT);
pinMode(CH_A, OUTPUT);
pinMode(CH_B, OUTPUT);
pinMode(CH_C, OUTPUT);
pinMode(CH_D, OUTPUT);
pinMode(CH_E, OUTPUT);
pinMode(CLK, OUTPUT);
pinMode(LAT, OUTPUT);
pinMode(OE, OUTPUT);

// Send Data to control register 11
digitalWrite(OE, HIGH); // Display reset
digitalWrite(LAT, LOW);
digitalWrite(CLK, LOW);
for (int l = 0; l < MaxLed; l++)
{
int y = l % 16;
digitalWrite(R1, LOW);
digitalWrite(G1, LOW);
digitalWrite(BL1, LOW);
digitalWrite(R2, LOW);
digitalWrite(G2, LOW);
digitalWrite(BL2, LOW);
if (C12[y] == 1)
{
digitalWrite(R1, HIGH);
digitalWrite(G1, HIGH);
digitalWrite(BL1, HIGH);
digitalWrite(R2, HIGH);
digitalWrite(G2, HIGH);
digitalWrite(BL2, HIGH);
}
if (l > MaxLed - 12)
{
digitalWrite(LAT, HIGH);
}
else
{
digitalWrite(LAT, LOW);
}
digitalWrite(CLK, HIGH);
digitalWrite(CLK, LOW);
}
digitalWrite(LAT, LOW);
digitalWrite(CLK, LOW);
// Send Data to control register 12
for (int l = 0; l < MaxLed; l++)
{
int y = l % 16;
digitalWrite(R1, LOW);
digitalWrite(G1, LOW);
digitalWrite(BL1, LOW);
digitalWrite(R2, LOW);
digitalWrite(G2, LOW);
digitalWrite(BL2, LOW);
if (C13[y] == 1)
{
digitalWrite(R1, HIGH);
digitalWrite(G1, HIGH);
digitalWrite(BL1, HIGH);
digitalWrite(R2, HIGH);
digitalWrite(G2, HIGH);
digitalWrite(BL2, HIGH);
}
if (l > MaxLed - 13)
{
digitalWrite(LAT, HIGH);
}
else
{
digitalWrite(LAT, LOW);
}
digitalWrite(CLK, HIGH);
digitalWrite(CLK, LOW);
}
digitalWrite(LAT, LOW);
digitalWrite(CLK, LOW);
}
// End of default setup for RGB Matrix 64x32 panel
///////////////////////////////////////////////////////////////

void setup(){

resetPanel(); // do this before matrix.begin

// If you experience ghosting, you will need to reduce the brightness level, not all RGB Matrix
// Panels are the same - some seem to display ghosting artefacts at lower brightness levels.
// In the setup() function do something like:

matrix.setPanelBrightness(10); // SETS THE BRIGHTNESS HERE. 60 OR LOWER IDEAL.

matrix.begin(R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK); // setup the LED matrix
// matrix.setFont(&TomThumb.h);
matrix.setTextColor(matrix.Color(96, 0, 96)); // r,g,b
matrix.setCursor(1, 15);
matrix.println("Hello World");

}

void loop(){

}
```

Galaxy-Man, 2020-07-22:
Well after fettling a few settings and change -1 to 14 I have two panels together 128x32 working. Not got the 64x64 mode working yet, panels not behaving. These are some examples of animated screens for my alarm clock and the Aurora Demo.
[image]
[image]
[image]
[image]
[image]

csloz, 2020-07-22:
The connector on the larger density panels is not called HUB75E for nothing.

You need the E line for > 32px x 32px…

eg
64x64, 128x32

Good to see you got it going.

I had trouble getting > 128 x 32 going (4096pixels), as I ran into memory problems. 
Larger than that becomes iffy.

mrfaptastic, 2020-07-22:
> Well after fettling a few settings and change -1 to 14 I have two panels together 128x32 working. Not got the 64x64 mode working yet, panels not behaving. 

Nice work. What is the second and third visualisation?

> Larger than that becomes iffy.

This is what I am going to change in the library next. Turn off the second DMA buffer which most people don't use and therefore immediately double the pixels.

Secondly, going to investigate reducing pwm gradients to about 16 bit color (most people can't tell the difference) to further increase resolution. These would be compile time settings.

Galaxy-Man, 2020-07-22:
Flowfield and Increment are the two screens. Agree on both parts, I saw no advantage in have the second DMA, however can you write in as an option to have 1 or 2 DMA buffers? 16 bit would be a better option as who can tell difference on a 64x32 screen? these are not exactly high end commercial products! I would not waste my time doing it other than proving I could do it!

Galaxy-Man, 2020-07-22:
As DMA is the constraint? does it stand to reason less color variations the more DMA you have and in return the more panels you could possibly have?

Galaxy-Man, 2020-07-23:
As an update in the void resetPanel(),  MaxLed is set to MATRIX_WIDTH and I have updated the library for support of 128x32 to get it to work. 64x64 does not work, yet ,  or I should say I cannot get it to work :-)

Galaxy-Man, 2020-07-23:
Working currently on mrfaptastic DMA driver, NTP no RTC and DailyAlarms all working together. I need to clean out old library files from VSC to eliminate conflicts. DailyTimers seems to work ok on ESP8266 not ESP32, however I tried your original source of forked version and that works with ESP32!

mrfaptastic, 2020-07-23:
> As an update in the void resetPanel(), MaxLed is set to MATRIX_WIDTH and I have updated the library for support of 128x32 to get it to work. 64x64 does not work, yet , or I should say I cannot get it to work :-)

I have created a 'development' branch of the library. Feel free to submit a pull request to that. 
https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/tree/development

mrfaptastic, 2020-07-24:
> DailyTimers seems to work ok on ESP8266 not ESP32, however I tried your original source of forked version and that works with ESP32!

Just to clarify is it my fork that works or is it the original that works?

Galaxy-Man, 2020-07-24:
Sorry when it's late I sometimes do not make myself clear Yours works on ESP8266 and the original (Bulldog) works on ESP32, I'm using VSC to build. I have not looked into why thats is yet, this was just an inital observation.

Thanks for the developemnt branch, I'll do a pull request this weekend.

Galaxy-Man, 2020-07-24:
Ok sorted it if I change the following it works...

sprintf(timeBuffer, "Time:%02d:%02d:%02d\tDate:%02d/%02d/%4d (mm/dd/yyyy)", hour(), minute(), second(), month(), day(), year());

to

sprintf(timeBuffer, "Time:%2d:%02d:%02d\tDate:%02d/%02d/%4d", hour(), minute(), second(), day(), month(), year());

Galaxy-Man, 2020-07-27:
Panel working with NTPClient, TIme, TimeLib and mrFaptastic DMA library and DailyTimers both excellent. Lesson learned here is make sure you avoid global libraries where possible, pull them into each project. Opinion VSC is far better than Arduino IDE. I had to play hide and seek with the Time/TimeLib using wrong versions etc but got there in the end.  

Working on adding audio and animation next.

mrfaptastic, 2020-07-27:
I moved from PlatformIO (VS Code) about a year ago and will never move back. It's so much better. I wish I knew about VSC when I was working on the original ESP32-RGB64x32MatrixPanel-I2S-DMA library, would have made debugging so much easier. Arduino IDE is painful for anything other than creating a 'blink' sketch.

Galaxy-Man, 2020-07-27:
Have you bought yourself one of these yet?

https://www.youtube.com/watch?v=psMqilqlrRQ&feature=youtu.be

mrfaptastic, 2020-07-28:
> 
> 
> Have you bought yourself one of these yet?
> 
> https://www.youtube.com/watch?v=psMqilqlrRQ&feature=youtu.be

I haven't but I might one day.

On another note, I have submitted a large change to the library to improve performance and reduce memory consumption significantly (disable double buffering). Also the possibility to use for much much longer chains of panels is possible, but as I don't have any more than one 32x64 panel. I can't confirm. Refer to the readme.

https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/tree/development

mrfaptastic, 2020-07-28:
> I had trouble getting > 128 x 32 going (4096pixels), as I ran into memory problems. Larger than that becomes iffy.

@csloz - Can you try again with something larger than that? Use the development branch and enable (uncomment) line `#define SPLIT_MEMORY_MODE 1`. I'm curious to know how far we can now push the ESP32.

csloz, 2020-07-29:
Unfortunately my equipment is back home in China, and I’m not. - will be a while due to covid lockdown, so someone else will need to test in the interim…

Galaxy-Man, 2020-07-29:
Hi mrfaptastic, thanks for the replies and your time doing the updates.

I'll order more panels today, hopefully I can find the correct panel on eBay or Amazon, when it comes to tech details the suppliers are not vey good at telling you what you are buying as all they do is box shift!

However my usual supplier is out of stock since March, I wonder why! I found someone else so will order a single 64x32 panel and if correct I'll order more.

How big (many) do you wish to go with the panels? I may need a bigger PSU lol

mrfaptastic which country are you in? if these are correct panels I don't mind sending you some in the post, it's a small price to pay for the effort you are putting in to this project.

Galaxy-Man, 2020-07-29:
404 on the development link, it was there before the weekend but has dropped off.

Galaxy-Man, 2020-07-29:
I am ordering extra panels today from within my country, will give it a go on arrival hopefully this week.

Galaxy-Man, 2020-07-29:
mrfaptastic, is thee a particular order for loading the includes after <Arduino.h> ?

Reason for question is other panel drivers had issues when loading <Wifi.h> after other developers panel drivers, locking up, flicker, wifi not working correctly very unstable.

The order of my code, reference for other developers is ..

#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
////////////////////////////////////////////////////////////////////
 
#include "WiFiConnect.h"
#include <WiFiClient.h>
#include <HTTPClient.h>
 
WiFiConnect wc;

WiFiClient client;
HTTPClient http;
 
//const char *ssid = "EE5G";
//const char *ssid = "testreserved";
//const char *password = "Pa55w0rd!6101";
//const char *ssid = "SSID";
//const char *password = "PASSWORD";
/////////////////////////////////////////////
////////////////////////////////////////////////////////////////////
#include "NTPClient.h"
#include <WiFiUdp.h>
#include "Time.h"
#include "DailyTimer.h"
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP);
/////////////////////////////////////////////
////////////////////////////////////////////////////////////////////
#include "ESP32-RGB64x32MatrixPanel-I2S-DMA.h"

RGB64x32MatrixPanel_I2S_DMA matrix;
/////////////////////////////////////////////
////////////////////////////////////////////////////////////////////
#include "GFX_fonts/Font4x7Fixed.h"
/////////////////////////////////////////////

Galaxy-Man, 2020-07-29:
In void setup()

//////////////////
resetPanel();                                                                       // do this before matrix.begin
// matrix.begin(); // use default settings in library
matrix.begin(R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK); // setup the LED matrix

startWiFi();

startTimeClient();

mrfaptastic, 2020-07-29:
> However my usual supplier is out of stock since March, I wonder why! I found someone else so will order a single 64x32 panel and if correct I'll order more.
>
> How big (many) do you wish to go with [...]

Hold on! You're putting me under pressure here! I just did some testing of the library (as one doesn't need to physically have panels connected to see if the library theoretically works), and managed to get this to work, if I reduced to colour depth to about 5 bit (65k) and enabled SPLIT_MEMORY_MODE. However there is no guarantee this will look any good in the real world as the refresh rate is pretty low and it may be super dim as a result. At least the library doesn't just die.

```
#ifndef MATRIX_HEIGHT
#define MATRIX_HEIGHT               128
#endif

#ifndef MATRIX_WIDTH
#define MATRIX_WIDTH                512
#endif

#ifndef MATRIX_ROWS_IN_PARALLEL
#define MATRIX_ROWS_IN_PARALLEL     2
#endif

#define MATRIX_COLOR_DEPTH 			    5     
```
 
> 404 on the development link, it was there before the weekend but has dropped off.

Sorry. It's back again now.

> mrfaptastic, is there a particular order for loading the includes after <Arduino.h> ?

None whatsoever. Are you referring to the examples? I probably need to clean these up if I have the includes in the wrong order.

Galaxy-Man, 2020-07-29:
Sorry there is no pressure, I thought I was helping. Even if I send you some panels they are free of charge I am not expecting anything from you. I know what its like doing these projects, people expect too much in return. This is onyl a hobby for me, a leaning experience nothing more. I do these things at my own pace and as long as I am enjoying myself then I continue to program and do electronics.

mrfaptastic, 2020-07-29:
Apoloiges. It was a humour fail on my behalf. I just didn't want you going out buying 50 panels with the library only being able to handle a couple haha!

Looks like 256x32 or even 256x64 should be doable however! YYMV of course!

Galaxy-Man, 2020-07-29:
Reference to start order is to do with something from another site saying ESP32 need to load Wifi drivers first as thye allocate memory in a particular way. Since then I have always loaded them first. I had a lot of issues with reset of ESP32 and wifi. This could be the way the other matrix drivers were written?  

All other panel drivers need the reset program to work or they did last time I tested them.

Galaxy-Man, 2020-07-29:
Thoughts;

If the panels are dumb using shift registers then the limitation is on how quickly and brightly we can light the LED to get POV?
There will be a compromise fps, to get more panels and as you said may be dimmer?

What if we find the compromise then use muliple ESP32 to sync each other to drive many panels?

Galaxy-Man, 2020-07-29:
Just ordered 2 panels which means I have 4 for testing if they work I'll send a couple to you.

I'll ask privately for your address once I confirm they are the correct type FM6126 based.

These are the panels I have ordered.

https://www.ebay.com/itm/LED-Matrix-P5-RGB-pixel-panel-64x32-LED-sign-screen-module/274396077566?_trkparms=aid%3D888008%26algo%3DDISC.CARDS%26ao%3D1%26asc%3D20140109134348%26meid%3Df35d358fb1fc414e824773b0a19e447d%26pid%3D100042%26rk%3D1%26rkt%3D1%26sd%3D274372793522%26itm%3D274396077566%26pmt%3D0%26noa%3D1%26pg%3D2508447%26algv%3DDiscV2%26brand%3DUnbranded&_trksid=p2508447.c100042.m2062

mrfaptastic, 2020-07-29:
> I had a lot of issues with reset of ESP32 and wifi. This could be the way the other matrix drivers were written?

I haven't had any issues with WiFi with my library, but with other Matrix libraries, they don't use DMA, and simply use a timer interrupt every millisecond to refresh the panels. This generally is what causes the ESP32 to crash and burn with hundreds of interrupts/second.

> If the panels are dumb using shift registers then the limitation is on how quickly and brightly we can light the LED to get POV?
> There will be a compromise fps, to get more panels and as you said [...]

You could have an infinite size array of panels if you have a cluster of ESP's driving say a single 256x64 portion.

If you're displaying a static image it would be easier to sync (just crop the image up or something). If you wanted to display video or something... probably impossible.

I think if you wanted to go down this interesting route of multiple ESP32 driving one massive panel of sub-panels. You'd have 1) One 'master' ESP sending packets of bitmap data to all of them (via UDP or whatever), which gets stored on each sub-panel's buffer.
2) A confirmation back from each ESP32 they've got all the packets
3) Then the master sends an 'OK, switch buffer!' message.

Frames per second would be limited to the WiFi throughput... would be super slow.

mrfaptastic, 2020-07-29:
> I'll ask privately for your address once I confirm they are the correct type FM6126 based.

Wow! Amazing!

Galaxy-Man, 2020-07-29:
It will be amazing if they are the correct panels lol

Galaxy-Man, 2020-07-31:
New Dev version up and running and working with FM6126 panels. Is there any particular test you need me to do or shall I just run some of my code through it?

Galaxy-Man, 2020-08-03:
Two new panels arrived today, I'll try and get them connected tonight in a 2x2 matrix. All going well I will send these to you. Can you email me your address galaxy-man at msdigital.co.uk

mrfaptastic, 2020-08-03:
Thank you @Galaxy-Man. Incredibly kind of you. I've sent you an email, check your spam folder.

Galaxy-Man, 2020-08-03:
Hi mrfaptastic, you get everything in the photo, except the Mr.Kipplings cakes sorry, lol

[image]

Galaxy-Man, 2020-08-03:
Ok done a quick test and the panels are Chipone icnd2038s based (no reset required) not the FM6126 (reset required). Not to worry they still work. 

The main test we both were trying to do was multi panel? so who cares what chipset :-) 

[icn2038s.pdf](https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/files/5017158/icn2038s.pdf)

mrfaptastic, 2020-08-03:
This is incredible! Thank you!

Yeah, I wish to expand on the multi-panel capabilities, as I made quite a few changes over the weekend to extract more out of the ESP32 SoC's DMA, and if I can get chaining working in random configurations with four panels, then it'll be super cool.

As for the reset code, I'll bake in what you wrote as an experimental function should people want to use it.

Galaxy-Man, 2020-08-03:
Two different panels, good colour match no issues.

[image]

Galaxy-Man, 2020-08-03:
Correction, there is a slight colour difference between panels. Not noticable until I did fillscreen 0xFFFF. The two panels I'm sending you are matched, so colours should be ok.

Galaxy-Man, 2020-08-04:
Panels being dispatched today.

Galaxy-Man, 2020-08-04:
Interesting use of the I2S DMA and Matrix Panels

https://github.com/yetifrisstlama/espirgbani_pio

Galaxy-Man, 2020-08-04:
Downloaded latest version of DMA https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA
I need to order more panels to test the other options.

Cannot find anyone ebay or Amazon UK with FM6126 panels, if you find any I will order more.

Galaxy-Man, 2020-08-04:
FM6126A dataSheet inc German but readable. Use Google to translate, it does a reasonable job of it.
Will continue to look for the english version. 

[FM6126A.de.pdf](https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/files/5022793/FM6126A.de.pdf)

Galaxy-Man, 2020-08-17:
HI mrfaptastic

Are you in tomorrow as I have another package being sent today.

2 x panels to match the previous set.

I thought the 2x2 same size would look better than having odd sizes! Also becuaes of the time you spent on this porject this is a small price to pay. ex gratia :-)

Have Fun

Galaxy-man

Galaxy-Man, 2020-08-17:
I have a full set of 4x FM 6126 panels arriving thsi week I will test with the reset code and your library, I am sure it will be ok.

I will let you know how I get on.

Galaxy-Man, 2020-08-17:
Just I sent one package another arrived! got my new panels, watch ths spot :-)

[image]

mrfaptastic, 2020-08-17:
> 
> 
> I thought the 2x2 same size would look better than having odd sizes! Also becuaes of the time you spent on this porject this is a small price to pay. ex gratia :-)

Wow that's amazing! Thank you. Indeed I will be around. 

> I will let you know how I get on.

Please do, even if it works OK.

Galaxy-Man, 2020-08-17:
mrfaptastic no problem, you are welcome. Please let me know when you have recieved them.

atc1441, 2020-08-20:
Hey so since the FM6126 did not started correctly after a software reset it was needed to Reset the I2S hardware first on a software reset and then send the FM6126 enable cmds. 

This is a function i use to reset the I2S: 

void i2s_reset(i2s_dev_t *dev) {
  dev->out_link.start = 0;
  dev->conf.tx_start = 0;
  dev->lc_conf.in_rst = 1; dev->lc_conf.out_rst = 1; dev->lc_conf.ahbm_rst = 1; dev->lc_conf.ahbm_fifo_rst = 1;
  dev->lc_conf.in_rst = 0; dev->lc_conf.out_rst = 0; dev->lc_conf.ahbm_rst = 0; dev->lc_conf.ahbm_fifo_rst = 0;
  dev->conf.tx_reset = 1; dev->conf.tx_fifo_reset = 1; dev->conf.rx_fifo_reset = 1;
  dev->conf.tx_reset = 0; dev->conf.tx_fifo_reset = 0; dev->conf.rx_fifo_reset = 0;
}

Added this into the https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/blob/master/esp32_i2s_parallel.c file so it has all the dependences already. and also to the .h then it can be called from Arduino when needed.

Also the Enable function for the FM6126 on the top of this issue is quite long this on is a bit shorter, but not really better:
int MaxLed = 256;

#define P_LAT 22
#define P_A 19
#define P_B 23
#define P_C 18
#define P_D 5
#define P_E 15
#define P_OE 2
#define CLK 14
#define R1 13
#define R2 27
#define G1 21
#define G2 17
#define B1 12
#define B2 4

void set_RGB(bool state) {
  digitalWrite (R1, state);
  digitalWrite (G1, state);
  digitalWrite (B1, state);
  digitalWrite (R2, state);
  digitalWrite (G2, state);
  digitalWrite (B2, state);
}
void init_matrix() {
  bool C12[16] = {1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0};
  bool C13[16] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0};

  pinMode(CLK, OUTPUT);
  pinMode(P_LAT, OUTPUT);
  pinMode(P_OE, OUTPUT);
  pinMode(B1, OUTPUT);
  pinMode(R1, OUTPUT);
  pinMode(R2, OUTPUT);
  pinMode(G1, OUTPUT);
  pinMode(G2, OUTPUT);
  pinMode(B2, OUTPUT);
  pinMode(P_A, OUTPUT);
  pinMode(P_B, OUTPUT);
  pinMode(P_C, OUTPUT);
  pinMode(P_D, OUTPUT);
  pinMode(P_E, OUTPUT);

  digitalWrite (P_OE, HIGH);
  digitalWrite (P_LAT, LOW);
  digitalWrite (CLK, LOW);

  // Send Data to control register 11
  for (int l = 0; l < MaxLed; l++) {
    int y = l % 16;
    set_RGB(LOW);
    if (C12[y]) set_RGB(HIGH);
    digitalWrite(P_LAT, (l > MaxLed - 12) ? HIGH : LOW);
    digitalWrite(CLK, HIGH);
    digitalWrite(CLK, LOW);
  }
  digitalWrite (P_LAT, LOW);
  digitalWrite (CLK, LOW);

  // Send Data to control register 12
  for (int l = 0; l < MaxLed; l++) {
    int y = l % 16;
    set_RGB(LOW);
    if (C13[y])set_RGB(HIGH);
    digitalWrite(P_LAT, (l > MaxLed - 13) ? HIGH : LOW);
    digitalWrite(CLK, HIGH);
    digitalWrite(CLK, LOW);
  }
  digitalWrite (P_LAT, LOW);
  digitalWrite (CLK, LOW);
}

mrfaptastic, 2020-08-20:
Hi @atc1441. Great youtube channel by the way.

Would you be happy to raise a pull request? I'll incorporate then.

Galaxy-Man, 2020-09-14:
mrfaptastic after lots of waiting the 6x led matrix panels are on my desk :-) I will dispatch them to you tomorrow . Are you goiong to be at home Wednesday?

Chipset in these are the same as the first batch I sent to you. CHIPONE ICND2038S, data sheet is on my repository.

[image]
[image]

mrfaptastic, 2020-09-14:
Hi @Galaxy-Man. Yep, delivery is fine on Wednesday. 

This is going to be immense. Can't wait to chain these up and test. Thank you !!!

Galaxy-Man, 2020-09-15:
Morning mrfaptastic, panels being collected today delivery tomorrow by 1pm, need to be signed for.  

Have fun, let me know you have recieved them and how you get on.  :-)

Are we going bigger than 12x panels or is this the limit before performance/colour/brightness/fps is compromised?

mrfaptastic, 2020-09-15:
This is phenomenal. 

> Are we going bigger than 12x panels or is this the limit before performance/colour/brightness/fps is compromised?

That I don't know until I experiment. There is all manner of tweaks that can be done to the library to get this to work, all which will have a trade off between brightness, refresh rate and color depth. I will let you know how I get on. Either way, my poor ESP32 is going to get put through its paces.

mrfaptastic, 2020-09-16:
@Galaxy-Man. All received! Thank you kindly, will start testing. Did you end up receiving any FM6126 based panels in the end? I think you did that they worked fine when chained with non-FM6126 panels?

Note: Cleaning up this issue to remove our messages to each other. Nothing personal :-)

Galaxy-Man, 2020-09-16:
Hi mrfaptastic

Glad you have the panels.

I have ordered more Fm panels, not holding my breath for delivery.

I chained a mix of panels and they worked fine, only minor issue is a very very slight colour difference when showing single colour across panels.

pilly52, 2020-10-01:
@Galaxy-Man  How did you finally achieve to chain FM6126 panels ?  Using a single 64x32 panel (same ref as yours from amazon) works perfectly, but if I try to simply chain them, I'm getting the same image on both matrices. 

Here is the code I'm using :

`// How to use this library with a FM6126 panel, thanks goes to:
// https://github.com/hzeller/rpi-rgb-led-matrix/issues/746

#include <Arduino.h>
#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>

RGB64x32MatrixPanel_I2S_DMA matrix;

////////////////////////////////////////////////////////////////////
// Reset Panel
// This needs to be near the top of the code
//
// Change these to whatever suits
// recommended settings and patches are by
//
// pinout for ESP38 38pin module
// http://arduinoinfo.mywikis.net/wiki/Esp32#KS0413_keyestudio_ESP32_Core_Board
//

// R1 | G1
// B1 | GND
// R2 | G2
// B2 | E
// A | B
// C | D
// CLK| LAT
// OE | GND

#define R1 25
#define G1 26
#define BL1 27
#define R2 5 // 21 SDA
#define G2 19 // 22 SDL
#define BL2 23
#define CH_A 12
#define CH_B 16
#define CH_C 17
#define CH_D 18
#define CH_E 14 // assign to pin if using two panels
#define CLK 2
#define LAT 32
#define OE 33

/////////////////////////////////////////////////////////////////
// how many pixels wide if you chain panels
// 4 panels of 64x32 is 256 wide.
int MaxLed = 128;

int C12[16] = {0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1};
int C13[16] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0};

void resetPanel()
{
  pinMode(CLK, OUTPUT);
  pinMode(LAT, OUTPUT);
  pinMode(OE, OUTPUT);
  pinMode(R1, OUTPUT);
  pinMode(G1, OUTPUT);
  pinMode(BL1, OUTPUT);
  pinMode(R2, OUTPUT);
  pinMode(G2, OUTPUT);
  pinMode(BL2, OUTPUT);
  pinMode(CH_A, OUTPUT);
  pinMode(CH_B, OUTPUT);
  pinMode(CH_C, OUTPUT);
  pinMode(CH_D, OUTPUT);
  pinMode(CH_E, OUTPUT);

  // Send Data to control register 11
  digitalWrite(OE, HIGH); // Display reset
  digitalWrite(LAT, LOW);
  digitalWrite(CLK, LOW);
  for (int l = 0; l < MaxLed; l++)
  {
    int y = l % 16;
    digitalWrite(R1, LOW);
    digitalWrite(G1, LOW);
    digitalWrite(BL1, LOW);
    digitalWrite(R2, LOW);
    digitalWrite(G2, LOW);
    digitalWrite(BL2, LOW);
    if (C12[y] == 1)
    {
      digitalWrite(R1, HIGH);
      digitalWrite(G1, HIGH);
      digitalWrite(BL1, HIGH);
      digitalWrite(R2, HIGH);
      digitalWrite(G2, HIGH);
      digitalWrite(BL2, HIGH);
    }
    if (l > MaxLed - 12)
    {
      digitalWrite(LAT, HIGH);
    }
    else
    {
      digitalWrite(LAT, LOW);
    }
    digitalWrite(CLK, HIGH);
    digitalWrite(CLK, LOW);
  }
  digitalWrite(LAT, LOW);
  digitalWrite(CLK, LOW);
  // Send Data to control register 12
  for (int l = 0; l < MaxLed; l++)
  {
    int y = l % 16;
    digitalWrite(R1, LOW);
    digitalWrite(G1, LOW);
    digitalWrite(BL1, LOW);
    digitalWrite(R2, LOW);
    digitalWrite(G2, LOW);
    digitalWrite(BL2, LOW);
    if (C13[y] == 1)
    {
      digitalWrite(R1, HIGH);
      digitalWrite(G1, HIGH);
      digitalWrite(BL1, HIGH);
      digitalWrite(R2, HIGH);
      digitalWrite(G2, HIGH);
      digitalWrite(BL2, HIGH);
    }
    if (l > MaxLed - 13)
    {
      digitalWrite(LAT, HIGH);
    }
    else
    {
      digitalWrite(LAT, LOW);
    }
    digitalWrite(CLK, HIGH);
    digitalWrite(CLK, LOW);
  }
  digitalWrite(LAT, LOW);
  digitalWrite(CLK, LOW);
}

// End of default setup for RGB Matrix 64x32 panel
/////////////////////////////////////////////////////////////////

void setup(){

resetPanel(); // do this before matrix.begin

// If you experience ghosting, you will need to reduce the brightness level, not all RGB Matrix
// Panels are the same - some seem to display ghosting artefacts at lower brightness levels.
// In the setup() function do something like:
// If using multi panels watch your power levels when running full brightness.

matrix.setPanelBrightness(10); // SETS THE BRIGHTNESS HERE. 60 OR LOWER IDEAL.
matrix.begin(R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK); // setup the LED matrix
matrix.setTextColor(matrix.color444(96, 0, 96)); // r,g,b
matrix.setCursor(1, 15);
matrix.println("Hello World");

}

void loop(){

}`

pilly52, 2020-10-01:
I added it in the code with :

#define MATRIX_WIDTH 128
#define MATRIX_HEIGHT 32

I haven't modified the .h, but looking at the H if it's already defined it shouldn't overwrite it.

mrfaptastic, 2020-10-01:
Can you hack the .h directly and set the width to 128 and see if it makes a difference? I have noticed some compilers don't respect defines not in the .h file itself.

pilly52, 2020-10-01:
That was it :) Thx for the hint !

[3 more replies]

closed 2020-11-28 by mrfaptastic