mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #26: Example of using multiple panels in a non standard layout

witnessmenow, 2020-07-29:
Hello,

First, thanks a lot of the library, it is working great! Much appreciated.

I have made a shield for using this library with the D1 mini style ESP32 board (if you want one @mrfaptastic , I would be more than happy to donate one) and someone wanted to arrange their panels vertically rather than horizontally and I came up with this fairly hacky, but very simple solution that I wanted to share.

https://github.com/witnessmenow/ESP32-i2s-Matrix-Shield/tree/master/examples/MultiplePanelVirtualScreenPatternPlasma

I'm creating a virtual display using AdafruitGFX where the user configures how their panels are laid out and it will look after the mapping of the pixels for them. A reference to this library is passed into the virtual displays constructor. I've implemented the same virtual calls as this library so you should be able to just swap out the matrix calls with the virtual display.

It should be flexible enough to be used with panels arranged in rows and columns too. There is possibly a better way of doing this, but this was one idea I had that I thought would be really simple for end users to use.

I noticed earlier that you updated to GFX, I haven't got a chance to try that out instead with the virtual display, its still using AdafruitGFX.

The example linked is patternPlasma, but I also tried it out on Aurora and it worked fine too. (I only have two panels though)

Just as a BTW, Aurora was running out of memory for me originally, but I noticed in #23 that you had mentioned that you updated some things to free up some, so I updated to version from about 8 hours ago (I see you have updated since!) and now it's working perfectly.

mrfaptastic, 2020-07-30:
Hi Brian! Firstly, you're the legend that got me interested in these panels with your fantastic library for the ESP8266 years ago!  So thank you for the kind words.

> I'm creating a virtual display using AdafruitGFX where the user configures how their panels are laid out and it will look after the mapping of the pixels for them. A reference to this library is [...]

This is fantastic, and is something that has often come up as a request. I'll look into your implementation and merge it into the overall library.

> I noticed earlier that you updated to GFX, I haven't got a chance to try that out instead with the virtual display, its still using AdafruitGFX.

I did, but as of a few minutes ago I've changed it back to Adafruit_GFX. GFX_Root didn't in the end provide any memory/program space benefits and I'd rather not break backwards compatability when people update to the new version of this library. Especially given you're using this library for your cool project.

> Just as a BTW, Aurora was running out of memory for me originally, but I noticed in #23 that you had mentioned that you updated some things to free up some, so I updated to version from about 8 hours ago (I see you have updated since!) and now it's working perfectly.

That's great to hear. You should be able to get it working on 4-6 panels even, but who know. I only have a sole 32x64 to test with.

> (if you want one @mrfaptastic , I would be more than happy to donate one)

That would be absolutely amazing if you were happy to do that? Absolutely no need to however. Can test the library out on this as well!

witnessmenow, 2020-07-30:
I'm not sure I can take credit for leading you down this dark, yet brightly illuminated, path! I didn't write PxMatrix, and only contributed very minor code changes to it! I did however use it in a lot of projects! On of the most common complaints I had from people using it was the second ribbon cable and a buddy mentioned this library to me, but I only really started using it recently.

If you'd like to merge it in that would be great. I have some instructions on top of the ino, but I think it might be better to describe it in more detail in a readme or something, with some high quality graphics, such as this beauty 😄 

[image]

Re: Adafruit_GFX ah ok, good to know. I was actually interested in what the advantage between the two was!

Well just to mention it did work with Aurora demo out of the box before, just with my virtual screen thing it didn't on the old version of the lib. 

That sounds great, I'd love to test it out with more panels! I have a friend who has a few of those pixel purses panels that she is looking to make into a larger screen, they are fairly low res though (32x16) but she has tried 3 so far with no problems, and she mentioned that she will try it with more when she gets time.

Re: board, no worries at all. Very happy to support the efforts of this library! Do you have the right ESP32 for it? If not I should have a spare one here I can throw in too. I'll send you an email 

P.S. 
I'm not sure if you ever seen this video before: https://www.youtube.com/watch?v=HKWDGangWU0

witnessmenow, 2020-07-30:
Actually, can you email me, I can't find your email! I didn't realise you didn't have to populate it in the library.properties file !

brian.d.lough@ gmail com

mrfaptastic, 2020-07-30:
Thank you! Email sent earlier today.

mrfaptastic, 2020-08-03:
I have created a VirtualDisplay class example, and a graphic.

https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/tree/master/examples/ChainedPanels

Although, the 2x2 pixel mapping code hasn't actually been written, especially with the rotated panels! Will get to that when I kindly recieve test panels from galaxy man.

Galaxy-Man, 2020-08-04:
Hi mrfaptastic, are you in on Wednesday as the package will need to be signed for.

witnessmenow, 2020-08-07:
I love the diagrams you added @mrfaptastic , but I think there is mistake with the 2x2 image

[image]

I don't think the bottom rows should be upside down

I think it should be:
[ 2 ][ 1 ]
[ 4 ][ 3 ]

It's possible not a bad idea to have the option of invert the second row, as it makes it easier to arrange (it would need a long ribbon cable from 2 to 3 in my config

mrfaptastic, 2020-08-07:
I've nearly finished code to allow S chaining only. As it is the only
practical way to do so with the supplied ribbon cables. Means every second
row has 180 degree rotated panels.

Galaxy-Man, 2020-08-07:
mrfaptastic, thanks for the update. Sorry not replied earlier, doing covid stuff.

witnessmenow, 2020-08-07:
I think it makes sense to make it optional, the ribbon cables that come with the screens are pretty short but longer ones are easy to get. But i'll leave it up to you!

as mentioned I can't test it but here is what I think the code should be (I literally have not checked this works though)

Adding `invertEvenRows` as a constructor param, can default it to true or false, or even remove this alltogether.
```
bool invertEven = false;

    RGB64x32MatrixPanel_I2S_DMA *display;

    VirtualGFX(RGB64x32MatrixPanel_I2S_DMA &disp, int vRows, int vCols, int screenX, int screenY, int vResX, int vResY, bool invertEvenRows = false)
      : Adafruit_GFX(vResX, vResY)
    {
      this->display = &disp;
      rows = vRows;
      cols = vCols;
      screenResX = screenX;
      screenResY = screenY;

      virtualResX = vResX;
      virtualResY = vResY;

      invertEven = invertEvenRows;
    }
```

And then updating the `getCoords` function. Just remove the `invertEven` flag if you don't want to make it optional.

```
inline VirtualCoords VirtualGFX::getCoords(int16_t x, int16_t y) {
  uint8_t row = (y / screenResY) + 1; //a non indexed 0 row number
  if(invertEven && (row % 2 == 0))
  {
    // First portion gets you to the correct offset for the row you need
    // Second portion inverts the x on the row
    coords.x = (y / screenResY) * (cols * screenResX) + (virtualResX - 1 - x);

    // inverts the y the row
    coords.y = screenResY - 1 - (y % screenResY);
  } else {
    coords.x = x + (y / screenResY) * (cols * screenResX);;
    coords.y = y % screenResY;
  }
  return coords;
}
```

witnessmenow, 2020-08-07:
Looking at the graphics agian, even with the serpentine, the still look wrong to me.

Screen 4 has the origin, so 1 and 2 should be upside down.
[image]

The vertical one is also incorrect, display 4 should be on the top. Also every second display should be inverted if you want to stick with inverting the even rows

Galaxy-Man, 2020-08-07:
mrfaptastic, I will be ordering more panels next week, let me know if you'd like another 2 for trying the 2x2.

witnessmenow, 2020-08-07:
Actually I can test it, I was thinking too big!

Here are two vertical panels, with the second one inverted

[image]

[image]

Galaxy-Man, 2020-08-07:
Ive looked at it and layed out the panels, too me it looks correct. When I put two panels together the 0rigin shifted to second panels qed the diagram looks correct for 4.

Galaxy-Man, 2020-08-07:
Good job on 64x64

mrfaptastic, 2020-08-09:
Alright. I think I can close this issue. Thanks @witnessmenow for the code, and huge thanks to @Galaxy-Man for the 2xP3's! 🥇
So I have three panels of different sizes to test the logic, which made it possible.

Simplified the diagram afer your feedbac as people won't really care about how it works to be honest.

[image]
[image]

[Commit](https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/commit/3ff7155bf0bc56fdc0e133e7887468766ebaadbb)

mrfaptastic, 2020-08-09:
I think where a bit of confusion came from is the fact I prefer to chain from the top-right down, and in your pattern plasma test you're chaining upwards.

So I've made it such the library can handle s-chaining as well as top-down or bottom-up (the latter being the default).

mrfaptastic, 2020-08-09:
> mrfaptastic, I will be ordering more panels next week, let me know if you'd like another 2 for trying the 2x2.

Very kind of you yet again, but not sure if it's required now. You don't happen to have a spare HUB75 ribbon cable though - I could do with one of those things.

witnessmenow, 2020-08-09:
I have plenty of them if you need them. I only sent the package Thursday so
it will probably be with you early this week. There is one ribbon cable in
it , but if you need more just let me know.

Galaxy-Man, 2020-08-10:
Hi mrfaptastic, looks like good progress on the panels. Once you have Brian's package if you still need cables just message me and I'll send more.

mrfaptastic, 2020-08-10:
> I have plenty of them if you need them. I only sent the package Thursday so it will probably be with you early this week. There is one ribbon cable in it , but if you need more just let me know.

Ah yes. Sorry I forgot about this. Thank you so much! Looking forward to receiving it. Thanks again.

> Hi mrfaptastic, looks like good progress on the panels.

If you end up buying more panels (i.e. 2x2) hook them up per the diagrams and test away. Should work fine with the library now. One could sell  a  cheap commercial LED panel signage product powered by just ESP32 with this.

Galaxy-Man, 2020-08-10:
Was thinking along same lines, stupidly high fps or 24 gazillion colours may not be needed as won't be playing video etc. However for an info panel this is perfect. Do you think 2x2 will be the limit?

mrfaptastic, 2020-08-10:
> Less is more, do you think 2x2 will be the limit?

Entirely dependant on colour depth. 2x3 should be easily doable I think. 2x4 with a lower color depth. Trial and error really to see what works / looks best!

mrfaptastic, 2020-08-10:
[image]

Thanks for the kind donation Brian [@witnessmenow ] of ESP32 and your new product. Looking forward to testing with this. As for the chaining, your cable has proven all is working now (ignore the fact I have different PX boards).

mrfaptastic, 2020-08-10:
[image]

A quick mod to the new [Animated GIF](https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/tree/master/examples/AnimatedGIFPanel) example to use the VirtualDisplay class shows no issues.

The pause in the video display is due to the GIF of neo flying, looping. Could easy show 30fps otherwise.

mrfaptastic, 2020-08-11:
Included in version 1.2.0 of the library + other improvements.

mrfaptastic, 2020-08-24:
Hi @Galaxy-Man. I thought I would respond to this quote in this issue:
 
> If you are thinking of going bigger let me know and Ill get you some more panels sent over. Im currently using FM6124/FM6126 and seems to be fine.

I chained all 6 panels I had up today (thanks again for the four), and it's working ridiculously well!
 
[image]

I'm going to run out of 5v power before memory.

Galaxy-Man, 2020-08-24:
Hey wow that’s good news, 6! So what is the limit? How much more power do you need?

[Image removed by sender. itworks]<https://user-images.githubusercontent.com/12006953/91067827-4ee78780-e62b-11ea-84ed-7d7bed60118a.jpg>

mrfaptastic, 2020-08-24:
I think about 12-16 64x32 panels would be the memory limit, if there's no other memory consuming stuff happening in the sketch (like a webserver etc.).

> How much more power do you need?

It's a self-induced issue. Basically I need to get more poundland power packs. As a ghetto solution to not having a bench power supply I have each second panel connected to a rechargeable USB power pack via  micro-usb plug soldered to the back of the panel, with a common ground running across all boards. Does the job for a short period for testing.

atc1441, 2020-08-24:
For me the maximum was 4 panels 64x64 with 8bpp after that there wasnt enough memory.
https://twitter.com/atc1441/status/1176855172204650496

mrfaptastic, 2020-08-24:
> You could simply try to init it with more panels and see when it stops working to get the max. in will just be not visible, have you tried that?

Yep. I just tried it with 7 x 64*64 'panels' and it works. 8 onwards it crashed.

12 x 64x32 panels works fine as well.

Also, the setPanelBrightness is actually any value from 0 to MATRIX_WIDTH, so in your demo, if you want to increase the brightness heaps, then setPanelBrightness to about 200 or something

mrfaptastic, 2020-08-24:
The problem I think is also dependant on the expressif IDF version etc. used. even a totally blank sketch with just setup() and loop() seems to have a fragmented memory map between versions.

Are you using the latest library from the git? I made some changes since the custom version you created for one of your projects.

atc1441, 2020-08-24:
That testing was quite old so it could be indeed be that its now working better/with more like with your tests. 

It is so dark in that example as the FM6126 Chips where new to me, now its a good brightness and can be set perfectly fine with setPanelBrightness

So these where quite old infos, sorry for that.

Will try to update my version in the project in the next days. 

By the way i really like this library, here is a comparison video i made back then https://youtu.be/HKWDGangWU0

mrfaptastic, 2020-08-24:
> That testing was quite old so it could be indeed be that its now working better/with more like with your tests.
>
> It is so dark in that example as the FM6126 Chips where new to me, now its a good [...]

Glad you like it! And that video is awesome! Love the slow motion.

Galaxy-Man, 2020-08-25:
Using your example animated library I’m running 1 x panel (FM6126/FM6124) and using the standard library and settings it is drawing no more than 0.325 amps. So for testing would it be fair to say if max panels are 16 then this times 0.325 amps = 5.200 amps?

I have all my ribbon cables so will try 4 panels tonight and see what power is used.

12-16 panels? is this using the library as is or have you an updated library? I’m ordering more panels tomorrow, hope they are all the same chip set, fingers crossed 😊

FYI I’m not into making a big screen TV, if I want one of those I’ll buy an OLED Samsung lol, I’m doing this for my own curiosity 😊 I find electronics and programming like a never ending puzzle, the size of the puzzle is only limited by your imagination.

Galaxy-Man, 2020-08-25:
Hi atc1441

Would you be able to release the code for your youtube video, because we can use that as a benchmark when we are all comparing panels, code, and mcus.

witnessmenow, 2020-08-25:
A 64*32 panel can draw up to 2.5 amps if all pixels are set to white at full brightness, so be careful with something variable like a gif , cause if some of the frames are mostly white it will draw alot more power

Galaxy-Man, 2020-08-25:
Thanks Brian, checked the logs and as mentioned in email using mrfaptastic default settings and brightness set to 10, max draw is on the neo matrix image at 0.325 amps, more white in gif. Only using this for test purposes, will try on12 panels linked shortly. Watch for the blue smoke lol 😊

Galaxy-Man, 2020-08-25:
Hi Brian
Forgot to mention the video you did Use AdafruitGFX To Bodge Solutions
really good, gave me a better understanding of the library.

mrfaptastic, 2020-08-25:
> FYI I’m not into making a big screen TV, if I want one of those I’ll buy an OLED Samsung lol, I’m doing this for my own curiosity 😊 I find electronics and programming like a never ending puzzle, the size of the puzzle is only limited by your imagination.

Same. And come Christmas time one can have a huge LED matrix screen (chained panels of course) blasting Merry Christmas or something out the window facing the street, illuminating everyone who walks by.

Galaxy-Man, 2020-08-25:
I’m still using my Pioneer 50 inch plasma, really great picture even after +10 years ! also makes a great heater lol

How many panels are you short of to try out 3x4, I’ll order more and send them when they arrive.

mrfaptastic, 2020-08-25:
> A 64*32 panel can draw up to 2.5 amps if all pixels are set to white at full brightness, so be careful with something variable like a gif , cause if some of the frames are mostly white it will draw alot more power

Very true! @Galaxy-Man I also suggest, if you haven't already, soldering a decent cap (like a 2200uf 10volt) directly to the back of each panel to help with variable loads.

I bought a bunch of 2200uf 10V low ESR caps from this guy: https://www.ebay.com.au/itm/5-X-Panasonic-Ultra-Low-ESR-105-Degrees-FR-Range-Radial-Electrolytic-Capacitors/351313817882

Helped greatly with overall power stability.

[image]

mrfaptastic, 2020-08-25:
> I’m still using my Pioneer 50 inch plasma, really great picture even after +10 years ! also makes a great heater lol 

It'd still be an order of magnitude more energy efficient than these panels! 

Some maths here for laughs then:
So a 64x32 panel showing full white chews 2 amps @ 5 volts (conservative), so let’s say about 1 amp per 1000 pixels.
DIY Full HD screen = 1920 x 1080 = 2073 amps @ 5 volts hahaha!
P3 modules are about 19cm wide, so you’d need 30 per row, which is approx. 6 meters wide.
Would need about 34 rows, for a total of 1020 panels.
Assuming 12 62x32 LED panels per ESP32, you'd only need 85 ESP32's running in parallel doing their image segment, to get your full-hd picture.

+1 ESP extra to do a UDP broadcast WiFi message to the other 85 ESPs (:-)), with 16 bit color = ~4MB per frame. Could probably get a frame change per 30 seconds. Pretty sure WiFi interference of so many ESP32's in the one spot would kill the ability to communicate to all of them. That and the fact anybody's WiFi internet in the surrounding areas would be caput. Hahaha!

> How many panels are you short of to try out 3x4, I’ll order more and send them when they arrive. 
Are you sure? That's like 6 more panels lol!

Galaxy-Man, 2020-08-25:
Everyone around to my bedsit to watch Oblivion, I’ll order Deliveroo and have plenty of 50p coins for the meter! We have 12 panels between us only 1008 to go! So we could watch a bit of the film 😊

At normal 24fps the film would be 7,200 frames in 5 mins, using the Superdooper Vision LED Matrix TV using 1x frame per 30 sec we’d probably all fall asleep in the first 5 mins of the film as only 10 frames of the 7,200 would have gone by or 5mins @ 24fps translates to 60hrs using Superdooper vison . lol

Galaxy-Man, 2020-08-30:
Morning mrfaptastic, I'm back online been a busy week. Good tip ref capacitor for mulitpanel setup. Panels should be with me this week, I'll ship them out to you as soon as they arrive in deliveries.
I keep forgetting these are not closed conversations, everyone on GitHub reads, copies and actions our communications. As Brian mentioned also a good tip, when running panels at full brightness carefull of current usage. I'm using multi Tenma bench supplies so should be ok, also running setbrightness around 10. 
Something I spotted one ESP32 I'm using only runs max 160mhz, and my others which are the 38 pin yellow header run at 240mhz. Not sure if that will make a difference to the fps as the DMA does the work, is my understanding correct?

mrfaptastic, 2020-08-30:
Hi @Galaxy-Man. Let me know if there's any comments you want deleted to protect identity, but I have been sending you the odd email directly - no reponse received yet. Thanks again for sending through panels!

The CPU speed is not of relevance - but it certainly will help with graphical calcs. If you want to speed up the DMA output to the GPIO's the frequency can be changed on about line 119 of 'ESP32-RGB64x32MatrixPanel-I2S-DMA.h'. It's currently set to 10Mhz which is a good trade off for 64x32 panels. For a huge chain it could be increased up to about 40Mhz to keep the refresh rate high. Increasing the frequency also increases memory usage so it's trial and error.

Galaxy-Man, 2020-08-30:
Hi mrfaptastic

Emails direct are no problem at all, sometimes I may come across as terse which is not intended and I forget others read our comments 😊

Thank you for pointing to the change of frequency at line 119, Ill give it a go.

Just finishing work, will email in week once panels are here. Suppliers really slow at moment!

Have a great weekend.

GM

witnessmenow, 2020-09-03:
Where are you guys getting the 64x32 panels at the moment? I just want to try link to ones that work with the library on my documentation!

Galaxy-Man, 2020-09-04:
Morning all,

I am awaiting delivery of panels before forwarding to mrfaptastic to test >12 panels.

Suppliers are very slow at moment, expected this week but probably next.

Galaxy-man

kosso, 2020-09-06:
Sorry if this sounds like a silly question, but I have a few 64x32 panels which I want to chain. 

The instructions say to set a GPIO pin for the 'E' pin. The panels don't have an 'E' pin (because they're only 64x32).  Leaving me confused by the instructions. (I thought the 'E' only exists on *physical* 64x64 panels ?)

So do I simply need to set an unconnected pin for 'E' to make this work with chained panels?

thx.

mrfaptastic, 2020-09-06:
Hi @kosso. No need to worry about the E pin. Only required for 64 (row) x64 panels.

kosso, 2020-09-07:
Hi. Thanks. I eventually figured it out early this morning. 

I also finally got many of the Aurora demos working on a pair of stacked 32x64 panels using the chained 'virtualDisp'(lay), after much editing of the patterns and Effects files. :+1:

mrfaptastic, 2020-09-07:
Good to hear. If you're feeling generous, please feel free to raise a pull request and contribute. I'll create another example with the virtualDisplay.

mrfaptastic, 2020-09-16:
Closing this issue as with Kosso's use of the VirtualDisplay library it's clearly proven to be working.

Removed a few comments in this thread which leaked personal details / chat.

closed 2020-09-16 by mrfaptastic

mrfaptastic, 2021-02-08:
@Galaxy-Man - FYI - As part of testing a new development branch, I have finally chained up the 12 x 64x32 panels of different types and it is working flawlessly. Managed to test this off one lone USB power pack.

[image]

mrfaptastic, 2021-02-08:
@vortigont - The library changes work a treat, only needed to change: 

```
#define NUM_ROWS 3 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 4 // Number of INDIVIDUAL PANELS per ROW
```

In the sketch and upload!!

Galaxy-Man, 2021-02-08:
Hey, how are you 😊 Hope you are keeping well?

That is brilliant news getting all 12 working, so obvious question what is the limit? Do you want more panels to try, may need bigger PSU?

I have ordered late Dec from China more panels but not arrived yet ☹

I have done very little on any Arduino stuff other than look at the Teensy with a mate who programs it in ASM. I wish I had more time, work is priority during Covid.

I read you were venturing into Android, how’s that working out? Have you tried the NDK yet?

[Image removed by sender. IMG_20210208_214900]<https://user-images.githubusercontent.com/12006953/107286654-c7ff5d80-6a58-11eb-8cb0-b01273f17c8f.jpg>

vortigont, 2021-02-08:
Those 12 looks awesome! Not even a glitch! :) So it makes it what... 32x64x12, right? Gotta go and throw out of the window my old tv panel... ha-ha :))))

mrfaptastic, 2021-02-09:
> Hey, how are you 😊 Hope you are keeping well? That is brilliant news getting all 12 working, so obvious question where is the limit, memory or panels? Would you like more panels to try, and may be bigger PSU?

Hi @Galaxy-Man. I am good thanks, thanks for asking - keeping as sane as one can during Covid. Luckily there's always a bug to fix or something to code to keep one entertained. Glad to hear you're well, if not busy.

Ideally I'd like to try and get this to work as a 4x4 grid (256x128) @ 8bpp color, so 16 panels in total - but I think that might be a tall ask given the memory required, and perhaps, the 2-3meters of chaining required. Always happy to try with a couple more panels.

> Have you tried the NDK yet? Very impressed with your work quality and speed, some where along the way you should make money from this, I hope you are.

My entry into Android coding was out of sheer boredom - only Java/Kotlin based stuff at the moment and for my own learning. I don't make any money off this or other projects - the goal is to learn and have fun above all else. Others (such as @vortigont @witnessmenow) have made major contributions to this project which keeps the wheels turning!

> Those 12 looks awesome! Not even a glitch! :) So it makes it what... 32x64x12, right? Gotta go and throw out of the window my old tv panel... ha-ha :))))

Correct! I tried the current vs. dev branch and both seemed to 'KABOOM' :) around 15 panels though.

Archer-1983, 2021-11-09:
> Hello,
>
> First, thanks a lot of the library, it is working great! Much appreciated.
>
> I have made a shield for using this library with the D1 mini style ESP32 board (if you want one @mrfaptastic , I [...]
 

> Hello,
>
> First, thanks a lot of the library, it is working great! Much appreciated.
>
> I have made a shield for using this library with the D1 mini style ESP32 board (if you want one @mrfaptastic , I [...]

hi
can you help me how to use this library for P10 1/4 scan RGB panels
thank

[8 more replies]