mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #28: Display resetting when using setFont

madnat, 2020-08-14:
If I comment out line 43 my project runs... when left as is - the display resets when the print function is called.

What am I missing?

Any help very much appreciated

// Test the RGB64x32MatrixPanel_I2S_DMA setFont Function

#include <Arduino.h>
#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>
#include <Fonts/FreeSans9pt7b.h>

RGB64x32MatrixPanel_I2S_DMA matrix;

////////////////////////////////////////////////////////////////////
// Font Test
//
//
// pinout for ESP32 38pin module
//

// R1 | G1
// B1 | GND
// R2 | G2
// B2 | E
// A | B
// C | D
// CLK| LAT
// OE | GND

#define R1 25
#define G1 27
#define BL1 26
#define R2 14 // 21 SDA
#define G2 13 // 22 SDL
#define BL2 12
#define CH_A 23
#define CH_B 19
#define CH_C 5
#define CH_D 17
#define CH_E -1 // assign to pin if using two panels
#define CLK 16
#define LAT 4
#define OE 15

void setup(){

matrix.setFont(&FreeSans9pt7b); // This command causes a reset

// If you experience ghosting, you will need to reduce the brightness level, not all RGB Matrix
// Panels are the same - some seem to display ghosting artefacts at lower brightness levels.
// In the setup() function do something like:

matrix.setPanelBrightness(10); // SETS THE BRIGHTNESS HERE. 60 OR LOWER IDEAL.
matrix.begin(R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK); // setup the LED matrix
matrix.setTextColor(matrix.color444(255, 255, 255)); // r,g,b
matrix.setCursor(6, 7);
matrix.print("Leeds   4");
matrix.setTextColor(matrix.color444(255, 0, 0)); // r,g,b
matrix.setCursor(6, 17);
matrix.print("Man Utd 0");

}

void loop()
{

}

csloz, 2020-08-14:
Matrix.begin needs to be set first to initialize the library.
eg 

RGB64x32MatrixPanel_I2S_DMA matrix;

...

void setup(){
// If you experience ghosting, you will need to reduce the brightness level, not all RGB Matrix
// Panels are the same - some seem to display ghosting artefacts at lower brightness levels.
// In the setup() function do something like:

matrix.begin(R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK); // setup the LED matrix

matrix.setFont(&FreeSans9pt7b); // This command causes a reset

matrix.setPanelBrightness(10); // SETS THE BRIGHTNESS HERE. 60 OR LOWER IDEAL.

matrix.setTextColor(matrix.color444(255, 255, 255)); // r,g,b
matrix.setCursor(6, 7);
matrix.print("Leeds 4");
matrix.setTextColor(matrix.color444(255, 0, 0)); // r,g,b
matrix.setCursor(6, 17);
matrix.print("Man Utd 0");
}

mrfaptastic, 2020-08-14:
Thanks csloz. 

@madnat - I've updated the README now to highlight this.

madnat, 2020-08-14:
Thanks!  Regrettably the code still aborts with this change :(

Again - if I comment out "matrix.setFont(&FreeSans9pt7b); // This command causes a reset" the code runs with default font (6x8)

// Test the RGB64x32MatrixPanel_I2S_DMA setFont Function

#include <Arduino.h>
#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>
#include <Fonts/FreeSans9pt7b.h>

RGB64x32MatrixPanel_I2S_DMA matrix;

////////////////////////////////////////////////////////////////////
// Font Test
//
//
// pinout for ESP38 38pin module
// http://arduinoinfo.mywikis.net/wiki/Esp32#KS0413_keyestudio_ESP32_Core_Board
//

// R1 | G1
// B1 | GND
// R2 | G2
// B2 | E
// A | B
// C | D
// CLK| LAT
// OE | GND

#define R1 25
#define G1 27
#define BL1 26
#define R2 14 // 21 SDA
#define G2 13 // 22 SDL
#define BL2 12
#define CH_A 23
#define CH_B 19
#define CH_C 5
#define CH_D 17
#define CH_E -1 // assign to pin if using two panels
#define CLK 16
#define LAT 4
#define OE 15

void setup(){
  
matrix.begin(R1, G1, BL1, R2, G2, BL2, CH_A, CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK); // setup the LED matrix

matrix.setFont(&FreeSans9pt7b);

matrix.setPanelBrightness(50); // SETS THE BRIGHTNESS HERE. 60 OR LOWER IDEAL.
matrix.setTextColor(matrix.color444(255, 255, 255)); // r,g,b
matrix.setCursor(6, 7);
matrix.print("Leeds   4");
matrix.setTextColor(matrix.color444(255, 0, 0)); // r,g,b
matrix.setCursor(6, 17);
matrix.print("Man Utd 0");

}

void loop()
{

}

mrfaptastic, 2020-08-14:
Apologies. I broke something in the most recent release. Let me debug and get back to this issue.

mrfaptastic, 2020-08-14:
Try again with the latest release 1.2.2 - download from github and try again. THanks

madnat, 2020-08-14:
That now works - thanks very much!

mrfaptastic, 2020-08-14:
Excellent. Closing this issue.

closed 2020-08-14 by mrfaptastic