mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #29: Great version 1.2.2, but little problem with slow brightness adjustment.

FrenchLab47, 2020-08-20:
Hello mrfaptastic,

I just updated your library to version 1.2.2 and it still works fine.
Since this version, the whites are perfect, and the colors are more contrasted.

I am using an ESP32 is two in-line RGB LED matrices (128x32)
And to get a good fluidity of the texts I increased min_refresh_rate to the value 1600 (I don't know the real limit)

Since this version 1.2.2, I just have a small problem with the "setPanelBrightness" command which no longer acts instantly.
I have a page an html page (in websocket) with a slider that allows you to adjust the brightness.
The slider value is read by setPanelBrightness each time the main loop is passed.

With the previous version the brightness can change immediately when I touch the slider.
Now I have to wait several loop (up to 5/10s) or a reset, for the brightness to change.

However if I display my instant value with Serial.print(myValue) this value changes well in same time with the cursor, but the setPanelBrightness command is much slower.

Did you change anything around the brightness setting?
Or do you have an idea?

Thank you for your work and your responses.

mrfaptastic, 2020-08-20:
Hi @FrenchLab47. Glad you find the new library working well. I did make a change to the code to try and improve the 'drawPixel' performance quite a bit and the brightness check was a casualty.

There are two options to resolve:

1) Immediately after you call setPanelBrightness, do a clearScreen() and repaint.
2) Or on line 436 comment out ` #define GO_FOR_SPEED 1` aka:

```
Turn

 #define GO_FOR_SPEED 1

to

// #define GO_FOR_SPEED 1
```

Above this #define you'll see what I was trying to get at. I think I might update the GIT so this is actually commented out by default. I got carried away trying to performance tune PatternPlasma to work smoothly on a  256x256 pixel matrix :-D!

mrfaptastic, 2020-08-20:
On that note, you wouldn't mind sharing the code you for a) the websocket server & brightness call and b) the html / js required to do the scroller that does the websocket call? I would love to use this in one of my projects.

FrenchLab47, 2020-08-20:
Thank you for this quick and precise answer, I confirm that it works.

I plan to share my code soon, with the web settings pages (mobile), the printed circuits, and the parts of the box in 3D printing.
I still have a few cosmetic bugs to resolve, a bit of cleaning up in the code, I'm about 95% of my job.

Best regards