mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #33: Alternative method for buffering on 2x2 chained panel

maglevize, 2020-09-19:
Hello @mrfaptastic 
Been great use this librarry, it’s so straight forward and easy to understand.
Currently I’m able to do 2x2 using class VirtualMatrixPanel on panel fm6124 for total res 128x64.
Try simple scroll text, which I need to refresh each frame with fillscreen(0). This make flickering issues.
And I see example about framebuffer, introduce to flipDmaBuffer(), but it’s not available on VirtualMatrixPanel class.

Can you suggest what best way to get rid this flickering on my code?

Thanks

mrfaptastic, 2020-09-20:
Can you post your sketch code? In any case I have made an addition to the virtualDisplay class to include the the flip/show dma buffer functions (even though these could be accessed directly from the dma_display instance).

Try again with the latest git version of the library.

maglevize, 2020-09-21:
this is my function within code to update display:
`void disp_update_msg(String msg) {
  dma_display.flipDMABuffer();
  virtualDisp.fillScreen(virtualDisp.color444(0, 0, 0));
  virtualDisp.setCursor(0, 20);
  virtualDisp.println(msg);
  dma_display.showDMABuffer();
}`
but display is blank / not updating.
I'll try latest version soon, will keep updating.

Thanks

mrfaptastic, 2020-09-21:
Refer to: https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/blob/master/examples/DoubleBufferSwap/DoubleBufferSwap.ino

Make sure you're starting with 'true' :

```
RGB64x32MatrixPanel_I2S_DMA     display(true);      // Note the TRUE -> Turns of secondary buffer - "double buffering"!
                                                    // Double buffering is not enabled by default with the library.
```

maglevize, 2020-09-22:
I can confirm that latest version flipDMABuffer() and showDMABuffer() with VirtualMatrixPanel is working.

> Refer to: https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/blob/master/examples/DoubleBufferSwap/DoubleBufferSwap.ino
> Make sure you're starting with 'true' : [...]

Yes you're right, I completely forgot to set to true.
Now flicker is all gone, great and thanks.

closed 2020-09-22 by mrfaptastic