mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #34: AnimatedGIFPanel example error

advadv00, 2020-09-22:
Hello, this is AnimatedGIFPanel example error.
No change was made in the original example.
Sketch data already uploaded.
I dont know why...

아두이노:1.8.13 (Windows 10), 보드:"WEMOS D1 MINI ESP32, 80MHz, Default, 240MHz (WiFi/BT), 115200"

C:\Users\inhyu\Documents\Arduino\libraries\ESP32-RGB64x32MatrixPanel-I2S-DMA-master\examples\AnimatedGIFPanel\AnimatedGIFPanel.ino: In function 'void ShowGIF(char*)':

AnimatedGIFPanel:171:82: error: invalid conversion from 'void* (*)(char*, int32_t*) {aka void* (*)(char*, int*)}' to 'void* (*)(const char*, int32_t*) {aka void* (*)(const char*, int*)}' [-fpermissive]

   if (gif.open(name, GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile, GIFDraw))

                                                                                  ^

In file included from C:\Users\inhyu\Documents\Arduino\libraries\ESP32-RGB64x32MatrixPanel-I2S-DMA-master\examples\AnimatedGIFPanel\AnimatedGIFPanel.ino:23:0:

C:\Users\inhyu\Documents\Arduino\libraries\AnimatedGIF\src/AnimatedGIF.h:138:9: note:   initializing argument 2 of 'int AnimatedGIF::open(const char*, void* (*)(const char*, int32_t*), void (*)(void*), int32_t (*)(GIFFILE*, uint8_t*, int32_t), int32_t (*)(GIFFILE*, int32_t), void (*)(GIFDRAW*))'

     int open(const char *szFilename, GIF_OPEN_CALLBACK *pfnOpen, GIF_CLOSE_CALLBACK *pfnClose, GIF_READ_CALLBACK *pfnRead, GIF_SEEK_CALLBACK *pfnSeek, GIF_DRAW_CALLBACK *pfnDraw);

         ^

exit status 1

invalid conversion from 'void* (*)(char*, int32_t*) {aka void* (*)(char*, int*)}' to 'void* (*)(const char*, int32_t*) {aka void* (*)(const char*, int*)}' [-fpermissive]

mrfaptastic, 2020-09-22:
Try again now with latest example .ino from git repository.

advadv00, 2020-09-22:
It solved!
Thank you!

closed 2020-09-22 by advadv00