mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #36: Consider changing the name of the library?

witnessmenow, 2020-09-29:
I wonder does the "64x32" part of the library name put people off using it? I have used it extensively on a 64x64 and have had no issues. I know several people who have used it on 32x16 panels and it has also worked perfectly

Just a suggestion :)

sauttefk, 2020-09-29:
What about `EPS32-HUB75-MatrixPanel-I2S-DMA`

mrfaptastic, 2020-09-30:
I could be convinced if it wasn't for issues like #35 and #37. Anybody got any of these panels to understand what the problem could be?

Galaxy-Man, 2020-11-10:
How about 'TheOneMatrix'?

mrfaptastic, 2020-11-28:
> What about `EPS32-HUB75-MatrixPanel-I2S-DMA`

I think this is the winner. Not having tested the library myself witha  64x64 panel, I have confidence from others it works ok, so will rename the library and get the arduino folks to update the URL.

closed 2020-11-28 by mrfaptastic