mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #37: 32x32 Matrix not working!

MythicalForce, 2020-09-29:
I know that this library only has been tested with 64x64 matrix but i've heard that people have got it working with a 32x16 so i thought it would be just plug and play here but sadly no :/

my board is
 **IFH6-32x32-16S-V1.0 (HUB75D)**
[](https://www.adafruit.com/product/607)

mrfaptastic, 2020-09-30:
Can you pst a link to the panel you bought? Can you try the earliest version of the library 1.0.2 and see if that makes any difference? Please triple check your cabling?

MythicalForce, 2020-09-30:
> Can you pst a link to the panel you bought? Can you try the earliest version of the library 1.0.2 and see if that makes any difference? Please triple check your cabling?

this is the board -> [](https://www.adafruit.com/product/1484)

i've gone over it 7 times!! everything is fine

Code running

```
#ifndef MYMATRIX_H
#define MYMATRIX_H

#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>

RGB64x32MatrixPanel_I2S_DMA matrix;

namespace Matrix 
{

void begin()
{
    Serial.println("...Starting Display");
    matrix.begin();

    matrix.fillScreen(matrix.color565(255, 0, 0)); 

    matrix.setPanelBrightness(16);
    //matrix.clearScreen();
}

void update()
{
}

};

#endif

MythicalForce, 2020-09-30:
[Here is a image of what im seeing, running the code i posted above here!](https://ibb.co/NtHZVZG)

MythicalForce, 2020-09-30:
@mrfaptastic so just a quick update! i've checked my wires 9 times now (i even de-solder and re-soldered the wires)

i have NO problems using fill to actualy fill the entire screen with a solid color! but when it comes to animations / image display and so on the display is just garbage :/ /using your example codes only)

mrfaptastic, 2020-09-30:
That's good news. Can you show a photo of the testshapes_32x64 sketch? It's almost impossible to diagnose what is going on. From the first screenshot of the failed red, it looked like D line wasn't connected.

Perhaps do a simple sketch that draws a line from the top left to the bottom right as well - show a photo.

MythicalForce, 2020-09-30:
soooo i did this in setup `matrix.drawRect(0, 0, 32, 32, matrix.color565(0, 255, 0));` and using default pinout (i've even hardwired it now to a breadboard!)

[and this is the outcome](https://ibb.co/DbZSQf0) (btw RL it's flickering like crazy)

mrfaptastic, 2020-09-30:
Can you try another library like pxMatrix or something? Also, can you do a drawLine instead of a drawrect? Something I think is wrong with the panel perhaps.

MythicalForce, 2020-09-30:
actualy i had my Matrix working 100% with PXMatrix library :) so i know it's working :)

yes i adjusted WIDTH and HEIGHT in the .h file :)

mrfaptastic, 2020-09-30:
This is turning into one of these mysteries. Try drawing a single pixel / single diagonal line etc.... might help understand what's going on. also try version 1.0.2 of this library etc.

mrfaptastic, 2020-10-01:
Can you try with the earlier version of this library? 1.0.2

I am hoping there isn't a regression that I have inadvertantly introduced for some 32x32 px panels!

MythicalForce, 2020-10-04:
im on 1.0.2 now and if i set

`#ifndef MATRIX_HEIGHT
#define MATRIX_HEIGHT               32 //64
#endif

#ifndef MATRIX_WIDTH
#define MATRIX_WIDTH                32
#endif`

the colors and pattern is fucked up and if i set to `32x64` then it works but is wrong size since i have a 32x32 so :S but color and pattern is correct

running the plasmapattern btw! please make a fix for 32x32 <3 would realy like to use this library!

MythicalForce, 2020-10-05:
@mrfaptastic this is the ESP32 im using dunno if it differs from the one you used to build this library

[DOIT ESP32 DEVKIT 1](https://circuits4you.com/wp-content/uploads/2018/12/ESP32-Pinout.jpg)

mrfaptastic, 2020-10-27:
Can you show a photo of your wiring? The shorter the cable the better.

mrfaptastic, 2020-11-09:
Hi, I tested this with 32x32 on a 64x32 (width, height) and as expected I got two 32x32 images showing on the single panel. I can't find a datasheet on this Adafruit panel on how it works, but this library won't work with anything with less rows than 32px.

closed 2020-11-09 by mrfaptastic