mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #38: Option to change how pixels are addressed on the grid.

pgooch, 2020-10-21:
I've set up a simple little demo with an EPS32 and 2 64x32 panels and it all works, this is more of a quality of life/ease of use sorta thing.

So instead of the traditional layout description where all the panels are connected in one long row I'd like to work with them in one long column. I'm not talking stacked I'd like to run them end to end. For instance lets say I have 5 panels; with the current chaining method I could (theoretically, not tested) create a chain of 320 wide by 32 high, however I'm looking for 32 wide and 320 high.

I think this could be faked by having the options to swap out where the defined coordinate lands, as the rest of the chaining seems to work fine. I considered just writing a shim function to do that but since X and Y change positions it doesn't seem like it'll be as smooth a process as I imagined (I come from a web dev background where higher level languages would make this simpler).

It's possible there is a way to do this with the virtual display library but it seemed like it was the only option not supported.

pgooch, 2020-10-21:
Just in case someone else comes across this I fixed it by using the following functions to get the X, Y, and Color values respectivly.

```
/*
 * These functions take an x and y at the desired orinentation and convert them to an x and y at the orintation the driver wants
 */
int gX(int x, int y){
  return y;
}
int gY(int x, int y){
  return abs(MATRIX_WIDTH-x)-1;
}
int gC(int r, int g, int b){
  return matrix.color565(r,b,g);
}
```

mrfaptastic, 2020-10-21:
Hi pgooch,

Glad the library is of use to you. Electrically / physically the only way to connect these panels up is by way of on 'horizontal' (i.e. Serial) chain of panels. That's how it works. 

Are you referring to rotating the horizontal chain vertically however, like this?

[image]

If so, custom code will be required here as code hasn't been written for this particular type of arrangement in the virtual display class... yet.

> **Edit** I've noticed another quirk about my panel, it's not RGB, it's RBG.

Bizzare. An alternative option is to swap the B_1, B_2, G_1, G_2 pins from the ESP to the first panel (unless you're using a ribbon cable or something)

pgooch, 2020-10-22:
Yes, thats the layout ideal I had (well, I was thinking the ESP on 4 and going up but thats a physical orientation thing.

I totally didn't think of swapping pins around, thats a much nicer solution that my current messy code one.

I've also notice what appears to be artifact caused by asking the matrix to do to much, pixels next to the intended one lighting up one of the LEDs or scan-line like effects. I imagine they are more hardware than software related, would that be a good guess given your experience?

mrfaptastic, 2020-10-22:
> I've also notice what appears to be artifact caused by asking the matrix to do to much, pixels next to the intended one lighting up one of the LEDs or scan-line like effects. I imagine they are more hardware than software related, would that be a good guess given your experience?

Refer to the double buffering example.

pgooch, 2020-10-24:
Thanks for the tip, will do once I'm done with getting other things working (not related to the library, it's rockin'). It seems to be related to power, if I power the matrix through the board it doesn't happen, but when I provide an external 5V it does. Just an interesting thing I noticed when playing around with it.

mrfaptastic, 2020-10-24:
These panels are a huge power suck, and need to be powered externally. Also, it's important to put a large  2000uf capacitor on the back of the panels between VCC and GND if possible as well to deal with power draw spikes due to the graphics.

pgooch, 2020-10-24:
I was only powering 2 pixels for testing some code, but thanks for the tip I picked up some caps that I'll install.

mrfaptastic, 2020-11-05:
Did you end up getting to the bottom of this issue. Can I close it?

closed 2020-11-09 by mrfaptastic

pgooch, 2020-11-11:
I did solve it, it was something in the mess of wires, I ordered a PCB and it works exactly as expected. Thanks for the library man, works great.