mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #39: Red flickering when two 64x32 panels are chained

dom96, 2020-11-01:
What i've done:

* Verified my wiring is correct using continuity test on multimeter
* Tried master + all released versions of this library

The displays work fine individually but when connected together the red flickers a lot. I have written this example to demonstrate:

```
#define MATRIX_WIDTH 128 // These don't work, something that would be nice to fix :)
#define MATRIX_HEIGHT 32

#include <ESP32-RGB64x32MatrixPanel-I2S-DMA.h>
#define MATRIX_WIDTH 128
#define MATRIX_HEIGHT 32
RGB64x32MatrixPanel_I2S_DMA matrix;

void setup()
{ 
  // MUST DO THIS FIRST!
  matrix.begin();  // Use default pins supplied within ESP32-RGB64x32MatrixPanel-I2S-DMA.h
  // matrix.begin(R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN );  // or custom pins

  matrix.fillRect(9,9, 9, 9,  matrix.color565(255,255,255)); // can do this after .begin() only
  matrix.fillRect(18,9, 9, 9,  matrix.color565(255,0,0)); // can do this after .begin() only
  matrix.fillRect(9,18, 9, 9,  matrix.color565(0,255,0)); // can do this after .begin() only
  matrix.fillRect(18,18, 9, 9,  matrix.color565(0,0,255)); // can do this after .begin() only
  Serial.begin(9600);
}

void loop()
{
    Serial.println("Loop");
    delay(1000);
  
}
```

With just one panel:

[image]

(Tough to take a good photo, but all rectangle are lit up correctly and don't flicker)

With two panels (chained but width set to 64):

[image]

Close up of right:

[image]

When width is set to 128 in the .h file the display is similar to the above picture. There is slightly less flickering.

Would love to get some hints on how to debug this, seems like it could just be a simple change somewhere so I'd love to give fixing it a shot.

mrfaptastic, 2020-11-02:
Hi Dom,

I'm slightly confused, the sketch code you is designed to display one 4x4 grid of colours on a single panel right? 

Ye, if you're seeing the same 4x4 grid of colours  on both chained panels with that sketch code, then it's because it isn't actually configured to 128x32 pixels. For example, if you have six 64x32 panels chained together, but the library is configured to 64x32px (not 384x32), then you will see the same image repeated 6 times - and this may have weird results like flicker etc.

dom96, 2020-11-02:
Hey @mrfaptastic,

Thanks for the reply! So yes, sorry, didn't mean to confuse. Just to be clear, I have set the width to 128px and still get the same results. The flickering is slightly less but still definitely there on the red colour. I included the image showing what the results look like with the width set to 64px because I thought it might give some hints as to the problems (no idea how these panels work tbh) but I understand that this is not how I should be running these panels.

So just to be clear:

* The code draws 4 squares, white, green, blue and red.
* When I run this on a single 64x32 P3 panel it all looks good.
* If I run this on two 64x32 P3 panels chained together I get flicker on the red square and the white square (since the white square also lights up red, and I can clearly see the red LEDs flickering with the others being steady)

> Do you have any capacitors on each panel between VCC and GND?

I've connected the panels to a typical "breadboard power supply module". One of these:

[image]

I connect the panels to the 5v on that module. I believe it regulates the power for me (and I see a capacitor on it). Is this good enough?

>In the .h file -> #define ESP32_I2S_CLOCK_SPEED (10000000UL) // @ 10Mhz - perhaps increase this to 20?
>In the .h file -> int min_refresh_rate = 99 // perhaps drop this to 50?
>In the .h file -> #define PIXEL_COLOR_DEPTH_BITS 8 // Perhaps drop to 5?

Will try these and report back :)

dom96, 2020-11-02:
So I tried all of these changes and nothing helped. But I did notice that the flickering doesn't begin immediately if I shut down the power and bring it back. It takes around 5s for the brightness to dim and then the flickering to start.

I should mention that this is when I am powering my ESP from a USB port on my PC and not the power supply I use to power the display. Perhaps this is to blame. 

I just tried connecting the ESP straight to my power supply, this time it runs fine for the first 12 seconds, then the whole display flicks off, flicks back on, then off again, then turns off completely. I'm honestly quite confused why this happens. I suppose it's possible I am not connecting the ESP directly to power correctly (I just connected the 5V and GND on the ESP32 NodeMCU I've got to the power supply's 5v/gnd), but then why would it be working fine for the first few seconds? I might double check by flashing a simple web server onto it to see if it's stable then.

I'd appreciate any other ideas. Thanks!

mrfaptastic, 2020-11-02:
I think there are two issues:
1) Power and capacitors; and
2) Width configuration.

For 1), you need to have a proper capacitors for each panel, they are very noisy in how they draw power, this is what I have in order to get it to work properly:
[image]

Make sure you do not have the ESP32 connected to USB/pc at the same time as you're running it from the power supply! You'll fry your computer, and if you don't fry it, you'll cause weird things to happen.

For 2) You need to directly edit the 'ESP32-RGB64x32MatrixPanel-I2S-DMA.h' file and change the width to 128px. Putting defines in the .ino file do not work! You should only see ONE grid of four colours, as your project only prints onto the 128 chain/display once (unless you had created another .info file that prints another expicit 2x2 grid of colors 64 pixels to the right).

[image]
[image]

dom96, 2020-11-03:
There are still some misunderstandings here, let me clarify :)

> For 2) You need to directly edit the 'ESP32-RGB64x32MatrixPanel-I2S-DMA.h' file and change the width to 128px. Putting defines in the .ino file do not work! 

Yes! I am aware of this and have been changing it in the .h file. So the width is not a problem for me. Once again, I still get flicker even when I set the width to 128. I am certain that I am doing it right because the square does get shown on only one panel.

> Correct output (when library treats chain of panels as one 128px wide panel):

Yep, just to confirm again, this is what I see too. It's just that there is flicker.

> Make sure you do not have the ESP32 connected to USB/pc at the same time as you're running it from the power supply! You'll fry your computer, and if you don't fry it, you'll cause weird things to happen.

Of course, I have never connected the ESP32 both to my PC and the power supply. But I have connected the displays to my power supply and the ESP32 to my PC, I thought that this could cause this flickering because the grounds are different. But maybe even doing this is dangerous? If so, do you literally flash your ESP32 separately and re-connect to your power supply every time you make a change to the code? (seems like a lot of faffing around :))

> For 1), you need to have a proper capacitors for each panel, they are very noisy in how they draw power, this is what I have in order to get it to work properly. Large 1000-2200uf caps soldered driectly to the back of each panel:

This is indeed something I am missing, I'd love to add this into the circuit now but I don't actually have any spare capacitors. I just ordered some 1000uF ones from Amazon (they don't seem to have a lot of 2200uf from what I saw) and will give them a try once they arrive.

What chips are on your panels btw? My initial thoughts were that this is due to the "exotic" chip that the panels I have use (CHIPONE ICN2038S), in fact I've been using PxMatrix and the artifacts it gave me seemed to imply this: https://github.com/2dom/PxMatrix/issues/251. Do you think the lack of a capacitor is to blame for my issues with PxMatrix too?

mrfaptastic, 2020-11-03:
> What chips are on your panels btw? My initial thoughts were that this is due to the "exotic" chip that the panels I have use (CHIPONE ICN2038S), in fact I've been using PxMatrix and the artifacts it gave me seemed to imply this: [2dom/PxMatrix#251](https://github.com/2dom/PxMatrix/issues/251). Do you think the lack of a capacitor is to blame for my issues with PxMatrix too?

Well that's always the other wildcard, who knows what chipset is next to come out of China. Looking at my panels they appears to be TCS chips, but I also have a bunch of CHIPONE ICN2038S as well (that @galaxyman donated) which I have not had any issues with as demonstrate here: https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/issues/26#issuecomment-679229017

Basically, I don't think there's anything wrong with the library. Check your power supply etc.

dom96, 2020-11-05:
**I found the culprit:** the "breadboard power supply module" I showed above. Either that or the power supply I was using. I bought a beefier power supply earlier (15A at 5V) and today it arrived so I connected it to the power supply module, the result was the flickering stopped. But it still wasn't working 100% (especially when I filled the screen with white pixels).

At that point I decided to double check that the voltage coming out of the module was 5V... it wasn't :). At this point I didn't realise that my new 15A power supply was itself outputting 5V, so I suppose it's expected that the module was dropping that further down to ~3V. But then I reconnected my old power supply which outputs 12V at 5A, which I expected to get dropped to a steady 5V by the module, but when I checked the output coming from it the voltage started at 4.9V and then began to drop fast down to 2V, and the more it dropped the more flicker and less red LEDs were lit up.

In the end I connected my new power supply (15A at 5V) directly to the displays and that works perfectly (there is some small noise but I think this will be fixed once I connect the ESP32 to the same power supply and/or add capacitors).

Thanks so much for your help! Hope this gives others some hints too :)

mrfaptastic, 2020-11-05:
Thanks for the update. Glad to know there wasn't some hidden bug in the library. I have added a reference in the library's [readme](https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA/blob/master/README.md#power-power-and-power), which hopefully may help others.

daveythacher, 2021-01-11:
You are probably over 15A. I am guessing your power draw at full white is 15.36. This does not include overhead from ICs, caps, controller, etc. Adding a few extra Amps to the power supply should probably account for this. Maybe 20A will stop it, but that is just a guess.

dom96, 2021-01-11:
@daveythacher I wasn't anywhere close to full white though. In any case, a 5V 15A power supply works fine for me when connected directly.

In the meantime I did also check the current draw and it never gets anywhere near 15A even at full white. I double checked this with a DC Bench Power Supply too and it never got further than 5A IIRC.

daveythacher, 2021-03-24:
It depends if you are driving all of the LEDs or not. There is a little amount of power on the drivers and supporting logic.

This code may be based on BCM stuff found in RPI Matrix code that would explain it also. It semi intentionally misconfigures itself for better performance which lowers the current. It also has an algorithms which shifts brightness curves. Probably intended for other applications but it is probably close enough. Then there is other wacky stuff which can do stuff. If you know what your doing you can tailor it some.

SMEKH, 2021-05-27:
[image]

how to close running text so as not crash into the next text thx.
i have used the command matrix.fillRect(0,0,32,8, matrix.color565(0,0,0));
but its not work

daveythacher, 2021-05-27:
> @daveythacher I wasn't anywhere close to full white though. In any case, a 5V 15A power supply works fine for me when connected directly.
> 
> In the meantime I did also check the current draw and it never gets anywhere near 15A even at full white. I double checked this with a DC Bench Power Supply too and it never got further than 5A IIRC. It surprised me to learn this so maybe I managed to screw up my measurements somehow, but I did get them from two different devices so... I'm stumped :)

Panel is based on CIE1931, which lowers average current. Peak current is still very high, and would need a scope with measurement circuit to see it. Power supply response time is a factor. Panel multiplex ratio and current limit is factor. You setup could use only 7.2A peak current based on LEDs, however additional power consumption will exist from caps and logic. Wiring can be a factor. Etc.

felaniMT, 2021-07-22:
> > What chips are on your panels btw? My initial thoughts were that this is due to the "exotic" chip that the panels I have use (CHIPONE ICN2038S), in fact I've been using PxMatrix and the artifacts it gave me seemed to imply this: [2dom/PxMatrix#251](https://github.com/2dom/PxMatrix/issues/251). Do you think the lack of a capacitor is to blame for my issues with PxMatrix too?
> 
> Well that's always the other wildcard, who knows what chipset is next to come out of China. Looking at my panels they appears to be [TC5020AP](https://lcsc.com/product-detail/LED-Drivers_TC5020AP_C126197.html) chips, but I also have a bunch of CHIPONE ICN2038S as well (that @Galaxy-Man donated) which I have not had any issues with chaining them both as demonstrate here: [#26 (comment)](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/26#issuecomment-679229017)
> 
> Basically, I don't think there's anything wrong with the library. Check your power supply etc, ensure that the ESP32 and each panel have the GNDs and VCC's all linked together etc.

Hi, mrfaptastic. 

Recently i bought 64x32 panel with chipset TC5020AP, bur the result on the panel only red led is functioning not show for green and blue..do you have any idea what is the problem? For your information, i run the basic sketch test shape, it works but only for red color, same for plasma sketch only show red led..

mrfaptastic, 2021-07-22:
You have probably been sent a dodgy panel. I have a feeling that due to chip shortages etc, that some Chinese sellers are selling busted panels that would have been destined for the bin.
Connect the R1/R2 pins to G1/G2 or B1/B2 and if you still don't get anything other than red, then it looks like a refund is in order.
Check you power supply and grounding. Might not be putting out 4.75 volts or whatever is required.

mrfaptastic, 2021-07-22:
You have probably been sent a dodgy panel. I have a feeling that due to chip shortages etc, that some Chinese sellers are selling busted panels that would have been destined for the bin.
Connect the R1/R2 pins to G1/G2 or B1/B2 and if you still don't get anything other than red, then it looks like a refund is in order.
Check you power supply and grounding. Might be putting out too low a voltage under load etc.

felaniMT, 2021-08-03:
> You have probably been sent a dodgy panel. I have a feeling that due to chip shortages etc, that some Chinese sellers are selling busted panels that would have been destined for the bin.
> Connect the [...]

Hi mrfaptastic, you are right, i received a panel, what can i say is incomplete panel, when i see on the back of the panel every section which is labelled RG and RB is missing component there, but for RR have 1.2K smd resistor. Then i try to soldered a 1k resistor to all RG and RB,and now the panel work as intended, it show for color green and blue. I contact the seller and he said the panel need using their control card.  

Now i want to order smd resistor to do the rework, what rate of resistor i need to put for RG and RB, do you have any idea?

mrfaptastic, 2021-08-03:
I don't know, but somebody else might. I would request a refund or do a chargeback to the seller - although if you've already done stuff to the panel that might be difficult.

daveythacher, 2021-08-03:
I have one panel for TC5020 which uses Red 1.1k, Green 2.55k and Blue 3.65k. Note this assumes the LEDs are similar to the ones in my panel. This is done for white color balancing at 5 rail. 

I would recommend trying: 
Green - 2.8kOhm 1%
Blue - 4.02kOhm 1%

The current through these is small. 1.24 V / R, so P = 1.24 * 1.24 / R. Meaning 1/10 W resistors should work.

If you want to lower the voltage of the rail you have to lower the current limit via resistors. You will want the same current ratios between colors. However some experimentation may be required if the color temperature does not move with the current shift.

daveythacher, 2021-08-03:
> I don't know, but somebody else might. I would request a refund or do a chargeback to the seller - although if you've already done stuff to the panel that might be difficult.

I have found that with a hot air tool, the resistors can come off without too much risk. Get flux pen and apply a decent amount of flux before applying hot air. Use a temperature close to the solder temperature, maybe a tad higher. Limit the air speed to avoid blowing parts off.

Warm the board slowly with the nozzle far away from the board. Slowly work down in spiral wide and narrow around the component. Eventually the solder will reflow, shiny/wet look. Test gently with tweezers. Do not pull or push, the PCB may not take much abuse. Just enough to see the surface tension. It should move a little on both sides. Then lift it. 

Do not bake the board. Do not bake other components. Do not put nozzle near stickers or paper. Once the component has been removed slowly spiral nozzle back to avoid thermal shock. Try to be quick in the entire process to avoid too much heat. However do not thermally shock it.

Let the board cool before soldering resistors back on. Board will hold heat and can warp. Keep board on flat surface at all times. Remember to remove plastic case before hand. Solder resistors back on with iron. Use care in cleaning pads, or skip it. Resistors may not look very pretty but will not likely cause issues. Remember to use flux here too.

I am kind of noob at this and I did this several times without issue. Generally I have found getting stuff off the boards hold up. Putting it back on requires more skill than I have. The boards did not hold up when I tried swapping the SSOP LED drivers a few years ago.