mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #40: Drawing in different orientations

VictorHofstede, 2020-11-18:
Hi everyone,

would it be possible to write text in different orentations? I have a 64x32 panel, but I would like to use it lengthwise and write text into it. So it would function as a 32x64 panel. Would this be possible and if so, how?

csloz, 2020-11-18:
Yes.  

Just turn the panel sideways.

You can use Adafruit_GFX to display text in whatever orientation you feel like. 
It's that simple.

VictorHofstede, 2020-11-18:
Hmm I tried using 
  dma_display.setRotation(rotation);
with rotation set to 0, 1, 2, 3 and no matter what parameter it did not change the orientation of the text I wrote.

Example:
` 
dma_display.begin(); // use default pins
dma_display.setRotation(3);

dma_display.setCursor(0, 0);
dma_display.println("Test line");
`

closed 2021-06-21 by mrfaptastic