mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #43: getTextBounds returning an incorrect width

witnessmenow, 2020-12-05:
I was centering some text and I realized getTextBounds is returning an incorrect width, I'm not sure if this is a bug with this library or Adafruit GFX.

Here is the code I'm using the test:

```
dma_display.setTextSize(2);     // size 2 == 16 pixels high

int16_t xOne, yOne;
uint16_t w, h;

dma_display.getTextBounds("BB", 0, 0, &xOne, &yOne, &w, &h);
xPosition = dma_display.width()/2 - w/2 + 1;
Serial.println(w);

dma_display.setCursor(xPosition, 24);
dma_display.print("BB");
```

- Physically counting the pixels that the string takes up: 22
- Serial print of W: 24

It seems to always return 2 pixels bigger, regardless of the length of the string (even a single character)

It can be worked around for centering by just adding 1 pixel to where you calculate, but I thought I would raise it when I saw it.

mrfaptastic, 2020-12-05:
Hmmm... I think I had similar problems with a completely seperate 'GFX_Layer' class I made that inherited AdaFruit_GFX.

Try something like this and see if you get the same co-ords. Pretty sure this library has nothing to do with what getTextBounds returns. Unless of course the WIDTH orHEIGHT that is supposed to be passed to AdaFruit is supposed to start at 0 and therefore WIDTH-1 etc... which would be stupid.

```
class Layer : public Adafruit_GFX // use GFX Root for now
{
	public:

		Layer () : Adafruit_GFX (64, 32) {
			
		}
}
```

mrfaptastic, 2020-12-05:
I just pushed a commit to the repo with a 'Layer' class. Have a look at it. There's a function called autoCentre which is my hack to brute force ensure that variable with fonts are always in centre, as what getTextBounds returns is utter garbage.

mrfaptastic, 2020-12-07:
Created a new example called GraphicsLayer which may be of interest.

witnessmenow, 2020-12-08:
Cheers, will check it out!

mrfaptastic, 2020-12-08:
Created this simple Arduino .ino file, which only implemented AdaFruit_GFX, and get 24 pixels. So it's a bug with Adafruit I think.

```
#include <Adafruit_GFX.h>

class TestGFX : public Adafruit_GFX {

  public:
      TestGFX() : Adafruit_GFX(64, 32) {
        
  }

  void drawPixel(int16_t x, int16_t y, uint16_t color) 
  {
    Serial.printf("draw pixel x,y: %d, %d\n", x, y);
  }
}; // super basic arduino class

TestGFX displayTest;

void setup() {

  delay(100);
  Serial.begin(115200, SERIAL_8N1); // The default is 8 data bits, no parity, one stop bit. 

  // put your setup code here, to run once:
  displayTest.setTextSize(2);     // size 2 == 16 pixels high
  
  int16_t xOne, yOne, xPosition;
  uint16_t w, h;
  
  displayTest.getTextBounds("BB", 0, 0, &xOne, &yOne, &w, &h);
  xPosition = displayTest.width()/2 - w/2 + 1;
  
  Serial.print("Width: ");
  Serial.println(w);
  
  displayTest.setCursor(xPosition, 24);
  displayTest.print("BB");  

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

mrfaptastic, 2020-12-08:
Raised: https://github.com/adafruit/Adafruit-GFX-Library/issues/327

closed 2020-12-08 by mrfaptastic

mrfaptastic, 2021-02-17:
Old issue, but it was annoying me - so I've fixed it in my copy of Adafruit_GFX that works perfectly with the Matrix I2S library when #define USE_GFX_ROOT 1 

https://github.com/mrfaptastic/Adafruit_GFX_Lite

tsctrl, 2021-07-25:
hi @mrfaptastic ,

is this fix works with original libs? having the same issue