mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #45: Can't compile after last updates.

dinuxlt, 2020-12-07:
Can't compile after last updates. Code was working two weeks ago.
Now i get:
```
In file included from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:9:0:
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h:13:17: fatal error: GFX.h: No such file or directory
compilation terminated.
```
[image]

I opened `Layer.h`  and changed line number 13 from `#include <GFX.h>` to ` #include <Adafruit_GFX.h>`

After this I get:
```
In file included from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:9:0:
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h:14:21: fatal error: FastLed.h: No such file or directory
compilation terminated.
```
[image]

I'm using linux and on linux capital letters matters, so I changed line number 14 from `#include <FastLed.h>`  to `#include <FastLED.h>`

I now i get a lot of errors:
```
In file included from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h:14:0,
                 from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:9:
/home/dinux/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
                     ^
In file included from /home/dinux/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h:14,
                 from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:9:
/home/dinux/Arduino/libraries/FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h:15:0,
                 from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:9:
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/ESP32-RGB64x32MatrixPanel-I2S-DMA.h:2:17: note: #pragma message: Please update your include to use ESP32-HUB75-MatrixPanel-I2S-DMA.h instead!
 #pragma message "Please update your include to use ESP32-HUB75-MatrixPanel-I2S-DMA.h instead!"
                 ^
In file included from /home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:9:0:
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h:32:1: error: expected class-name before '{' token
 {
 ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h: In constructor 'Layer::Layer(RGB64x32MatrixPanel_I2S_DMA&)':
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.h:38:46: error: class 'Layer' does not have any field named 'GFX'
   Layer(RGB64x32MatrixPanel_I2S_DMA &disp) : GFX (LAYER_WIDTH, LAYER_HEIGHT) {
                                              ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp: In member function 'void Layer::drawCentreText(const char*, textPosition, const GFXfont*, CRGB, int)':
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:100:21: error: 'setTextWrap' was not declared in this scope
    setTextWrap(false);
                     ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:103:26: error: 'setFont' was not declared in this scope
      setFont((GFXfont *)f);
                          ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:105:14: error: 'setFont' was not declared in this scope
      setFont(); // use default
              ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:109:45: error: 'getTextBounds' was not declared in this scope
    getTextBounds(buf, 0, 0, &x1, &y1, &w, &h); //calc width of new string
                                             ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:126:40: error: 'setCursor' was not declared in this scope
      setCursor((LAYER_WIDTH - w) / 2, 0); // top 
                                        ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:128:55: error: 'setCursor' was not declared in this scope
      setCursor((LAYER_WIDTH - w) / 2, LAYER_HEIGHT - h);
                                                       ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:130:61: error: 'setCursor' was not declared in this scope
      setCursor((LAYER_WIDTH - w) / 2, (LAYER_HEIGHT - h) / 2); // top 
                                                             ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:142:33: error: 'setCursor' was not declared in this scope
      setCursor(wstart, h+yadjust); // top 
                                 ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:144:50: error: 'setCursor' was not declared in this scope
      setCursor(wstart+1, (LAYER_HEIGHT-1)+yadjust);
                                                  ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:146:61: error: 'setCursor' was not declared in this scope
      setCursor( wstart, ((LAYER_HEIGHT/2) + (h/2)) + yadjust);
                                                             ^
/home/dinux/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/Layer.cpp:153:58: error: 'setTextColor' was not declared in this scope
    setTextColor(this->color565(color.r, color.g, color.b)); // Need to confirm from FastLed CRGB to adafruit 565
                                                          ^
exit status 1
Error compiling for board ESP32 Dev Module.
```

What i'm doing here wrong? 
Should I install some additional libraries?

mrfaptastic, 2020-12-07:
I've nuked the Layer.cpp and Layer.h files... will move these to a seperate example.

Please delete these from your clone of the repository, or re-sync to repository.

closed 2020-12-07 by mrfaptastic

dinuxlt, 2020-12-08:
Thanks, it works now.