mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #47: Problem with chaining panels 2x2

nospamsam2011, 2020-12-16:
Good day and thank you in advance for such a great library and examples!

I had no issue with using the library with a single panel, as everything worked as expected.  However, if I chain up a total of 4 panels (each are 64x32 in size) I get the following error on the serial port:

ERROR: MATRIX_WIDTH and/or MATRIX_HEIGHT in 'ESP32-HUB75-MatrixPanel-I2S-DMA.h'
is not configured correctly for the requested VirtualMatrixPanel dimensions!

I am using the example ChainedPanels sketch with the added defines of 

#define MATRIX_WIDTH    128
#define MATRIX_HEIGHT   64

to represent the total pixels of the virtualized 4 panels.

The other defines are unchanged:

#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW

I am confused as to why the error message is generated, as my setup seems to match the examples listed in the comments.  Can anyone shed some light as to how I am misunderstanding or misconfiguring the library?

Thanks in advance!

Cheers,

Sam

mrfaptastic, 2020-12-16:
You need to edit the ESP32-HUB75-MatrixPanel-I2S-DMA.h directly and change the MATRIX_WIDTH and HEIGHT.

Adding defines in the .ino doesn't work.

mrfaptastic, 2020-12-16:
You need to edit the ESP32-HUB75-MatrixPanel-I2S-DMA.h directly and change the MATRIX_WIDTH and MATRIX_HEIGHT defines in place.

Adding new defines in the .ino or anywhere else doesn't. #define statements only have scope/visability within the file they exist.

mrfaptastic, 2020-12-16:
Also, just re-reading your requirement, you are chaining FOUR x 64px (width) * 32px (height) panels right?

So in 'ESP32-HUB75-MatrixPanel-I2S-DMA.h'

```
#ifndef MATRIX_WIDTH
 #define MATRIX_WIDTH                64*4   // CHANGE THIS VALUE IF CHAINING
#endif

#ifndef MATRIX_HEIGHT
 #define MATRIX_HEIGHT               32   // CHANGE THIS VALUE ONLY IF USING 64px HIGH panel with E PIN
#endif
```

These are fine as you had them:
```
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
```

nospamsam2011, 2020-12-16:
Good day,
Understood and thank you for the prompt response and the info!  I will modify as you suggest and will report back later with the results.
Thanks again!
Cheers,
Sam

nospamsam2011, 2020-12-17:
Good day Again,

Yup, that did the trick!  My apologies for being such a dolt!

Thanks again!

Cheers,

Sam

mrfaptastic, 2020-12-17:
No problems. Glad the library is of use.

hallard, 2021-02-01:
@nospamsam2011 , I had the same "question" took me some time to understand how to have 2 x 64x32 panels to build 1 of 64x64 and declare a looks like 128x32 to achieve that. It's not brain natural but once you understood the (IT) concept it's fine :-)

You can avoid modifying the lib source code to change size, because we must admit it's not very convenient especially if you use different code with different display size. Trick is to setup these `define` before `#include` statement (and **each** time you include the lib headers)

So something like that prevent modifying the lib source:

```cpp
#define MATRIX_WIDTH  	64   // Single panel of 64 pixel width
#define MATRIX_HEIGHT	64   // CHANGE THIS VALUE to 64 IF USING 64px HIGH panel(s) with E PIN

// Once constants defined, you can include the lib
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// These one are personal 
#if MATRIX_WIDTH==64 && MATRIX_HEIGHT==32 
#define DISPLAY_64x32
#elif MATRIX_WIDTH==64 && MATRIX_HEIGHT==64 
#define DISPLAY_64x64
#else
#endif

```

kj96, 2021-02-12:
Hi, I had no problem with sinngle 64x32 panel but when I'm trying to chain it into square, 64x64,  Vertical serprntine 'S' chain/stack. I only get both the scree running and mirroring with this definition values

`
#define PANEL_RES_X 64
#define PANEL_RES_Y 32

#define NUM_ROWS 2
#define NUM_COLS 1

#define MATRIX_WIDTH    64*2
#define MATRIX_HEIGHT   32
`

The output is like a mirror of another. 

What is wrong with these values or my connection. Please suggest.

kj96, 2021-02-12:
Hi, I had no problem with a single 64x32 panel but when I'm trying to chain it into a square, 64x64, Vertical serpentine 'S' chain/stack. I only get both the screen running and mirroring with this definition values

```#define PANEL_RES_X 64
#define PANEL_RES_Y 32

#define NUM_ROWS 2
#define NUM_COLS 1

#define MATRIX_WIDTH 64*2
#define MATRIX_HEIGHT 32``

The output is like a mirror of another.

What is wrong with these values or my connection. Please suggest.

mrfaptastic, 2021-02-13:
Start again using this example: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/examples/ChainedPanels/ChainedPanels.ino

GrimorJander, 2022-09-18:
> Comience de nuevo usando este ejemplo: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/examples/ChainedPanels/ChainedPanels.ino

I am facing the same problem trying to run the  gif animation skecht with 4 panels (64x32), chained 2x2 (serpent=true, topdown=false), making a total of 128x64 pixels.

I can run the Chainedpanels skecht correctly, but not port the chained to the GIF player example.

Do I need to modify the ESP32-HUB75-MatrixPanel-I2S-DMA.h file?

What definitions should I add or modify to the gifplayer .ino?

mrfaptastic, 2022-09-18:
@GrimorJander. Attach a zip of your sketch and I'll take a look

GrimorJander, 2022-09-18:
[SD_Card_AnimatedGIFPanel_4paneles.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/9593942/SD_Card_AnimatedGIFPanel_4paneles.zip)
Really apreciated your support

GrimorJander, 2022-09-18:
> @GrimorJander. Attach a zip of your sketch and I'll take a look

[SD_Card_AnimatedGIFPanel_4paneles.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/9593953/SD_Card_AnimatedGIFPanel_4paneles.zip)
Really apreciated your support

GrimorJander, 2022-09-21:
I have managed to chain the 4 panels and show the animated gifs with this attached code:
[SD_Card_AnimatedGIFPanel_4paneles.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/9619002/SD_Card_AnimatedGIFPanel_4paneles.zip)
But I am getting a horizontal band in the top third of the array where the colors are not displaying well. I don't know if I've messed up something by changing the "virtualDisp" or it's a connection error.
@mrfaptastic any clue to guide me?

mrfaptastic, 2022-09-22:
@GrimorJander - The library either works for all chained panels (if the same type) or doesn't at all. Any deviation from this would indeed mean that panel is broken and needs to be replaced.

GrimorJander, 2022-09-22:
Yes, that was my thought. The problem was on panel 3 & 4. First I think it was the data wire in between, and changing it the problem persist. Then I change position 3-4 of the panels, and problem was on the last defective panel. If I put it on position 1, the failure expand to all panels