mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #48: Cannot chaining panel 128x32

arinurcahyo, 2020-12-21:
Hi mrfaptastic, 
I used smartmatrix pinout and the display runs good using single panel.
However, the displays is always cloned when I used : 
` #define MATRIX_WIDTH 128
  #define MATRIX_HEIGHT 32`

I  run  example sketches and no modification other than MATRIX WIDTH 128.
Did I just go wrong?

[image]

mrfaptastic, 2020-12-21:
Hello. Impossible for me to say, the library works with chaining, so double check the wiring and that it's actually compiling with the amended library header .h file.

arinurcahyo, 2020-12-21:
Thank you.

So I also have to change MATRIX WIDTH value that lies in ESP32-RGB-64-32-MATRIX-PANEL-I2S-DMA.h and 2 chaining panels work well.

This is not an issue at all.

But can we just add #define MATRIX_WIDTH in sketch file and leave the .h library file untouched?

vortigont, 2020-12-22:
@arinurcahyo you can't use #defines in sketch to control library options since it a compile-time flag.
Although if you use platformio instead of Arduino IDE you can define "build_flags =  -DMATRIX_WIDTH=128"

arinurcahyo, 2020-12-22:
Thank you all.
This is due to my lack of knowledge.

closed 2020-12-22 by arinurcahyo