mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #49: I2S clock speed formula

vortigont, 2020-12-21:
@mrfaptastic I was playing with some i2s values and found some inconsistencies in code/values/formulas, whatever...
So, this code in esp32_i2s_parrallel.c sets divisors for clock generator. 
```
dev->clkm_conf.clkm_div_a=63;
dev->clkm_conf.clkm_div_b=63;
dev->clkm_conf.clkm_div_num=80000000L/(cfg->clkspeed_hz + 1);
```
To my great surprise I was able to run matrix at ESP32_I2S_CLOCK_SPEED=40000000. Hard to believe, eh, so I checked with scope what was going on at the clk line.

[image]

X-scale is 10ns per square. So it looks like exactly 20MHz from rising edge to rising edge.

More to this, according to the esp [docs](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#page=310) i2s clock is sourced from PLL_D2_CLK running at 160MHz which makes it even more mystery to me. Calculating the formula based on 160MHz source should give 80 MHz clk, not 40 and 20.
What do you think?

mrfaptastic, 2020-12-21:
Hi vortigont,

The esp32_i2s_parallel.c code I copied from the esp32 official forums years ago as the basis of this library and haven't really challenged it or bothered to read the ESP32 datasheet to see if it could be improved.

I'm confused as to what you are suggestion is changed? I would prefer to keep the default to 10Mhz as it is a good balance between brightness, memory consumption (due to color depth), and flicker.

If you're saying the actual calculation is wrong and when ESP32_I2S_CLOCK_SPEED=40Mhz, it doesn't actually output a 40mhz signal, then yeah, I agree - needs to be fixed. Does that mean the 10mhz default of the library is actually 5Mhz output clock?

vortigont, 2020-12-21:
No, I'm not about fixing something right away. Just researching how this is all works, for fun :) Thought you may have insite on this.
Default 10 is fine and it gives 10MHz rise to rise on the scope also. I've tried some other values too - it works more or less correct up to somewhere around 26.5 MHz (according to the scope). Curious. Was thinking about making clock selection a runtime tunable. Or just make a note in the lib about available steps. Not a hurry, I will do some other test also.

vortigont, 2020-12-21:
One more question - using -DPIXEL_COLOR_DEPTH_BITS=7 seriously degrades colours. Actually anything less than 8 makes any gradient horrible. Does not look like just one LSB reduction in colour. So is that OK? I mean it's not just my setup is wrong, is it?

BTW, have you tried github [discussions](https://github.com/github/docs/discussions)?
Might be useful for anything that is not a real issue actually :)

Cheers!

mrfaptastic, 2020-12-21:
> 
> 
> Default 10 is fine and it gives 10MHz rise to rise on the scope also. I've tried some other values too - it works more or less correct up to somewhere around 26.5 MHz (according to the scope). 

This is all a result of the black-magic that is the ESP32 SoC. Perhaps experiment with the values in esp32_i2s_parallel.c and see what happens.

```
 dev->sample_rate_conf.tx_bck_div_num=

dev->clkm_conf.clkm_div_num=80000000L/(cfg->clkspeed_hz + 1); 

etc...
```

Not really well documented this stuff is. Apparently when using LCD model (16bit parallel), the maximum frequency is 80Mhz. According to line 404 of ESP32-HUB75-MatrixPanel....cpp: `Acceptable values 26.67MHz, 20MHz, 16MHz, 13.34MHz...` so that seems in line. This is all ESP32 silicon stuff.

> One more question - using -DPIXEL_COLOR_DEPTH_BITS=7 seriously degrades colours.

I have never actually used the library with a PIXEL_COLOR_DEPTH_BITS less than 8. In fact, I don't think the library is coded to handle it.  Perhaps I should remove that configurable define from the library alltogether - pretty much useless.

> Might be useful for anything that is not a real issue actually :)

Thanks for the tip. I'll look into this  :-)

mrfaptastic, 2020-12-21:
> Was thinking about making clock selection a runtime tunable. 

To be honest, not sure it's worth the effort. You'd need to write a routine to restart the DMA engine etc.

vortigont, 2020-12-21:
> when using LCD model (16bit parallel), the maximum frequency is 80Mhz. According to line 404 of ESP32-HUB75-MatrixPanel....cpp: `Acceptable values 26.67MHz, 20MHz, 16MHz, 13.34MHz...` so that seems in line with your observations. Again, this is all ESP32 silicon stuff.

Oh, thanks! How could I have missed this!? It must be 'cause of is a 404 line :)))

 
> I have never actually used the library with a PIXEL_COLOR_DEPTH_BITS less than 8. In fact, I don't think the library is coded to handle it. Perhaps I should remove that configurable define from the library alltogether - pretty much useless.

Well, but the idea is great actually! DMA buffs are mem hogs, so loosing a couple of last colour bits looks promising. Let it be... maybe it just requires some more bit math, could worth the efforts.

> To be honest, not sure it's worth the effort. You'd need to write a routine to restart the DMA engine etc.
Yeah, you right... just for research not worth it. I've made a demo sketch to play with options like brightness/patterns via websocks UI to make testing more comfortable :)

vortigont, 2020-12-23:
so, it is all HW specific. Not an issue

mrfaptastic, 2021-02-15:
@vortigont - Can you do me a favour given you have a proper scope, and see what frequency you get when clkm_div_num = 1?

``    dev->clkm_conf.clkm_div_num=1; //80000000L/(cfg->clkspeed_hz + 1);``

vortigont, 2021-02-15:
Sure, it is 20MHz, the same for "2" and "3"

mrfaptastic, 2021-02-15:
Can you try again with this overwritten version of esp32_i2s_parallel?

I'm hoping you see 40mhz.

[esp32_i2s_parallel.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/5983277/esp32_i2s_parallel.zip)

mrfaptastic, 2021-02-15:
I've uploaded this into a new 'testing' branch. I think I managed to break my panel with a bitrate that was too high, but need to confirm.

Adjust the value `int sdm2 = 3;` value to 1,2,3,4,5,6 etc. Any higher than 7 and my ESP didn't boot ! ha!

vortigont, 2021-02-16:
So, here is what I was able to measure:
sdm = 3 17,5 MHz
sdm = 4 20 MHz
sdm = 5 22,5 MHz
sdm = 6 25 MHz
sdm = 7 27,5 MHz
sdm = 8 30 MHz
sdm = 9 32,5 MHz
sdm = 11 37,5 MHz
And by that frequency I mean full clock time from rising edge to rising edge.  Everything above 11 results in ESP boots but hangs on "Performing I2S setup:" My panels works only up to 20Mhz flawlessly. Starting from 22.5 it results in increased noise and distortions.

mrfaptastic, 2021-02-16:
> And by that frequency I mean full clock time from rising edge to rising edge. Everything above 11 results in ESP boots but hangs on "Performing I2S setup:" My panels works only up to 20Mhz flawlessly. Starting from 22.5 it results in increased noise and distortions.

Thank you @vortigont !! I'm going to assume the distortions are due to the panels being driven too fast opposed to the esp32 generating out-of-sync or garbled clock and data signals (due to the 'overclock'). If so then there's no much point in this overclocking activity. Between 10-20Mhz is therefore perfect.

vortigont, 2021-02-16:
yup, according to datasheet 6126 could be clocked up to 25 MHz, and I believe this could be done with 5 volts level logic drivers and some better cabling maybe. Agreed current 10-20 values is enough to choose from. For lesser clock rates better use some other lib with bit banging.
I was thinking if there are ribbon cables for HUB75 like a good-old IDE 80-pin's with every even wire grounded. :)

vortigont, 2021-02-16:
ugh... removed enum clk_speed control in master was a little bit unexpected :) Had some projects using it. Is that so bad to have this runtime control available for some of the 10-20 MHz values?

mrfaptastic, 2021-02-16:
I've reverted my change. I got a bit carried away :)

It's back.

vortigont, 2021-02-16:
Thanks! Already got used to it :)

mrfaptastic, 2021-02-17:
I've replaced the dma logic with a re-written cleaner version. When you get some spare time can you just check that 10Mhz and 20Mhz is actually sends out 10Mhz and 20Mhz?  Thanks again :)

mrfaptastic, 2021-02-17:
I've replaced the dma code with a cleaner version which fixes an issue with double buffer flipping. 

However, when you get some spare time can you just check that when the config is set to 10Mhz or 20Mhz it actually sends out 10Mhz and 20Mhz? Thanks again :)

vortigont, 2021-02-17:
Sure, I will check it and let you know if any issues with clock speed with a newer code

vortigont, 2021-02-19:
@mrfaptastic I've checked current driver from master with a scope and it looks like something is messed there. The default build clocks at about 8.8 MHz and if I set `mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;` it clocks at about 16 MHz.
Will you look into it? I believe It could be not that easy without a scope at hand.
Anyway let me know if you need to test something or try to play with that clocking

mrfaptastic, 2021-02-19:
Fixed. I did the 'shake the panel test' and it looks about 2-4mhz better. :-)

Ha!

closed 2021-02-19 by mrfaptastic

vortigont, 2021-02-19:
Ahah! Scopes are for weak ones :)))
Now it's fine! Thanks!

daveythacher, 2021-03-24:
Rail voltage of the buffer and driver matters for max clock. The impedance of the wire, grounding and PCB (all of them) matter. Cap load of the IO drivers can matter along with termination stuff. If you look at the Henner's "speed tests" you will notice that in some cases a square wave can become a sine wave. This is hell on inputs, and a function of some stuff mentioned above.

Likely in that range it should work. You may want to allow it down to 1-3 MHz. However that may not even be worth the time.

HUB75 is cost optimized, it is not electrically sound. Honestly I am shocked it works. My guess is they down clock internally and use a huge grounded metal enclosure. Maybe multilayer PCB with ground plane. That is how the fixed PATA till they moved to the 80 pin.

This frequency determines the max pixels you can drive at for a given quality. Even with a few fancy tricks applied. There is a number of ways to get around this outside the scope of this project.