mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #50: FM6126A - garbage after ESP reboot

vortigont, 2020-12-22:
Maybe, if someone else could confirm it, or not.
After successful FM6126A initialization it works fine. But if ESP reboots afterward than some part of the combined screen contains garbage. It is almost always NOT the first panel connected to CPU but all other in chain.
Looks like somehow a random values got itself locked in 6126 registers and usual LAT/OE signals can't write new data to it any more.
The only proven way is to power down both ESP and the whole panel than power up both at the same time.
Not sure how to fix this since specs for 6126 are scarce. Need more experiments with reg vals.

mrfaptastic, 2020-12-22:
Hi vortigont,

This is annoying. Seemingly when restarting the ESP32, the output from the GPIO's is indeterminate which must screw up the registers as you say.

A poor copy paste into Google Translate reveals there's a reset mode for the panels. I guess will need to expand the shiftDriver function to do this - but how it is done... no idea, I don't have one of these panels yet to trial-and-error a solution.

Two clock pulses whilse OE is high = RESET? Not sure.

```
Instruction definition
The chip contains a simple 16bit shift register, the gray value and configuration value will be latched into the shift register. By counting the LE signal
The length is used to analyze the control commands, and different LE lengths represent different commands. For example, an LE signal with a length of 3 represents the "Data_Latch" command, use
To control the shift register to latch the gray value, and send the 16bit data in the shift register to the output channel. The following table lists all commands and their definitions.
Note: The length of LE refers to the number of rising edges of CLK when LE is high. As shown in the figure below, the length of the first LE signal is 3.
Let it be the "Data_Latch" command.
Build hold time
The setup and hold time of LE signal and SIN signal are shown in the following table.
Signal name MIN Remark
TSU_LE 7ns
THD_LE 7ns
TSW_LE 10ns
TSU_SIN 3ns
THD_SIN 3ns
Command name LE command description
RESET_OEN >1 soft reset signal
DATA_LATCH 3 Latch 16bit data to the output channel
``

mrfaptastic, 2020-12-22:
Attempt:

```
      // Reset Matrix
      for (int l = 0; l < MATRIX_WIDTH; l++){
		  
          if (l > MATRIX_WIDTH - 3){         // pull the latch 11 clocks before the end of matrix so that REG1 starts counting to save the value
              digitalWrite(dma_lat_pin, HIGH);
          }
          digitalWrite(dma_clk_pin, HIGH);    // 1-clock pulse
          digitalWrite(dma_clk_pin, LOW);
      }

		// Drop Latch and reset
      digitalWrite(dma_lat_pin, LOW);
      digitalWrite(dma_clk_pin, LOW);

```

mrfaptastic, 2020-12-22:
Blind attempt at code:

```
      // Reset Matrix
      for (int l = 0; l < MATRIX_WIDTH; l++){
		  
          if (l > MATRIX_WIDTH - 3){         // pull the latch 2 clocks before the end of matrix so that SOFT_RESET occurs
              digitalWrite(dma_lat_pin, HIGH);
          }
          digitalWrite(dma_clk_pin, HIGH);    // 1-clock pulse
          digitalWrite(dma_clk_pin, LOW);
      }

		// Drop Latch and reset
      digitalWrite(dma_lat_pin, LOW);
      digitalWrite(dma_clk_pin, LOW);

```

vortigont, 2020-12-22:
OK, will try this, thanks.
Actually that is itself interesting. So there is some hidden way to lock a particular data in REG from being modified even while proper LAT/OE signals are in-place.

mrfaptastic, 2020-12-22:
The datasheet seems to imply that sending three high latches after three clocks puts the panels into DATA_LATCH mode which then treats all input as for the pixels only.

So perhaps at the end of your shiftDriver function you need this at the end:
```
      // Reset Matrix
      for (int l = 0; l < MATRIX_WIDTH; l++){
		  
          if (l > MATRIX_WIDTH - 4){         // pull the latch 3 clocks before the end of matrix so that DATA_LATCH mode enabled
              digitalWrite(dma_lat_pin, HIGH);
          }
          digitalWrite(dma_clk_pin, HIGH);    // 1-clock pulse
          digitalWrite(dma_clk_pin, LOW);
      }

		// Drop Latch and reset
      digitalWrite(dma_lat_pin, LOW);
      digitalWrite(dma_clk_pin, LOW);
``

vortigont, 2020-12-22:
No, this doesn't work :( there must be something more. I found that flipping specific bits in REG2 leads to some test-like patterns on matrix and the data input is blocked when those stripes are displayed. More and more mysteries )

mrfaptastic, 2020-12-22:
Actually, the latch and clock pin goes to all the chips right, so couldn't we do this for reset? We're not transferring 'data' unlike the REG1/2 stuff.

```
      // Reset Matrix
      digitalWrite(dma_lat_pin, LOW);
      digitalWrite(dma_clk_pin, LOW);

                digitalWrite(dma_lat_pin, HIGH);

               digitalWrite(dma_clk_pin, HIGH);    // 1-clock pulse
               digitalWrite(dma_clk_pin, LOW);

               digitalWrite(dma_clk_pin, HIGH);    // 1-clock pulse
               digitalWrite(dma_clk_pin, LOW);

	// Drop Latch and reset ?
      digitalWrite(dma_lat_pin, LOW);
      digitalWrite(dma_clk_pin, LOW);
```

vortigont, 2020-12-22:
Will try that too, thanks! reflashing esp for every change is so frustrating. :)

hallard, 2021-02-18:
Guys,

Tried this piece of code also, does not work either on reboot, I'm facing same issue. 
But I can test easily so don't hesitate do ask.

https://user-images.githubusercontent.com/2471931/108402878-5f656d00-721e-11eb-904a-1e66ec5665eb.mov

vortigont, 2021-02-20:
Nice clock indeed :) No, I was unable to unlock all the secrets of 6126A (yet). So how do you reset it than? By powercycling? In the video it seems pretty good to me, just a short flicker and than it's OK.

hallard, 2021-02-20:
Thanks,

I saw afterward that something was existing in current code like `clearScreen`, and seems to work better to call this just before reseting like following code (but of course on reset button or power recycle, this does not works) but with OTA it's far much better.

```
    display->clearScreen();
    // Just to be sure all work is done
    delay(50);
    // Finally reset device
    ESP.restart();
```

mrfaptastic, 2021-02-20:
Hi @hallard @vortigont, so it's a case of simply performing a clearScreen prior to reset to avoid visual artifacts? I couldn't see anything wrong in your video @hallard.

On another note, how does on calculate the % complete of an OTA update in progress? I like the graphics.

hallard, 2021-02-20:
@mrfaptastic 
Check at 0:08 from 0:09 the green line due to reseting the board, but as soon as I'm clearing screen before it's ok now.

hallard, 2021-02-20:
Here my simple OTA code

```cpp

void OTAStart() {
#ifdef USE_MQTT
  if ( mqttClient.connected() ) {
    mqttClient.disconnect();
  }
#endif // MQTT
  display->clearScreen();   // turn off display during update
  setTextColor(colorBrightness(CRGB::Magenta));
  centerText("Firmware Update", row1, false);
  OUT.println(F("\nOTA: Start"));
}

void OTAProgress(int percent) {
  char line[32];
  sprintf_P(line, PSTR("Progress : %3d%%\r"), percent);
  setTextColor(CRGB::Green);
  centerText(line, row3+2, true);
  OUT.print(line);
  //display->fillRect(0, row4 -7, 64, 8, myBLACK);
  drawProgressBar(5, row2-4, 64-(5*2)+1, 6, percent , CRGB::Gray, CRGB::Gray);
}

void OTAEnd() {
  OUT.println(F("\nOTA: End"));
  setTextColor(CRGB::Green);
  centerText("Done", row3, false);
  OUT.flush();
  display->clearScreen();
  delay(100);
  ESP.restart();
}

```

in `setup()`
```cpp
  ArduinoOTA.setHostname(HOST);
  ArduinoOTA.onStart(OTAStart);
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    int percent = progress / (total / 100);
    OTAProgress(percent);
  });
  ArduinoOTA.onEnd(OTAEnd);
  ArduinoOTA.onError([](ota_error_t error) {
    OUT.print(PSTR("\nError[") + String(error) + PSTR("]: "));
    if (error == OTA_AUTH_ERROR) OUT.println(F("Auth Failed"));
    else if (error == OTA_BEGIN_ERROR) OUT.println(F("Begin Failed"));
    else if (error == OTA_CONNECT_ERROR) OUT.println(F("Connect Failed"));
    else if (error == OTA_RECEIVE_ERROR) OUT.println(F("Receive Failed"));
    else if (error == OTA_END_ERROR) OUT.println(F("End Failed"));
    else OUT.println(F("unknown error"));
    setTextColor(colorBrightness(CRGB::Red));
    centerText("Error", row3, false);
    delay(200);
    display->clearScreen();
    ESP.restart();
  });
  
  ArduinoOTA.begin();

```

mrfaptastic, 2021-02-20:
> @mrfaptastic
> Check at 0:08 from 0:09 the green line due to reseting the board, but as soon as I'm clearing screen before it's ok now.

I think the solution for your problem is to manually stop ESP 32 DMA transfers *after* you have cleared the screen (zero'd out the DMA buffer) as this really isn't an issue with this library because when ESP.restart() is called it's anyone's guess what the SoC will transmit - library has absolutely no control of whatsoever. However, turning off the DMA hardware engine before a reboot seems (from my 5 seconds of testing) to stop random pixel noise.

Also, if you have any Serial.begin stuff, do this AFTER you do the .begin() for this library, as I have noticed that the ESP32 can sometimes send random serial output garbage to GPIO pins that may be connected to the HUB75, before this library has a chance to 'take control' of them.

```
        // Stop all ongoing DMA operations
        i2s_dev_t *dev = &I2S1;

        dma_display.clearScreen();

        dev->out_link.stop = 1;
        dev->out_link.start = 0;
        dev->conf.tx_start = 0;

        delay(1000);
  
        Serial.println("Reset..");
        ESP.restart();
```

Thanks for the OTA code!

mrfaptastic, 2021-02-20:
FYI @hallard , can you test your sketch with the 'clkphase' branch of the library?

I've added a function to do what I outlined above, in one call. 

```
        // Stop all ongoing DMA operations
        dma_display.stopDMAoutput();
        
        delay(1000);
  
        Serial.println("Reset..");
        ESP.restart();
```

hallard, 2021-02-22:
Just tested, this does works :-)

closed 2021-06-21 by mrfaptastic