mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #54: 64x64 Matrix not showing anything, when used as 64x32 works (partly)

jonasf21, 2020-12-26:
Hello,

ive got a P2 64x64 (1/32 scan) module that works fine with the PxMatrix library. Now with the DMA library i have a hard time getting it working correctly. The aurora demo works only when i define the panel to be a 64x32, then it will run the demo but the picture has "gaps" as x32 is wrong of course. If i configure it to be 64x64 the screen stays blank, console output is working. I assume this has to do with the E pin (which is working fine with PxMatrix) but not DMA. I tried relocating it to a different pin, same effect. 
Not sure if this is related: with PxMatrix i had to set a muxDelay for the B channel, otherwise i would get gaps in the image.

What could be wrong or how can i debug this further?

mrfaptastic, 2020-12-26:
Hello. A quick Google revealed this issue, is it the same, do you have the
same chips?

https://github.com/2dom/PxMatrix/issues/104

This library doesn't have the ability to arbitrarily delay outputs as it
doesn't manually toggle gpios. The only work around I can think of is to
reduce the DMA frequency to something less than 10mhz (8, 4mhz) if its due
to chips with issues with speed.

jonasf21, 2020-12-26:
The chips are partly the same. I have the RUC7258 and 74HC245C too but a ICN2037BP (instead of DP5020B from that thread).

I have just tested this via
#define ESP32_I2S_CLOCK_SPEED
with 8, 4, 1Mhz as well as 500k and 100k. None works.

I also somehow now doubt that this is a delay issue because with PxMatrix default settings and 64x64 i atleast got a image to display (with gaps). With your library it simply doesnt display anything at all when set to 64x64. With 64x32 it works (with gaps, as its the wrong resolution of course). So i somewhat think its a different problem.

mrfaptastic, 2020-12-27:
Can you try again with the earliest version (1.0.2) and see if that works?

jonasf21, 2020-12-27:
Just tested the old releases:
1.0.2: seems to not honor 64x64, instead will always display 64x32
1.1.0: same
1.2.2: 64x64: does not work, displays nothing
1.2.3: 64x64: does not work, displays nothing

Is there a known list of 64x64 displays/chipsets that were tested successfully?

mrfaptastic, 2020-12-27:
There isn't I'm afraid. This library has historically been focussed on 64x32 px panels. Ironically, FM6162A panels apparently work OK now after years of issues.

This seems like some obsure hardware bug with those panels which only trial and error with the hardware could fix.

mrfaptastic, 2020-12-27:
There isn't I'm afraid. This library has historically been focussed on 64x32 px panels. Ironically, FM6162A 64x64 panels apparently work OK now after years of issues.

This seems like some obsure hardware bug with those chips which only trial and error with the panels can find the fix. I would triple check the wiring because this seems like some weird addressing issue. Versions 1.0.2-1.1.0 of this library do honour 64x64 with other panels I believe.

mrfaptastic, 2020-12-27:
Can you post some photos and videos of what you're seeing?

csloz, 2020-12-27:
I’ve used / been using P2.5 64 x 64 since the early days of the library.
Haven’t tested recent versions as I don’t have my equipment handy, but haven’t had issues previously.

Taking a look at 2 panels I have here  that work, these have the following chipsets:

SunMoon - SM16106SC 
CHIPONE-  ICND2038

I made my own pcb to connect to the boards.

jonasf21, 2020-12-27:
To outrule any cableing issue i completly changed it (i was using different/custom pins before) and now changed it to the library's default (except E, as it has no default). Same issue.

Meanwhile i've done some more testing: If i configure the panel to correct 64x64 it stays dark (as said before) but as soon as i disconnect the OE pin (i.e. leave it floating) the panel will display the demo however in 64x32 and with heavy flickering.

Then i configured it with wrong 64x32 and ran the demo and disconnected the OE on the run: display works as before but gets brighter (no flickering in anyway).
Even starting the ESP32 with OE disconnected does the same. Isnt OE required to turn the display on or is it just for brightness controll? Ive found conflicting reports on that via google.

My display has no pinout label on the back however it has a writing "DCHY-P2-6464-1515-VP" and someone has found/created the schematric here:
https://raw.githubusercontent.com/esden/led-panel-docs/master/DCHY-P2-6464-1515-VP/led-panel-DCHY-P2-644-1515-VP-sch.pdf

@csloz thanks for confirmation. Your chips are different, so might still be a chips issue...still wonder why the work fine with PxMatrix.

mrfaptastic, 2020-12-27:
I have no idea I'm afraid. Yet another chinese chip variant that tries to to the same thing yet has its own issues.

vortigont, 2020-12-27:
@jonasf21 you may also play around with 'brightness' value, 'cause it affects toggling OE pin. Try to set "64" for example.

Max-X3, 2021-01-10:
@jonasf21: Maybe worth a try, i experienced similar issues with my 64x64 panels with TC7258GN and ICN2037B chips. I worked on the AuroraDemo adjusting E_PIN for my panel and "#define MATRIX_HEIGHT 64" in AuroraDemo.ino but had no luck.

It looked like MATRIX_HEIGHT have been partly overwritten in the sourcefiles (especially ESP32-HUB75-MatrixPanel-I2S-DMA.cpp). After i changed the "#define MATRIX_HEIGHT 64" in ESP32-HUB75-MatrixPanel-I2S-DMA.h in line 33 too everything worked ...

mrfaptastic, 2021-01-10:
@Max-X3. I've updated the examples in the git to remove this confusion. Indeed you are correct, #define's in the .ino files are not 'propogated' to this library.

closed 2021-01-10 by mrfaptastic

jonasf21, 2021-01-11:
@Max-X3 you are absolutly right, that was the issue, now its working! Thanks alot. Unfortunately i didnt had time the past days to do further testing. 
So after all my first impression (bug in the library/examples) prevailed. Anyway, thanks for everyone helping!