mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #56: Any issue with AsyncWebServer and AsyncMQTT

hallard, 2021-01-08:
Until there I'm using PxMatrix, even if it's working fine, when added heavy stuff such as ayncWebServer or AsyncMQTT there are some issues (ESP32 crash), I'm thinking moving to this DMA version, but has anyone experienced better results and no issues with MQTT and or WebServer implemented on the same time ?

Thanks

mrfaptastic, 2021-01-08:
Shouldn't be any issues.

closed 2021-01-10 by mrfaptastic

hallard, 2021-02-01:
Confirmed, compiled, uploaded, works fine, no crash no flicker, with both combined async libraries even with OTA

Thanks for this awesome lib.

mrfaptastic, 2021-02-01:
Hi @hallard. Thanks for providing follow-up feedback!

Any chance you could also test with the dev branch as well? We're going to roll this out at some point in the future.

hallard, 2021-02-01:
Ha ha didn't even took time to see existence of this branch, for sure, I will use it instead.

hallard, 2021-02-05:
Hi @mrfaptastic,

just tested the dev branch, so far, so good on my side.

Just one thing, looks like missing `setfastmode()` on dev branch :-).

vortigont, 2021-02-06:
@hallard, setfastmode() is deprecated in dev, it is always 'fast' now :)