mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #57: Using a 16x32 Display - possible?

mrRobot62, 2021-01-08:
I have a 16x32 HUB75 Display running.
If I use the rainbow example (with original MATIRX_HEIGHT 32, MATRIX_WIDTH 64 and I compile the file - it works, I will see fancy color shades on the display

If I try to work with printing text - it didn't work, can't read any characters :-O

Change (hardcoded) inside EPS32_HUBxxxx .h file height/width to 16 and 32. It make no real difference not readable.

What can I do to use such a 16x32 display?

mrfaptastic, 2021-01-09:
Hello! Unfortunately 1/8 and 1/4 scan (yours) panels aren't supported by
this library. Not to say that couldn't, but I don't have the panels to test.

I see a similar issue happened with another library and that got around it
by sending the library 'fudged' coordinates. Not sure if you could use any
code referred to in this issue.

https://github.com/hzeller/rpi-rgb-led-matrix/issues/529

mrRobot62, 2021-01-09:
I made a little progress with my 16x32 display. Unfortunately I tried to use PxMatrix without success.
But with ESP32-HUB75-MatrixPanel I made today a little step forward :-)

What I did:
for-x-loop starting fo
r x up from 32 to 64
for-y-loop starting for y from 0 to 32

Problem:
drawing row 1, than row 2, than repeat row 1 and row 2
than 6 + 7 and repeat
than 3 + 4 and repeat
than 8 + 9 a.s.o

good visible on the video

https://user-images.githubusercontent.com/949032/104105118-622e8300-52ac-11eb-9034-dfd2331d1e46.mp4

Any ideas?

P.S.
I tried to change h-file with width/height to 32 and 16 and played around with MATRIX_ROWS_IN_PARALLEL without success.

Maybe the video is helpful to fix this (I hope small issue)

cheeers
Bernd

mrfaptastic, 2021-01-09:
Very usefull video. Can you post the code of your pin mapping, and a photo of the HUB75 connector? What HUB75 pin is connected to what ?

mrfaptastic, 2021-01-09:
Very usefull video. Can you post the code of your pin mapping, and a photo of the HUB75 connector? What HUB75 pin is connected to what ?

Also what happens if you do :

for-x-loop starting for x from 0 to 16
for-y-loop starting for y from 0 to 32 

Please post video of this as well.

mrfaptastic, 2021-01-09:
Also, have you taken a look at the SmartMatrix library for ESP32? That could also be an option.

mrfaptastic, 2021-01-09:
Finally, I think the dirty and easy solution would be to simply create a remap_y_function
```
int remap_y(int y) // y coord starts at 0
{
case 0:
case 1:
  return y;
case 2: 
case 3:
  return y+8;
case 4:
case 5:
etc....
}
```

bit of trial and error required

mrRobot62, 2021-01-09:
will do this tomorrow - late in Germany :-O

Your quick and dirty - honestly that was my thought too :-O

Videos tomorrow

cheers
Bernd

mrRobot62, 2021-01-10:
**OMG**
seems that I crashed my matrix display :-(
Nothing works - I installed libs new, but it totally dark - ah shit

I double check connectors but nothing helps. Don't know what I did to distroy the board

coming back if I have a new one (I ordered yesterday a 64x32 board)

mrRobot62, 2021-01-10:
update:
seems my esp32 wroom crashed - exchange to another one - works :-)

ok
for loop for x from 0-32 - black screen
for loop for y from 0-32 that's shown in my last video

I implemented a remap_y function (quick & dirty). Result is shown in below video
But this can't be a real solution, because I want to show text or draw lines. Ergo a remapping should be done inside library, right?

https://user-images.githubusercontent.com/949032/104129128-869d6480-536b-11eb-850d-598586ca9085.mp4

**Simple-dimpel source code**

[main.cpp.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/5792944/main.cpp.zip)

mrfaptastic, 2021-01-10:
> But this can't be a real solution, because I want to show text or draw lines. Ergo a remapping should be done inside library, right?

Look at the VirtualDisplay class and create something NEW for your 1/4 scan panels (feel free to then contribute to this project):

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/ESP32-VirtualMatrixPanel-I2S-DMA.h

Essentially, create a C++ class that inherits this library class, but then re-maps pixels as it sees fit. Then instead of calling ESP32-HUB75.... (which is named 'dma_display' in the examples), call your own custom class, that inherits this library, but with the y-coord re-mapped.

mrRobot62, 2021-01-10:
good hint @mrfaptastic ,

my first step was to patch the master class 
`    MatrixPanel_I2S_DMA(bool _double_buffer = false, bool _is16x32 = false)
#ifdef USE_GFX_ROOT	
      : GFX(MATRIX_WIDTH, MATRIX_HEIGHT), double_buffering_enabled(_double_buffer), use16x32(_is16x32)  {
#else
      : Adafruit_GFX(MATRIX_WIDTH, MATRIX_HEIGHT), double_buffering_enabled(_double_buffer), use16x32(_is16x32)  {
#endif		  
    }
`and of course implement the remapping inside drawPixelxxxxxx() functions.

work 

But will reimplement it in that way you suggested ! - it's a long time ago that I wrote c/c++ code :-O

mrfaptastic, 2021-01-10:
Here's a possible class that you could use. Totally untested by me.

```
#ifndef _ESP32_4SCAN_PANEL_I2S_DMA
#define _ESP32_4SCAN_PANEL_I2S_DMA

#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

struct VirtualCoords {
  int16_t x;
  int16_t y;
};
 
class QuarterScanMatrixPanel : public Adafruit_GFX
{

  public:

    MatrixPanel_I2S_DMA *display;

    QuarterScanMatrixPanel(MatrixPanel_I2S_DMA &disp)  Adafruit_GFX(32, 16)
    {
      this->display = &disp;

    }

    VirtualCoords getCoords(int16_t x, int16_t y);

    // equivalent methods of the matrix library so it can be just swapped out.
    virtual void drawPixel(int16_t x, int16_t y, uint16_t color);
    virtual void fillScreen(uint16_t color); // overwrite adafruit implementation
    void clearScreen() { fillScreen(0); }
    void drawPixelRGB565(int16_t x, int16_t y, uint16_t color);
    void drawPixelRGB888(int16_t x, int16_t y, uint8_t r, uint8_t g, uint8_t b);
    void drawPixelRGB24(int16_t x, int16_t y, RGB24 color);
    void drawIcon (int *ico, int16_t x, int16_t y, int16_t module_cols, int16_t module_rows);

    uint16_t color444(uint8_t r, uint8_t g, uint8_t b) {
      return display->color444(r, g, b);
    }
    uint16_t color565(uint8_t r, uint8_t g, uint8_t b) {
      return display->color565(r, g, b);
    }
    uint16_t color333(uint8_t r, uint8_t g, uint8_t b) {
      return display->color333(r, g, b);
    }
	
	void flipDMABuffer() { display->flipDMABuffer(); }
	void showDMABuffer() { display->showDMABuffer(); }

    void drawDisplayTest();

  private:
    VirtualCoords coords;

}; // end Class header

inline VirtualCoords QuarterScanMatrixPanel::getCoords(int16_t x, int16_t y) {

  int y_remap[] = { 0,1,8,9,4,5,12,13,16,17,24,25,22,23,30,31 };

  coords.x = x;
  coords.y = y_remap[y];
  
  return coords;  

}

inline void QuarterScanMatrixPanel::drawPixel(int16_t x, int16_t y, uint16_t color)
{
  VirtualCoords coords = getCoords(x, y);
  this->display->drawPixel(coords.x, coords.y, color);
}

inline void QuarterScanMatrixPanel::fillScreen(uint16_t color)  // adafruit virtual void override
{
  // No need to map this.
  this->display->fillScreen(color);
}

inline void QuarterScanMatrixPanel::drawPixelRGB565(int16_t x, int16_t y, uint16_t color)
{
  VirtualCoords coords = getCoords(x, y);
  this->display->drawPixelRGB565( coords.x, coords.y, color);
}

inline void QuarterScanMatrixPanel::drawPixelRGB888(int16_t x, int16_t y, uint8_t r, uint8_t g, uint8_t b)
{
  VirtualCoords coords = getCoords(x, y);
  this->display->drawPixelRGB888( coords.x, coords.y, r, g, b);
}

inline void QuarterScanMatrixPanel::drawPixelRGB24(int16_t x, int16_t y, RGB24 color)
{
  VirtualCoords coords = getCoords(x, y);
  this->display->drawPixelRGB24(coords.x, coords.y, color);
}

// need to recreate this one, as it wouldnt work to just map where it starts.
inline void QuarterScanMatrixPanel::drawIcon (int *ico, int16_t x, int16_t y, int16_t module_cols, int16_t module_rows) { }

#endif
```

Then in the ino of your project include like the 'VirtualMatrixPanel' example: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/examples/ChainedPanels/ChainedPanels.ino

and in your project, something like

```
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
MatrixPanel_I2S_DMA dma_display;
QuarterScanMatrixPanel qscan_display(dma_display);

in Serup:

dma_display.being() etc...

Then,

   qscan_display.setFont(&FreeSansBold12pt7b);
   qscan_display.setTextColor(virtualDisp.color565(0, 0, 255));
   qscan_display.setTextSize(2); 
   qscan_display.setCursor(10, virtualDisp.height()-20); 

```

etc.  :-)

mrRobot62, 2021-01-12:
Below first impression of the new virtual 16x32 panel

https://user-images.githubusercontent.com/949032/104366370-8fcc3400-5519-11eb-9842-389f6afcd58d.mp4

I will implement for function - next step is to draw text. I forked this git and will contribute the new panels next days.

As attachment : virtual panel and a quick & dirty demo
ESP_xxxxx header file should be saved in the same folder as the library
main.cpp is and example file from this video

[Archiv.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/5804487/Archiv.zip)

mrfaptastic, 2021-01-12:
Very good work! The quick & dirty solution is usually the best value, 80/20 rule. Seems to work perfectly.

mrRobot62, 2021-01-17:
YES - it works, upload int my repo next days

https://user-images.githubusercontent.com/949032/104838449-4aae5600-58bb-11eb-802f-a358b49a9315.mp4

WattsC-90, 2021-03-31:
did you get this confirmed working? I have tried to use the code from both repos but get the following: 
I can only assume the adafruit libs have been updated? I would welcome some help/feedback on this please?

[image]

csloz, 2021-03-31:
You’re both getting the same error about RGB24 not being defined.

Can you try add this define at the top above the library defines, see if it solves it for you?

#define COLOR_DEPTH 24

WattsC-90, 2021-03-31:
@csloz sadly that did nothing same errors, RGB24 not defined :(

mrRobot62, 2021-03-31:
Hi, strange, yes on my machine this lib works well.
RGB24 is a typten located in #include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
and should be a part of the QuarterScanMatrixPanel.

Could u please check this

WattsC-90, 2021-03-31:
[image]
proud to be different lol

WattsC-90, 2021-03-31:
I am including the lib in platformio through the git repo, i think that brings in the bleeding edge.. should i be using a different lib_dep reference?

WattsC-90, 2021-03-31:
Aha! that is indeed what it is, using the git repo method, the struct has been removed in the latest code.. using the mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display reference, it has pulled the last good release, and has made that error disappear, now it looks like it just needs an adafruit SPI lib!

mrRobot62, 2021-03-31:
on my machine I use
Adafruit BusIO
Adafruit GFX Library
ESP Hub75......

I'm not on my dev machine - sorry - above dependencies are "out of my mind" :-O

mrfaptastic, 2021-03-31:
To make it clear.

I broke something with my last release. It's not an issue with your machine.

Use the latest verison in the repository. Should be fixed.

M10CUBE, 2021-08-06:
Hi everybody. I jumped here and enjoying this excellent work done because I am trying to make to work an old 16X32 RGB Matrix I have (while waiting for 32X64 t come).
Sorry I I hijack this thread but generally speaking which example must follow for printing text or values in any part of the display? I am a bit confused.That is text, coordinates, font size. It is for my M10CUBE Sensor project I am building.
https://hackaday.io/project/171770-m10cube
Thanks all in adavance

DANILmaster9790, 2024-02-28:
> Ниже первое впечатление от новой виртуальной панели 16x32.
> 
>  IMG_4436_2.mp4 
> Я реализую функцию — следующий шаг — нарисовать текст. Я создал форк этого git и в ближайшие дни представлю новые панели.
> 
> Во вложении: виртуальная панель и заголовочный файл ESP_xxxxx с быстрой и грязной демонстрацией должны быть сохранены в той же папке, что и библиотека. main.cpp — это пример файла из этого видео.
> 
> [Архив.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/5804487/Archiv.zip)

Hello! Help me how to make a code for text output in the form of a running line on the P5 64x32 panels. The text is output, but the text in the form of a running line is not

board707, 2024-02-28:
> how to make a code for text output in the form of a running line on the P5 64x32 panels.

"P5" says nothing about panel pattern. What is the scan factor of your 64x32 matrix? Most common 64x32 16scan panels are supported without any modifications of the library code.

DANILmaster9790, 2024-02-28:
> > how to make a code for text output in the form of a running line on the P5 64x32 panels.
> 
> "P5" says nothing about panel pattern. What is the scan factor of your 64x32 matrix? Most common 64x32 16scan panels are supported without any modifications of the library code.

P5 64x32pix, 1/16scan, smd 2121. I just can't understand the very principle of scrolling text output.

board707, 2024-02-28:
> I just can't understand the very principle of scrolling text output.

Print the still text and than move it by one pixel left or right with defined intervals

DANILmaster9790, 2024-02-29:
> > P5 64x32пикс, 1/16скан,
> 
> Хорошая новость, панель 64х32 s16 должна работать без какой-либо специальной настройки.
> 
> > Я просто не могу понять сам принцип вывода текста с прокруткой.
> 
> Распечатайте неподвижный текст и переместите его на один пиксель влево или вправо с заданными интервалами.

[Line 156.txt](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/files/14445963/Line.156.txt)

DANILmaster9790, 2024-02-29:
> What is your question about the code?

so that these 20 news headlines are alternately displayed in running text on the screen