mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #58: Render 64x128 in 4 panel

tsctrl, 2021-01-10:
Hi @mrfaptastic ,

i did some test by chaining 2x1 panel to display 128(W)x32(H) by changing MATRIX_WIDTH to 128 and it render the gifs without any issue. i am planning to create a 128(W)x64(H) by chaining 4 panel in 2x2 configuration. is it possible to render gif using 2x2 configuration?
if i set the MATRIX_WIDTH to 256. It will be define by the library as 4(W)x1(H) and will not able to display in 2x2 configuration right?. Any specific changes that i could do to achieve 2x2 configuration?

Thank you.

mrfaptastic, 2021-01-10:
Easily done but you'll need to write custom code. Take a look at the
virtual display example, and rework the animated gif example to output to
the virtual display class.

>
>

tsctrl, 2021-01-11:
ok, thank you @mrfaptastic

closed 2021-01-11 by tsctrl