mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #62: Color change in between two panels

reto88, 2021-01-13:
I have two panels 64x32 panels chained with default pins and an esp32. I can control everything. But the color changed in the middle of each panel (from red to green).(see diagonal Line)

[image]

 Do anyone now what the problem is?

Settings:
```
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

/**************************
 * Create physical DMA panel class AND virtual (chained) display class.   
 **************************/
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
MatrixPanel_I2S_DMA dma_display;
VirtualMatrixPanel       virtualDisp(dma_display, NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, true, true);
```

Connection ESP<-> Panel
 HUB 75 PANEL              ESP 32 PIN
```
+-----------+   
|  R1   G1  |    R1  -> IO25      G1  -> IO26
|  B1   GND |    B1  -> IO27
|  R2   G2  |    R2  -> IO14      G2  -> IO12
|  B2   E   |    B2  -> IO13      E   -> N/A	(required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32 )
|   A   B   |    A   -> IO23      B   -> IO19
|   C   D   |    C   -> IO 5      D   -> IO17
| CLK   LAT |    CLK -> IO16      LAT -> IO 4
|  OE   GND |    OE  -> IO15      GND -> ESP32 GND
+-----------+
```

mrfaptastic, 2021-01-13:
Can you try again with an earlier release of the library ?

(Earlier versions may not have the virtual display class, so you might need to copy this across)

mrfaptastic, 2021-01-13:
Actually, check your pin wiring. I think you have G2 and R2 wited the wrong way around.

mrfaptastic, 2021-01-13:
Check your wiring.

reto88, 2021-01-13:
thanks for the reply. I switched it and it works. now it is R2 ->IO12 and G2 ->IO14. . May you can change it in the Wiring **ESP32 with the LED Matrix Panel**  Indeed it it is fantastic library.

reto88, 2021-01-13:
Whats the method to change the brightness of the virtual display?

mrfaptastic, 2021-01-13:
VirtualDisplay class is just a pixel mapping class.

To adjust the brightness call the underlying hardware object dma_display.setBrightness(XX)

reto88, 2021-01-13:
ok, thanks, and is it possible to change the brightness of each pixel individual?

mrfaptastic, 2021-01-18:
You can do that by scaling the value of each pixels RGB value accordingly. i.e. drawPixel(1,1, matrix.color565(255,0,0)) (bright red) vs. drawPixel(1,1, matrix.color565(64,0,0)) (not so bright red)

vitalikdm, 2021-04-26:
Hello @mrfaptastic. Thank you for the great library that really helps and inspires.

I have a quite different problem with colours but decided not to create a new issue. 
I have created a project that is based on this library and tested it with 2 64x64 panels with FM6126A chip, everything works fine. Recently I ordered the same LED panel at the same seller but I hot panels with FM6126B chip. They also work but I get Green colours instead of Blue and vice versa. I can rewrite all entry where colours are used in my project to swap blue and green but it won't work with Gifs and other places. 

I double-checked wires connections - everything connected properly. 
Could you please help solve that issue?

mrfaptastic, 2021-04-26:
Hi @vitalikdm. I am glad you didn't raise an issue as the problem you are having is not an issue caused by this library.

Because you didn't get the same batch of panels, and it seems one of them is defective with the R & G pins swapped, you'll need to hack this library.

Around line 459 of ESP32-HUB75-MatrixPanel-I2S-DMA.cpp, you'll need to make it swap the blue and green values depending on where this defective panel sits in your chain on panels. Assuming it's the LAST panel in the chain you'll need to do something like:

```cpp
void IRAM_ATTR MatrixPanel_I2S_DMA::updateMatrixDMABuffer(int16_t x_coord, int16_t y_coord, uint8_t red, uint8_t green, uint8_t blue)
{
    if ( !initialized ) { 
      #if SERIAL_DEBUG 
              Serial.println(F("Cannot updateMatrixDMABuffer as setup failed!"));
      #endif         
      return;
    }

   // NEW CODE HACK FOR @vitalikdm 
  uint8_t swap = 0;
  if ( x_coord > 127)
  {
     swap = green;
     green = blue;
     blue = swap;
  }
   // END NEW CODE HACK FOR @vitalikdm 

  /* 1) Check that the co-ordinates are within range, or it'll break everything big time.
  * Valid co-ordinates are from 0 to (MATRIX_XXXX-1)
  */
  if ( x_coord < 0 || y_coord < 0 || x_coord >= PIXELS_PER_ROW || y_coord >= m_cfg.mx_height) {
    return;
  }
```

You're on you're own with this problem I'm afraid. You'll need to experiment.

vitalikdm, 2021-04-27:
Thank you for the fast reply. I tried the code above but without `if ( x_coord > 127)`  because I chained 2 new panels with FM6126B chip when green and blue are swapped.  The new code helped but not in all places (some text became green, some left blue).

It was fixed by changing wires without changing the code this way: B1-G1, B2-G2. After that everything works as expected. It looks like B and G were mixed up at the factory. There are no silkscreens on the boards. I just took it into account for boards with FM6126B

mrfaptastic, 2021-04-27:
Glad you found the fix. Indeed, there's always the option up flip the wires around 😁 !