mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #69: random pixel issue

tsctrl, 2021-01-22:
Hi @mrfaptastic ,

I am using dev branch to test the led using the double buffer example with 6 panel virtual matrix. i am getting additional fixed pixel display in several part of the panel as shown in image below:

[image]

i confirm this issue as comparison output from the master branch. 
1. the circle image are exist on both master and dev branch. only visible on 3rd and 6th panel on 6 chained panel.
2. the arrow are shifted pixel on every panel. is 10 pixel shift to be exact.

the dev config are default SHIFT HZ_10M 255 brightness and MinRefreshRate 100. 
changing refresh rate/ brightness/i2s freq/shift does not fix the issue.

mrfaptastic, 2021-01-22:
Can you also test with the earlier version of this library. v 1.0.2

There's no virtual panel class with this release, but you can still use it to test for artifacts. Just out of curiosity, what are you building with a 6 x 64px panel?

tsctrl, 2021-01-22:
Hi @mrfaptastic 

i have switching branch for this test to confirm. issue no 2 gone when using release v1.0.2 with no virtual panel. i just open the buffer_swap_test example. change the port and run. however still seeing issue no 1 but lesser. 
[image]
top: image on the v1.0.2. only visible in 3rd and 6th panel.

My finding lowering the current would hide the visible issue in no 1 for library in dev branch and master branch but not in the v1.0.2. lowering current in v1.0.2 does not impacting the display. i am using 30A bench power supply. 

[image]
top: clean display on lower current for dev and master branch

[image]
top: Issue 1. display on max current for master branch as stated in the initial comment. only visible in 3rd and 6th panel.

No issue if using single panel. let me try with fixed led power supply 40A and share the result later. i am building a notification board btw.

Thanks!

tsctrl, 2021-01-22:
I have switched to 4x4 panel. issue with panel no 3 done but still having artifacts in the last panel. Not sure if this is wiring/psu or a dupon issue.

Thanks!

vortigont, 2021-01-27:
You might try latest dev, there is a new method MatrixPanel_I2S_DMA::setLatBlanking(uint8_t v) to control the number of clocks output is disabled before/after LAT signal. May help with ghosting in some cases. Value coud be set between 1 (default) to 4.

tsctrl, 2021-01-30:
Hi @vortigont, thank you so much on the fixes.
The ghosting issue no 2. was fixed! thank you. but not able to see any difference when changing LatBlanking for 1 to 4 (bit dimmer in 4).

however issue 1. which showing random pixel in panel 3 and 6 for 6 panel, and panel 4 for 4 panel config still exist but not as bad as before. still 3 dots in panel 3 and quite a lot in panel 6 for 6 panel config. have used 60A power supply. still the same. but i am ok with it. probably issue with my dupon or because i am using 3 different panel in a chain or because i am using 1.5m idc cables. @mrfaptastic is this is as expected? are you able to replicate the issue on you end?

[image]
3 dots in panel 3.

thank you @mrfaptastic , @vortigont

tsctrl, 2021-01-30:
i am using DoubleBufferSwap.ino and add chained panel codes from ChainedPanels.ino. with 1 NUM_ROWS and 1 NUM_COLS to replicate the issue.

mrfaptastic, 2021-01-30:
Do you have any capacitors on each panel? You need to have at least 2000uf caps across VCC and GND on each panel to reduce noice which can cause artifacts.

tsctrl, 2021-01-30:
Did not have at the moment.

vortigont, 2021-01-31:
> i am using DoubleBufferSwap.ino and add chained panel codes from ChainedPanels.ino. with 1 NUM_ROWS and 1 NUM_COLS to replicate the issue.

I'm quite confused, Why are you using 1x1 configuration if you have 6 or 4 panels? Anyway it looks more like some hardware/interference issue.
LatBlanking might or might not help for some corner cases, just thought it'd better have this knob available to try than hardcode some constant value.

tsctrl, 2021-01-31:
using 1x1 configuration to draw the same pattern for all panels just to do the test. although i having same issue when not using virtual panel with 3x2 panel connected. i will swap all with same panel and recheck all pin configuration again.

tsctrl, 2021-02-01:
I have tested using different hardware and the display work in the master and dev branch with no false pixel. Closing this one

thank you.

closed 2021-02-01 by tsctrl

vortigont, 2021-02-01:
@tsctrl awesome! would appreciate if you could share some details what was the issue - cabling, power, panel ordering?