mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #70: ESP32 Task watchdog got triggered.

tsctrl, 2021-01-22:
Hi @vortigont 

i am having watchdog got triggered. when moving the matrix placeholder pointer declaration to the dedicated .h file.
Happens on the dev branch. How to place the matrix declaration outside the main files? i want to create specific files for the matrix. 

.h files include:
```
// placeholder for the matrix object
extern MatrixPanel_I2S_DMA* matrixPanel = nullptr;

// placeholder for the virtual display object 
extern  VirtualMatrixPanel* matrix = nullptr;
```

main .ino / .cpp file:
```
void setup() 
{
HUB75_I2S_CFG::i2s_pins _pins =
{ R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN };
HUB75_I2S_CFG mxconfig(
	PANEL_RES_X,   // module width
	PANEL_RES_Y,   // module height
	PANEL_CHAIN,   // chain length
	_pins,
	HUB75_I2S_CFG::shift_driver::SHIFT,
	true,
	HUB75_I2S_CFG::clk_speed::HZ_10M
	);

matrixPanel = new MatrixPanel_I2S_DMA(mxconfig);
matrixPanel->setBrightness8(225); 
matrixPanel->setMinRefreshRate(100);

// Allocate memory and start DMA display
if (not matrixPanel->begin()) {
Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
}

matrixPanel->clearScreen();
matrixPanel->print("JUST TO TRY");
matrixPanel->showDMABuffer();
}

```
Error stack trace:
E (6360) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6360) task_wdt:  - IDLE (CPU 0)
E (6360) task_wdt: Tasks currently running:
E (6360) task_wdt: CPU 0: main
E (6360) task_wdt: CPU 1: IDLE
E (6360) task_wdt: Aborting.

abort() was called at PC 0x400dc2b5 on core 0

Backtrace:0x40088484:0x3ffbe770 0x40088e8d:0x3ffbe790 0x4008e461:0x3ffbe7b0 0x400dc2b5:0x3ffbe830 0x400829f1:0x3ffbe850 0x400d4471:0x3ffbd200 0x400f07b0:0x3ffbd230

Decoding stack results
0x40088484: uxPortCompareSetExtram at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h line 94
0x40088e8d: pcTaskGetTaskName at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2331
0x400dc2b5: get_arg at ../../../.././newlib/libc/stdio/vfprintf.c line 2264
0x400829f1: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 46
0x400d4471: base_of_encoded_value at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-pe.h line 102

vortigont, 2021-01-22:
Not sure about your layout. Why do you need to put the declaration only to the other file but still keep all setup/work in main? You should define somewhere MatrixPanel_I2S_DMA* matrixPanel = nullptr;
and declare extern MatrixPanel_I2S_DMA* matrixPanel; in main

But better create you own class though than mess with external's, just a rough example:

```
//mymatrix.h

Class Mymatrix {
private:
 MatrixPanel_I2S_DMA* matrixPanel = nullptr;

public:
  void init(){
   matrixPanel = new MatrixPanel_I2S_DMA();
   matrixPanel->setBrightness8(225); 
  }
}

//main.cpp
#include mymatix.h
Mymatrix panel;

void setup(){
  panel.init();
}

```

tsctrl, 2021-01-23:
hi @vortigont, thank you. i created extern to make the matrix as static. so i can call the matrix function from multiple .c/cpp file. i have multiple .h and cpp files for different logic files that calling the matrix.
i would like to use the class approach as you ve mentioned. but because of the declaration of the class was on the main file as below:

`Mymatrix panel;`

i could only call panel inside my main class. how to call it from other .h files?

Thanks!

tsctrl, 2021-01-23:
Hi @vortigont,

i am able to create the classes as per your suggestion and can compile and without any issue. but still having the same error. my esp32 keep restarting once the matrix was initialized in different .h files, this issue did not happen on the master branch where i am able to refer the matrix from any files then the main file. 

below is the way i initialize the matrix:
```
class MatrixPanel {
	public:
		static void init();
	public:
		static MatrixPanel_I2S_DMA* getMatrixPanel();
		static VirtualMatrixPanel* getMatrix();
	private:
		static MatrixPanel_I2S_DMA *matrixPanel;
		static VirtualMatrixPanel *matrix;
};
```
and below is the error:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core  0 register dump:
PC      : 0x400dbaec  PS      : 0x00060d30  A0      : 0x800fb0ae  A1      : 0x3ffe3be0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x0000000f  A8      : 0x800dbaec  A9      : 0x3ffe3bc0  
A10     : 0x00000000  A11     : 0x3f41b50e  A12     : 0xffffffe9  A13     : 0x3f41b50b  
A14     : 0x00000000  A15     : 0x3ffb7a70  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x40086008  LEND    : 0x40086024  LCOUNT  : 0xffffffff  

Backtrace:0x400dbae9:0x3ffe3be0 0x400fb0ab:0x3ffe3c00 0x40081696:0x3ffe3c40 0x40079239:0x3ffe3c80 |<-CORRUPTED
ELF file SHA256: 007e3de72e92aa69

PC: 0x400dbaec: get_arg at ../../../.././newlib/libc/stdio/vfprintf.c line 2114
EXCVADDR: 0x0000000c

Decoding stack results
0x400dbae9: get_arg at ../../../.././newlib/libc/stdio/vfprintf.c line 2114
0x40081696: esp_timer_impl_set_alarm at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_timer_esp32.c line 161

I will try comment out the changes in dev branch to see where is the root cause. Appreciate if you could provide example on how to init the matrix and call the instance anywere within the project.

Thank you.

tsctrl, 2021-01-23:
Hi @vortigont ,

i could not reproduce the issue anymore. what i did is just build incrementally in each code comment and uncomment. adding curly braces in all if condition that was missing. not able to confirm the root cause. then seems the issue was not reproduceable. hence i am closing this issue.

Thank you!

closed 2021-01-23 by tsctrl