mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #72: Compiling Master > unable to locate FreeRTOS

curtischolon, 2021-02-04:
Hello @mrfaptastic ,

First, I'm very appreciative of the work you have done on this package. I've been able to create some interesting projects for my kids to enjoy and hopefully many more in the future.

I'm creating this issue because it seems that after setting up a new dev environment I am unable to compile your master branch, nor several prior tagged releases.

After install Adafruit_GFX and checking relevant dependencies, I get the following error message just trying to verify the stock code in the AuroraDemo and plasma.ini examples.

I would greatly appreciate your help, I'm currently running Arduino 1.8.13, along with fresh installs of the required dependencies.

```
Arduino: 1.8.13 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:3MB OTA:~512KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /home/curtis/git/ESP32-HUB75-MatrixPanel-I2S-DMA/examples/AuroraDemo/AuroraDemo.ino:24:0:
/home/curtis/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/ESP32-HUB75-MatrixPanel-I2S-DMA.h:70:31: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
                               ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Verbose Output:
```
Arduino: 1.8.13 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:3MB OTA:~512KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/home/curtis/Downloads/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/curtis/Downloads/arduino-1.8.13/hardware -hardware /home/curtis/.arduino15/packages -tools /home/curtis/Downloads/arduino-1.8.13/tools-builder -tools /home/curtis/Downloads/arduino-1.8.13/hardware/tools/avr -tools /home/curtis/.arduino15/packages -built-in-libraries /home/curtis/Downloads/arduino-1.8.13/libraries -libraries /home/curtis/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=160,vt=flash,exception=legacy,ssl=all,eesz=4M3M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10813 -build-path /tmp/arduino_build_982314 -warnings=none -build-cache /tmp/arduino_cache_796728 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/curtis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/home/curtis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/home/curtis/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/curtis/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=/home/curtis/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=/home/curtis/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=/home/curtis/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/home/curtis/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -verbose /home/curtis/git/ESP32-HUB75-MatrixPanel-I2S-DMA/examples/AuroraDemo/AuroraDemo.ino
/home/curtis/Downloads/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/curtis/Downloads/arduino-1.8.13/hardware -hardware /home/curtis/.arduino15/packages -tools /home/curtis/Downloads/arduino-1.8.13/tools-builder -tools /home/curtis/Downloads/arduino-1.8.13/hardware/tools/avr -tools /home/curtis/.arduino15/packages -built-in-libraries /home/curtis/Downloads/arduino-1.8.13/libraries -libraries /home/curtis/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=160,vt=flash,exception=legacy,ssl=all,eesz=4M3M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10813 -build-path /tmp/arduino_build_982314 -warnings=none -build-cache /tmp/arduino_cache_796728 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/curtis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/home/curtis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/home/curtis/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/curtis/.arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.mklittlefs.path=/home/curtis/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=/home/curtis/.arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=/home/curtis/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/home/curtis/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -verbose /home/curtis/git/ESP32-HUB75-MatrixPanel-I2S-DMA/examples/AuroraDemo/AuroraDemo.ino
Using board 'nodemcuv2' from platform in folder: /home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3
Using core 'esp8266' from platform in folder: /home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3
Detecting libraries used...
/home/curtis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/tools/sdk/include -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/tools/sdk/lwip2/include -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_982314/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/cores/esp8266 -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/variants/nodemcu /tmp/arduino_build_982314/sketch/AuroraDemo.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for ESP32-HUB75-MatrixPanel-I2S-DMA.h: [ESP32-HUB75-MatrixPanel-I2S-DMA-master@1.2.4]
ResolveLibrary(ESP32-HUB75-MatrixPanel-I2S-DMA.h)
  -> candidates: [ESP32-HUB75-MatrixPanel-I2S-DMA-master@1.2.4]
/home/curtis/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/tools/sdk/include -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/tools/sdk/lwip2/include -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_982314/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/cores/esp8266 -I/home/curtis/.arduino15/packages/esp8266/hardware/esp8266/2.7.3/variants/nodemcu -I/home/curtis/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master /tmp/arduino_build_982314/sketch/AuroraDemo.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for freertos/FreeRTOS.h: []
ResolveLibrary(freertos/FreeRTOS.h)
  -> candidates: []
In file included from /home/curtis/git/ESP32-HUB75-MatrixPanel-I2S-DMA/examples/AuroraDemo/AuroraDemo.ino:24:0:
/home/curtis/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master/ESP32-HUB75-MatrixPanel-I2S-DMA.h:70:31: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
                               ^
compilation terminated.
Using library ESP32-HUB75-MatrixPanel-I2S-DMA-master at version 1.2.4 in folder: /home/curtis/Arduino/libraries/ESP32-HUB75-MatrixPanel-I2S-DMA-master 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

DaveDavenport, 2021-02-04:
This library is for the ESP32 (esp32 sdk includes freertos), not for the ESP8266 (nodeMCU) that you are using.

For nodeMCU you want to use the PxMatrix library.

curtischolon, 2021-02-04:
@DaveDavenport - Thanks for the response, I literally just arrived at that conclusion after doing a deep dive into additional documentation. I did not realize that the compilation requirements were different for the respective boards.

Cheers,
Curtis

closed 2021-02-04 by curtischolon