mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #73: buffer allocation and clear buffer manually

tsctrl, 2021-02-07:
Hi @vortigont @mrfaptastic ,

how the buffer allocated and what changes affect the buffer size allocation?
how to make sure the display use the right amount of buffer and refresh the usage during runtime?
how to clear buffer manually or to refresh the display to clear up some buffer?

simply to ask, how to make sure the display run efficiently?
thank you at the earliest and regards,
Ts

mrfaptastic, 2021-02-07:
Short answer: Nothing that needs to be done / you need to do.

 DMA memory is allocated as efficiently as possible for the given resolution required on start-up (once).

vortigont, 2021-02-07:
just to add 5 cent's to @mrfaptastic reply - currently the fastest method to clear buffer to 'black' (doesn't mean releasing memory) is to use `fillScreen(0)`.
I have some ideas to implement methods to stop/resume DMA transfer to the matrix on-demand, releasing/allocating buffer memory (for example to use as a kind of "turn-off" switch for a matrix/cotroller). But I do not see any practical use for this for now, at least until I find some free time to research FM6126 HW brightness control feature.
Also there might be a benefit in saving buffer ram using reduced color depth, but it also requires some research and free time to implement, stay in touch :)

mrfaptastic, 2021-02-08:
I'm going to do some more testing this weekend if I get time... but I feel the dev branch is almost ready for release to the world.

If you decide to implement any further groundbreaking stuff like handling < 8 bit color depth etc. please create a new branch of dev.

mrfaptastic, 2021-02-08:
Once again though, great work and thanks for your efforts.

closed 2021-02-12 by tsctrl