mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #74: (some) ESP32 crashes on boot when allocating larger size (for chained panel)

BodoMinea, 2021-02-08:
Hello,

I am encountering a strange issue.

I am testing 4 ESP32 Devkit V1 boards, all from the same supplier, batch and visually identical. Out of these, two work properly with the library and two do not. Let me elaborate:
- all boards will happily drive a single HUB75 display (64x32 in my case, that is with `#define MATRIX_WIDTH 64` )
- two of the boards will happily drive two chained HUB75 displays (so 128x32, I set that up with `#define MATRIX_WIDTH 64*2`)
- two of the boards, with the exact same wiring, code, computer, power supply, panels etc will crash and boot-loop if in scenario 2 (with the code set up for two matrixes). Using these with the define set up for only one panel will work, but the two chained displays will obviously show the same image.

The following is recorded into Serial Monitor when the crash occurs:

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
ets Jun  8 2016 00:22:57
```

Apart from just not using those boards... is there anything else I can try? I am also afraid that purchasing other boards will make me encounter some more incompatible ones. Also, these boards that seemingly cannot drive 128x32 displays work fine otherwise - WiFi is working, they can switch a relay on/off or communicate with other devices via I2C.

The tests have been conducted with the latest release of the library, version 1.2.4.

BodoMinea, 2021-02-08:
This was instantly solved by switching the Flash Mode from Tools from QIO to DIO. I have no idea what that means or how it impacted the ability of the controller to drive a specific size of display, but now all the boards work flawlessly!

Kudos for developing such a great library, it is of tremendous help for me.

closed 2021-02-08 by BodoMinea

mrfaptastic, 2021-02-08:
These are the best kind of issues - I get the closer notification before I see it having been raised. Glad you like the library.