mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #76: Single Matrix 128x64 with FM6126A

hallard, 2021-02-10:
Hi There,

Anyone has succeed to use a 128x64 single panel 128x64 with FM6126A?

Display is correct but missing part, or superposed part, can't have full area.

As one single was not working, then tried to set it up with chained mode (virtual) 

- 2 x 64x64 2 rows 1 col
- 2 x 64x64 1 row 2 cols
- 4 x 64x32 2 rows 2 cols

No success at all, of someone got the correct config, I would give a try (I'm using latest dev branch)

Thanks

mrfaptastic, 2021-02-10:
Refer to this example, see if this works: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/dev/examples/3_FM6126Panel

hallard, 2021-02-10:
ah ah ah that did the trick as expected, not sure why it does not works with my code since I used the same init.

indeed I love the new init stuff which allow just to setup the overwritten pins.

```cpp
// It's my config to use with electrodragon I2S DMA
// https://www.electrodragon.com/product/rgb-matrix-panel-drive-interface-board-for-esp32-dma/
#if defined (ARDUINO_ESP32_PICO)
#define CH_D   22
#define CH_E   9
#define CLK    21
#elif defined (ARDUINO_ESP32_DEV)
#define CH_D   17
#define CH_E   18
#define CLK    16
#endif

  HUB75_I2S_CFG mxconfig( PANEL_RES_X,  PANEL_RES_Y, PANEL_CHAIN  );

#if defined (CH_D) && defined (CH_E) && defined (CLK)
  // override default pin mappings
  mxconfig.gpio.d = CH_D;                    
  mxconfig.gpio.e = CH_E;
  mxconfig.gpio.clk = CLK;     
#endif

  // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  mxconfig.driver = HUB75_I2S_CFG::FM6126A;     

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
```

mrfaptastic, 2021-02-10:
Glad it works :-)

Can you post a link to the panel you bought?

hallard, 2021-02-10:
Sure, but take care looks very fragile got one with an issue and looks the price raised, now 2 panels cost $61.4 and I paid 2 for $51.4 same seller one month ago.
https://www.aliexpress.com/item/1005001623824711.html

[image]
[image]

And here a picture of the back
[image]

vortigont, 2021-02-10:
Hmmm, interesting. So am I right that it is just two 64x64 chained in one case?

closed 2021-02-16 by mrfaptastic

daveythacher, 2021-03-24:
> Hmmm, interesting. So am I right that it is just two 64x64 chained in one case?

My guess is yes. They are generally speaking highly predictable. Outdoor and high density panels generally have the fun stuff in them. This is kind of mid range density panel. There are a few problems that pop up around them.

These modules have a few variations in terms of multiplex ratio, LED drivers, row scan topologies, etc. The panels target applications like outdoor (low multiplex for high brightness), high density, etc. Generally speaking there will be two RGB data chains. GCLK could be OE for memory mapped drivers. A, B, C, D, E can mean decoder, direct, shift register, or hybrid. Sometimes the multiplexing with full panel while on others it per data chain. Outdoor modules will sometimes strip the panels to in a few configurations to increase brightness and refresh traits.