mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #79: Task Handler to run Matrix

tsctrl, 2021-02-12:
Hi @mrfaptastic , @vortigont 

can implement task handler with priority for the led matrix? is this will help to init and render the display gracefully?

Thank you

tsctrl, 2021-02-12:
such as this code: while(!i2s_parallel_is_previous_buffer_free()) {}    
without taskDelay it will trigger task watchdogs

tsctrl, 2021-02-12:
TimeTask processing is running
E (7717) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (7717) task_wdt:  - IDLE (CPU 0)
E (7717) task_wdt: Tasks currently running:
E (7717) task_wdt: CPU 0: main
E (7717) task_wdt: CPU 1: IDLE
E (7717) task_wdt: Aborting.

abort() was called at PC 0x400e49c9 on core 0

```
0x400889a0: panic_abort at C:/Development/esp-idf/components/esp_system/panic.c line 367
0x40089365: esp_system_abort at C:/Development/esp-idf/components/esp_system/system_api.c line 112
0x4008ee9d: abort at C:/Development/esp-idf/components/newlib/abort.c line 46
0x400e49c9: task_wdt_isr at C:/Development/esp-idf/components/esp_common/src/task_wdt.c line 182
0x400da8f6: MatrixPanel_I2S_DMA::updateMatrixDMABuffer(short, short, unsigned char, unsigned char, unsigned char) at ../components/ESP32-HUB75-MatrixPanel-I2S-DMA/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp line 497
0x400daedd: MatrixPanel_I2S_DMA::updateMatrixDMABuffer(short, short, unsigned char, unsigned char, unsigned char) at ../components/ESP32-HUB75-MatrixPanel-I2S-DMA/src/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp line 447
0x400d57d5: VirtualMatrixPanel::drawPixel(short, short, unsigned short) at ../components/ESP32-HUB75-MatrixPanel-I2S-DMA/include/ESP32-HUB75-MatrixPanel-I2S-DMA.h line 562
0x400fb30a: Adafruit_GFX::writePixel(short, short, unsigned short) at ../components/arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.cpp line 191
0x400d811f: Adafruit_GFX::drawChar(short, short, unsigned char, unsigned short, unsigned short, unsigned char, unsigned char) at ../components/arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.cpp line 1225
0x400d82c2: Adafruit_GFX::write(unsigned char) at ../components/arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.cpp line 1277
0x400fb085: Print::write(unsigned char const*, unsigned int) at ../components/arduino/cores/esp32/Print.cpp line 42
0x400d6b13: Print::print(char const*) at ../components/arduino/cores/esp32/Print.h line 67
```

vortigont, 2021-02-12:
actually it should not take long to switch buff, if it causes WD trigger than something is wrong. Simply adding delay might lead to endless wait here. Could you provide an example sketch which triggers WD?

mrfaptastic, 2021-02-13:
> 
> 
> such as this code: while(!i2s_parallel_is_previous_buffer_free()) {}
> without taskDelay it will trigger task watchdogs

Not sure why this happens, could be sketch or idf specific. Anyway, added delay(1) to feed to watchdog.  
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/commit/77e0abff9057ad73d5700ee2be3ea502ce14622e

closed 2021-02-13 by mrfaptastic

tsctrl, 2021-02-13:
ya, probably the panel is too big and the memory is low :(