mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #80: Request for Enhancement (RFE)

tsctrl, 2021-02-13:
Hi,

Thank you again @mrfaptastic for this awesome library. 
I have feature request as below: 

1. Add support for lower color depth. down to 1?. 
2. Add max memory allocation like 50% of total memory up to 100% of memory.
3. Dynamic reclaim memory allocation from current consumption. Simply the panel not always show white all the time. so the allocation can be reduce dynamically? is this possible? So that when display in blank memory can be allocated to other process. not sure if clearFrameBuffer() is for this purpose.
4. Dynamic color depth. Color depth can be change on demand.
5. Current statistic resource usage fps, clk, memory, sequence etc. Small text in panel during display like benchmark concept.
6. Sometime the display are blocked with other slower process that running make the display are flickering. Is there any approach to see maybe like progress bar for one cycle to show that the panel process are deferred?

cc: @vortigont 

Thank you.

mrfaptastic, 2021-02-13:
Hi tsctrl,

Thanks for your kind words. 

1. Change the 'PIXEL_COLOR_DEPTH_BITS' define in the library .h file. Not sure what the output will look like.
2. I am not sure how this would work. If you want to make sure there's memory available in your sketch, check 'heap_caps_get_largest_free_block' function.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html
3. Memory cannot be changed dynamically. This library has to allocate memory for the DMA engine on start-up.
4. Memory is allocated according to color-depth and per 3) above so not possible.
5. Refer to 2.  & Enable 'DEBUG_MODE' in the .h file to see statistics.
6. The display shouldn't be flickering as it's sent via. DMA, but if animations are jerky then that's a byproduct of your sketch - you will need to experiment with pinning your animation task to another CPU.

The fact is, the ESP32 only has a small amount of RAM, and it isn't intended for chaining lots of panels. I would consider using a Raspberry Pi Nano W if you want to have a bigger display. This library isn't intended to chain more than a couple of panels.

tsctrl, 2021-02-13:
Hi @mrfaptastic , thank you for your reply.

yup for now i can only go up to 3 bits after enabling wifi, higher then 3, wifi will not work anymore. display as now still ok as i havent go with abstract animation on it as i not sure its purpose other than benchmark btw. 

migrate is the last choice. i will try the fruit and the pi. look like really overkill to runs a matrixs and i have to re-port all my codes. so far still ok. will see how i could balance and improve the sketch. hope to see espressif to have new multicore and bigger ram and more gpios mc in the future. so far esp32 and s2 are good choice. i could add more ram from the extra pins. will squeezz this bread till dry. btw, hope you can consider if there is option or approach to add the feature above in the feature.

cc: @vortigont 

Thank you so much!

vortigont, 2021-02-13:
Hi All,
although it might be interesing, but not easy (if possible) to implement
1) what's the fun in RGB's with 1-bit color? :)
2) it is all or nothing, either you allocate all the required ram for specific resolution or this won't work. The amount of memory could be calculated in advance. Maybe it would be easier to make some kind of script or table to estimate

Reduced color depth is interesting and may help to same lot's of RAM. Although seems to me color calculation for non 8-bit is broken currently, it looks ugly even with 7-bit color. Need to take deep dive into this.

4) it is possible to implement I believe, but what's the use? You can test some depths in advance and choose the one that suits you. Anyway, we need to fix ugly colors first (if possible). That was the reason I had not included color_depth option into the HUB75_CONFIG struct.

5) Nope, drawing on-screen is too complicated. Need to load font's, intercept draw functions, adjust to the topology etc... it will take more resources and efforts than the lib itself :)

> for now i can only go up to 3 bits after enabling wifi, higher then 3, wifi will not work anymore

Hmmm... I was able to drive 128x128 at 8 bits with WiFi and async server, worked fine. Had not tried to stress test the wifi, but at least some simple controls worked good for me.

tsctrl, 2021-02-14:
Hi,

Thank you so much @mrfaptastic and @vortigont for the reads and reply.
128x128 is higher than my setup. Definitely i got memory leaks somewhere. Will check and improve.

Closing this issue.
Thank you.

closed 2021-02-14 by tsctrl

mrfaptastic, 2021-02-14:
I have made a change to the library to improve visual colour quality at a lower colour-depth, and therefore you might be able to run the library at 5 or 6 bit colour and free up some memory as a result.

Change

` #define PIXEL_COLOR_DEPTH_BITS      8`

to say

` #define PIXEL_COLOR_DEPTH_BITS      5`

or 6, or 7.

See what the output looks like. If the image output looks bad, there is nothing that can be done, as less colour = less colour depth. But depending on if you're showing only text, or an image, you might not notice.

tsctrl, 2021-02-15:
Thank you so much @mrfaptastic for the changes!