mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #83: Cloned lines issue

radical-las, 2021-02-16:
When drawing any image on led panel it draws "cloned" lines. I made a little sketch, that should fill display line by line and writes to console number of line, that is drawing right now. I don't relly know what's wrong.
[link to sketch](https://pastebin.com/zvUZfdkx)
[link to video how it works](https://youtu.be/AMCZxoxgggA)

mrfaptastic, 2021-02-16:
Can you try an older version of this library ? 1.0.2  and see if this issue persists?

radical-las, 2021-02-16:
> 
> 
> Can you try an older version of this library ? 1.0.2 and see if this issue persists?

Yep, issue still persists.

mrfaptastic, 2021-02-16:
You'll need to use pxMatrix library sadly.

mrfaptastic, 2021-02-16:
> PS: my panel uses rul5358 (SHIFTREG_ABC_BIN_DE) decoder

You will need to use PxMatrix which supports this type of panel.

radical-las, 2021-02-16:
> 
> 
> You'll need to use pxMatrix library.

PxMatrix library should support my panel, but it uses very odd cabling

closed 2021-02-16 by mrfaptastic