mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #84: Debug Sketch

tsctrl, 2021-02-16:
Hi @mrfaptastic 

I have suggestions to add debug sketch which can be use as entry point for users to check compatibility and issues with the panel and library.

this probably a simple line scan sketch which could be a main sketch for issue investigation. by having this. when user want to create new issue. he could run this sketch and reports the behavior. This would help to reduce potential non issue or un necessary checking on your end. 

to add, this also give starting point for user to learn how the line feed and scan works. it also can be a sketch for benchmarking purpose which will reflect the effect on changing configuration such as row scan, bit depth, freq etc. by this simple sketch user can easily tune the configuration to meet their usage purposes.

this came as use case for new users who just step in and needs a library that could be complex to understand and require learning curve for then to write scan codes for debugging purposes.

thank you.

vortigont, 2021-02-16:
[This](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/PIO_TestPatterns) example could be a good starting point to begin with. I use it for various tests myself.

mrfaptastic, 2021-02-16:
Have now linked to @vortigont example in the README.md guide.

closed 2021-02-16 by mrfaptastic

vortigont, 2021-02-16:
Thanks, @mrfaptastic I will update an example with some comments to be more newbie friendly than

tsctrl, 2021-02-16:
hi @mrfaptastic , @vortigont thank you.

how to run the project? import as pio project in vs and run the command?

mrfaptastic, 2021-02-17:
Open the project folder and configure the project.ini as required for your setup.

If one is not totally up-to-speed with PlatformIO then I suggest checking youtube as there are some good tutorials. It is confusing to start with.

Alternatively, try the simple Arduino sketches like Test Patterns to test the setup.

vortigont, 2021-02-17:
If you are windows user than VSCode with PlatformIO plugin is your friend. But I run it under Linux shell usually :)
Also I believe it could be build with Arduino IDE too. Something like putting an empty src.ino file into src dir and opening this dir in IDE should work. But it was so long ago when I gave up struggling with ArduinoIDE, not sure.