mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #97: Problem to generate bitmap

tsctrl, 2021-02-25:
Hi @mrfaptastic ,

Generated weather icons bitmap thats in the example work well. but i am not able to get proper display for my own bytes generated from custom bitmap.

wondering what is the proper settings to generate the bitmap. only change to monochrome? how about the main, line scan, trailing bits? do you use custom script to generate the bytes?

thank you.

tsctrl, 2021-02-25:
one more, do we have to use the bitRead and the loop to render the image? no existing gfc function that can do this automatically?

mrfaptastic, 2021-02-27:
I'm confused as to the issue? Is it an issue with the example? i.e. How to generate a bitmap for the example?

tsctrl, 2021-02-28:
yes, how to generate the PROGMEM bitmap for the example. the normal bitmap which working with drawBitmap not working with the example.
how to create bitmap for the drawXbm565 monochrome or color?
how to utilize the drawXbm565 example and how to use drawXbm565 with bitmap with color information..

thank you @mrfaptastic

mrfaptastic, 2021-02-28:
Updated instructions: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/BitmapIcons

You can only draw a bitmap of a single colour. You will need to come up with your own solution if you want to draw full-colour images. Use: https://github.com/bitbank2/JPEGDEC or something.

closed 2021-02-28 by mrfaptastic