mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #98: ESP32 Bootloop

luxorbis, 2021-03-02:
### ESP32 bootlooping when using example sketch "2_Pattern_Plasma" on 64x64 Panel

Hey there! First things first: I successfully tested this library before with the example sketch "2_Pattern_Plasma". Of course I had to edit the code to get it to work properly with my 64x64 display (~81fps btw).
But when I added WiFi-funtionality to the ESP like [this](https://learn.sparkfun.com/tutorials/live-spotify-album-art-display), it had very inconsistent connecting issues. I figured the power wasn't sufficient and connected a 1mF Capacitor on its input, but since then the bootloop occured. I disconnected everything, ran the working example sketch but this time the bootloop persisted.
I tried erasing the flash and changing the "Flash Mode" like #74 but that didn't help either.

**Stack-Dump:**
Rebooting...
Fatal exception (29): StoreProhibited
epc1=0x400798b1, epc2=0x00000000, epc3=0x00000000, excvaddr=0xffffffff, depc=0x00000000
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
+++++++++++++++++++++++++++++++++++++++
+ESP32-HUB75-MatrixPanel-I2S-DMA DEMO       +
+++++++++++++++++++++++++++++++++++++++++
*** ESP32-HUB75-MatrixPanel-I2S-DMA: Memory Allocations Complete ***
Total memory that was reserved: 38 kB.
... of which was used for the DMA Linked List(s): 6 kB.
Heap Memory Available: 333196 bytes total. Largest free block: 117280 bytes.
General RAM Available: 248072 bytes total. Largest free block: 117280 bytes.
Fill screen: RED
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1f9f  PS      : 0x00060530  A0      : 0x800d12b8  A1      : 0x3ffb1f10  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x3ffb85c8  A5      : 0x3ffb84c8  
A6      : 0x00000000  A7      : 0xffffffc0  A8      : 0x00000009  A9      : 0x00000001  
A10     : 0x00000001  A11     : 0x3f4001c5  A12     : 0x00000000  A13     : 0x3ffb85c0  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x800d12b8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d1f9f:0x3ffb1f10 0x400d12b5:0x3ffb1f40 0x400d468a:0x3ffb1fb0 0x4008626d:0x3ffb1fd0

****Decoded Error-Code:****
PC: 0x400d1f9f: MatrixPanel_I2S_DMA::updateMatrixDMABuffer(unsigned char, unsigned char, unsigned char) at C:\Users\maxik\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-I2S-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp line 567
EXCVADDR: 0x800d12b8

Decoding stack results
0x400d1f9f: MatrixPanel_I2S_DMA::updateMatrixDMABuffer(unsigned char, unsigned char, unsigned char) at C:\Users\maxik\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-I2S-DMA-master\ESP32-HUB75-MatrixPanel-I2S-DMA.cpp line 567
0x400d12b5: setup() at C:\Users\maxik\Documents\Arduino\libraries\ESP32-HUB75-MatrixPanel-I2S-DMA-master/ESP32-HUB75-MatrixPanel-I2S-DMA.h line 693
0x400d468a: loopTask(void*) at C:\Users\maxik\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5\cores\esp32\main.cpp line 32
0x4008626d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Looking at the Error-Code, `dma_display->fillScreenRGB888(255, 0, 0);` seems to be where the problem first occurs.

I've read that my issue could be a problem where the ESP tries to reference a nullpointer, caused by a previous stack-overflow, but I don't know to fix it.

Your help is much appreciated!

mrfaptastic, 2021-03-02:
Post the code please

luxorbis, 2021-03-02:
Wow, got the ESP to work again!! I accidentally disconnected the HUB connector, then uploaded the code and what do you know, after connecting the cable again, it worked!! Apparently some pins cause the ESP to not flash or boot properly, but since I used the pins recommended as default in the example sketch, I do not know which pin(s) caused this issue.

***The pins (GPIO) I used:***
```
#define R1 25
#define G1 26
#define BL1 27
#define R2 14
#define G2 12
#define BL2 13
#define CH_A 23
#define CH_B 19
#define CH_C 5
#define CH_D 17
#define CH_E -1 // assign to any available pin if using two panels or 64x64 panels with 1/32 scan
#define CLK 16
#define LAT 4
#define OE 15

// Configure for your panel(s) as appropriate!
#define PANEL_WIDTH 64
#define PANEL_HEIGHT 64    // Panel height of 64 will required PIN_E to be defined.
#define PANELS_NUMBER 1   // Number of chained panels, if just a single panel, obviously set to 1
#define PIN_E 32

#define PANE_WIDTH PANEL_WIDTH * PANELS_NUMBER
#define PANE_HEIGHT PANEL_HEIGHT
```

mrfaptastic, 2021-03-02:
Not sure why that would cause a crash with the sketch though. ESP32 isn't happy with something electrically.

https://circuits4you.com/2018/12/31/esp32-devkit-esp32-wroom-gpio-pinout/

These ports are dangerous to have stuff connected to on boot:

    GPIO3 (U0RXD)
    GPIO1(U0TXD)

closed 2021-03-02 by mrfaptastic