mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #100: Add support for psram on wrover module

tsctrl, 2021-03-03:
Hi @mrfaptastic ,

sorry again to ask alot. 

i would like to suggest capability to allocate ram usage to use ps ram on wrover module. seems like adding this is not a big change.

adding flags or variable to switch malloc to ps_malloc should suffice to have the capability and allow more option to run higher bit graphic on the wrover module.

thank you!

mrfaptastic, 2021-03-03:
ESP32 doesn't support DMA from PSRAM

mrfaptastic, 2021-03-03:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/55

sheepov, 2021-04-15:
Hi,

Any issue using this ESP32-HUB75-MatrixPanel-I2S-DMA & PSRAM (wrover) ?
Do you have test the CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL ("Reserve this amount of bytes for data that specifically needs to be in DMA or internal memory") option ? 
=> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-spiram-malloc-reserve-internal

I need more ram for my project..
thx for your work!

mrfaptastic, 2021-04-15:
I don't understand what your question is.

sheepov, 2021-04-15:
is it possible to use this lib & PSRAM?

see README.md, memory constraints => it is not possible 
but  using "CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL" may solve the problem?

test using this lib with PSRAM enable on wrover : 
My program crash when i use this instruction  :
    if( not dma_display->begin(//mini esp prod
        12, //r1, 
        14, //g1, 
        27, //b1,
        22, //r2,
        23, //g2,
        2, //b2
        15, //a
        21, //b,
        16, //c,
        17, //d,
        34, //e,
        18, //lat,
        19, //oe,
        5 //clk
        ))
      Serial.println("****** !KABOOM! Insufficient memory - allocation failed ***********");      

=> no error msg but sometimes a weird character is printed : 13:40:00.630 -> l ⸮  

sheepov, 2021-04-16:
Hi,

I understand : this is not possible to use this lib from internal RAM and
enable PSRAM.

mrfaptastic, 2023-01-01:
PSRAM support is now available on ESP32-S3 only, but output throughput ('clock rate') is limited to 10Mhz due to hardware limitations.

20Mhz is available if you continue to use the limited SRAM - but then you face the same old problem of not being able to have a huge resolution..

In summary: There's not much benefit in using PSRAM as the low througput negates the value of having extra memory/possible chains of panels = flicker.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-DMA/discussions/343

closed 2024-08-24 by mrcodetastic