mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #99: Problem with 64x32 1/16 panel - row index not correct

mrRobot62, 2021-03-03:
Hi, got my 64x32 1/16 panel from aliexpress.
(this one: https://de.aliexpress.com/item/32870054325.html?spm=a2g0s.9042311.0.0.e7c34c4d9Kp83H)

I use on my esp32 default wiring.

Any hints, what I do wrong?

https://user-images.githubusercontent.com/949032/109812610-8fecd400-7c2c-11eb-8239-bcf1cd6821ba.mp4

`   HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
   HUB75_I2S_CFG mxconfig(MATRIX_WIDTH, MATRIX_HEIGHT, MATRIX_PANELS, _pins);
   mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;

   display = new MatrixPanel_I2S_DMA(mxconfig);`

Configuration copied from example file.

mrfaptastic, 2021-03-03:
Double, triple, quadruple check your A, B, C & D pin wiring.

Probably wrong. D_PIN and C_PIN are probably wrong way around.

mrRobot62, 2021-03-03:
ahhhhh - **shame on** me :-O

I swapped B & D wire :-O - Oh noooooo.

Ok, can start now to work on my RGMMatrixMenu lib

Thanks mrfaptastic !

closed 2021-03-03 by mrRobot62

mrfaptastic, 2021-03-03:
Happens to the best of us. Glad to help :)