mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #101: 128x64 matrix issues

bogd, 2021-03-13:
I am trying to use this library to control a 128x64 RGB module. However, I cannot get things to display properly.

When trying to run the simpleTestShapes example, this is what I get:
[text](https://bogdan.sass.ro/tmp/mx1.jpg)
[panel_with_X](https://bogdan.sass.ro/tmp/mx2.jpg)
[panel_with_X_and_circle](https://bogdan.sass.ro/tmp/mx3.jpg)

The patterns seem to be visible, but only half of the rows are lit.

The full code that I am using is below (basically, it is the exact simpleTestShapes sample code, to which I have added some additional settings to customize pins and width/height).

I know that this is most likely _not_ an issue with the library, but a problem with my code. If that is the case, please accept my apologies for posting here. And if you could point me in the right direction, I would really appreciate it! (even a simple link to a forum would be enough).

Thank you!

```
#define PANEL_WIDTH 128
#define MATRIX_WIDTH PANEL_WIDTH
#define PANEL_HEIGHT 64
#define MATRIX_HEIGHT PANEL_HEIGHT
#define NUM_LEDS PANEL_WIDTH*PANEL_HEIGHT

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// Change these to whatever suits
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 22
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

MatrixPanel_I2S_DMA *dma_display = nullptr;

// Input a value 0 to 255 to get a color value.
// The colours are a transition r - g - b - back to r.
// From: https://gist.github.com/davidegironi/3144efdc6d67e5df55438cc3cba613c8
uint16_t colorWheel(uint8_t pos) {
  if(pos < 85) {
    return dma_display->color565(pos * 3, 255 - pos * 3, 0);
  } else if(pos < 170) {
    pos -= 85;
    return dma_display->color565(255 - pos * 3, 0, pos * 3);
  } else {
    pos -= 170;
    return dma_display->color565(0, pos * 3, 255 - pos * 3);
  }
}

void drawText(int colorWheelOffset)
{
  
  // draw text with a rotating colour
  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // Don't wrap at end of line - will do ourselves

  dma_display->setCursor(5, 0);    // start at top left, with 8 pixel of spacing
  uint8_t w = 0;
  const char *str = "ESP32 DMA";
  for (w=0; w<strlen(str); w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print(str[w]);
  }

  dma_display->println();
  dma_display->print(" ");
  for (w=9; w<18; w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print("*");
  }
  
  dma_display->println();

  dma_display->setTextColor(dma_display->color444(15,15,15));
  dma_display->println("LED MATRIX!");

  // print each letter with a fixed rainbow color
  dma_display->setTextColor(dma_display->color444(0,8,15));
  dma_display->print('3');
  dma_display->setTextColor(dma_display->color444(15,4,0));
  dma_display->print('2');
  dma_display->setTextColor(dma_display->color444(15,15,0));
  dma_display->print('x');
  dma_display->setTextColor(dma_display->color444(8,15,0));
  dma_display->print('6');
  dma_display->setTextColor(dma_display->color444(8,0,15));
  dma_display->print('4');

  // Jump a half character
  dma_display->setCursor(34, 24);
  dma_display->setTextColor(dma_display->color444(0,15,15));
  dma_display->print("*");
  dma_display->setTextColor(dma_display->color444(15,0,0));
  dma_display->print('R');
  dma_display->setTextColor(dma_display->color444(0,15,0));
  dma_display->print('G');
  dma_display->setTextColor(dma_display->color444(0,0,15));
  dma_display->print("B");
  dma_display->setTextColor(dma_display->color444(15,0,8));
  dma_display->println("*");

}

void setup(){
  HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
  HUB75_I2S_CFG mxconfig(
	128, // Module width
	64, // Module height
	1, // chain length
	_pins // pin mapping
  );
  mxconfig.gpio.e = E_PIN;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

	// MUST DO THIS FIRST!
	dma_display->begin(); // Use default values for matrix dimentions and pins supplied within ESP32-HUB75-MatrixPanel-I2S-DMA.h
	// Draw a single white pixel
	dma_display->drawPixel(0,0, dma_display->color565(255,255,255)); // can do this after .begin() only
  dma_display->drawPixel(127,0, dma_display->color565(255,255,255)); // can do this after .begin() only
  dma_display->drawPixel(0,63, dma_display->color565(255,255,255)); // can do this after .begin() only
  dma_display->drawPixel(127,63, dma_display->color565(255,255,255)); // can do this after .begin() only

  
  // fix the screen with green
  dma_display->fillRect(0, 0, dma_display->width(), dma_display->height(), dma_display->color444(0, 15, 0));
  delay(500);

  // draw a box in yellow
  dma_display->drawRect(0, 0, dma_display->width(), dma_display->height(), dma_display->color444(15, 15, 0));
  delay(500);

  // draw an 'X' in red
  dma_display->drawLine(0, 0, dma_display->width()-1, dma_display->height()-1, dma_display->color444(15, 0, 0));
  dma_display->drawLine(dma_display->width()-1, 0, 0, dma_display->height()-1, dma_display->color444(15, 0, 0));
  delay(500);

  // draw a blue circle
  dma_display->drawCircle(10, 10, 10, dma_display->color444(0, 0, 15));
  delay(500);

  // fill a violet circle
  dma_display->fillCircle(40, 21, 10, dma_display->color444(15, 0, 15));
  delay(500);

  // fill the screen with 'black'
  dma_display->fillScreen(dma_display->color444(0, 0, 0));

  drawText(0);

}

uint8_t wheelval = 0;
void loop() {

    // animate by going through the colour wheel for the first two lines
    drawText(wheelval);
    wheelval +=1;

    delay(20); 
}

```

mrfaptastic, 2021-03-13:
Hi @bogd - Can you triple check your wiring? In particular the A,B,C,D,E pins. 

Somebody had a similar issue and it was a wiring issue. Perhaps also set

`mxconfig.driver = HUB75_I2S_CFG::FM6126A; ` 

In case it's some FM6126A panel or something.

bogd, 2021-03-13:
The panel is FM6124 (I checked before posting).

Indeed, the wiring might be problematic (at the moment, I am using jumper wires for everything, so there might be some loose contacts). But I did check, and I will check again - also, I have already ordered a dedicated PCB for the connection, so hopefully in about 1 week I will have a better alternative for connecting everything.

We can put this on hold until then (or even close it, if you prefer, since it is most likely not a library issue). To be honest, I posted here hoping that it was just something stupid that I did - or something stupidly simple that I forgot to do in the code :) .

mrfaptastic, 2021-03-13:
Well the shift driver definitely needs to be configured to FM6124 on setup if it's an FM6124 panel. Other than that, the issues look electrical from the pictures.

bogd, 2021-03-16:
As expected, it actually was a wiring issue. With blocks of 4 rows lit (and a binary shift register), it just had to be pin C :) . And lo and behold - instead of connecting it to IO5, I had managed to connect it to 3V3 :) . And, of course, I managed not to notice it when rechecking the wiring.

I will close this issue as a PICNIC (problem in chair, not in code). Thank you once again for taking the time to answer, and my apologies for not noticing my mistake!

closed 2021-03-16 by bogd