mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #105: What would be needed for 1/8th scan support

kustom666, 2021-03-28:
Hello!

First thing first: thank you so much for this incredible library.

I've sourced a lot of panels from china over the last year, as part of a bigger initiative. Most of my code is tied to this lib and I wouldn't want to switch to another since this one fits the bill perfectly as I need to be able to squeeze out as many frames as possible.

However, some of my batches are 1/8th scan rate which is expressly not supported. I'd be completely ok with working on a branch/fork that implements support for it, but while I don't mind digging through code, It be really awesome if you could give pointers as to where to look/what would be needed for 1/8th support. 

Thanks in advance!

mrfaptastic, 2021-03-28:
Glad to hear this library is of use!

I think your best bet would be to experiment with this example: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/P6_32x16_1_4_ScanPanel

Perhaps hack it to see if you can make it work for 1/8th scan. Would be a good addition to the library. The core DMA code the librray) won't need to be changed in this case.

mrfaptastic, 2021-03-28:
Just to clarify a bit more, lines 399-410 of https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/examples/P6_32x16_1_4_ScanPanel/QuarterScanMatrixPanel.h

Specifically line 401.

```  const int y_remap[] = { 0,1,8,9,4,5,12,13,16,17,24,25,22,23,30,31 };```

I suspect a solution for 1/8 scan panels may be able to be found with playing with this mapping table.

kustom666, 2021-03-28:
Awesome. Thanks for the pointers! I'll look into it and play with it. I'll PR if I get something satisfactory

kustom666, 2021-03-29:
Ok so after a whole day of working on those panels and almost pulling my entire head of hair over it, I've figured out that what I thought were 1/8th scan panels were actually multiplexed panels, where pixels basically can be turned on/written twice, the second time making the pixel brighter. 
This appears to be common on outdoors panels according to the documentation of rpi-rgb-led-matrix https://github.com/hzeller/rpi-rgb-led-matrix#types-of-displays

I've gotten it to work with a pi, rpi-rgb-led-matrix and the led-multiplex=1 option. 

The real challenge will be to make MatrixPanel-I2S-DMA work with it. 
Where do you think should I start? I'm not sure a remap trick would work?

kustom666, 2021-03-29:
I'm gonna close this issue in favor of a new once, since the core issue has changed

closed 2021-03-29 by kustom666