mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #106: I2S audio

dmb-220, 2021-03-29:
hello,
 can i2S audio be used?
https://github.com/schreibfaul1/ESP32-audioI2S
do you need any change? pin change understandable.
internet radio project, with rgb led matrix.

mrfaptastic, 2021-03-29:
Unlikely.  Not enough spare gpios.

dmb-220, 2021-03-29:
#define R1_PIN_DEFAULT  2
#define G1_PIN_DEFAULT  4
#define B1_PIN_DEFAULT  5
#define R2_PIN_DEFAULT  14
#define G2_PIN_DEFAULT  13
#define B2_PIN_DEFAULT  12

#define A_PIN_DEFAULT   23
#define B_PIN_DEFAULT   19
#define C_PIN_DEFAULT   17
#define D_PIN_DEFAULT   32
#define E_PIN_DEFAULT   33 
          
#define LAT_PIN_DEFAULT 18
#define OE_PIN_DEFAULT  15
#define CLK_PIN_DEFAULT 16

SPI I will not use, 

I2S 25, 26, 27 pins, web radio.

+i2c pins, get temperature, humidity.

interesings i2S audio works?

mrfaptastic, 2021-03-29:
You will have to experiment to find out if it works.

mrfaptastic, 2021-03-29:
If you actually have enough GPIO, I don’t see why that wouldn’t work. ESP32 will need to use I2S0 because this library uses I2S1. You should contact the developers of audioI2S for any other questions.

closed 2021-04-01 by mrfaptastic

mrweaver, 2021-11-03:
That actually may work out well if there is an intention for using the DACs to output the audio.

According the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html):
> I2S0 output can be routed directly to the digital-to-analog converter’s (DAC) output channels (GPIO 25 & GPIO 26) to produce direct analog output without involving any external I2S codecs. I2S0 can also be used for transmitting PDM (Pulse-density modulation) signals.

marianomd, 2023-10-25:
I'm trying this and it's not working. Using esphome I2S audio. I'm not sure if ESP32 allows two independent DMA transfers simultaneusly. Will report if I make any progress.

mrfaptastic, 2023-10-27:
It should. Try lowering the clock speed of the hub 75 output. Could be memory bandwidth related.

marianomd, 2023-10-29:
Also this:

[image]

https://www.youtube.com/watch?v=6PGrsZmYAJ0